betaflight/betaflight #9156: PYRODRONEF4 4.1.0 firmware

Whatsuppeter, 2019-11-09:
**Describe the bug**
After flashing my FC (PyroDronef4) to the new 4.1.0 (PYRODRONEF4) firmware, the Ports Tab does not show UART 2 or UART5.  My board has 6 UARTS and UART2 and UART5 are missing. My VTX is on UART5, so I can configure my VTX on BF 4.1.0
Also, when uploading the font in OSD tab, it freezes.
I typed in "defaults" in the CLI and nothing happened, it just booted me out and I had to connect again. Ports tab still missing the UART 2 and UART5
I came from BF 3.5.7 and when Re flashed 3.5.7 all UARTS showed up and the OSD font uploads correctly.

- Flight controller:  PyroDrone F4

Asizon, 2019-11-09:
Please, update your Betaflight Configurator version to 10.6.0
https://github.com/betaflight/betaflight-configurator/releases/tag/10.6.0

Whatsuppeter, 2019-11-09:
I am using the BF configurator version 10.6.0.

etracer65, 2019-11-09:
Nothing has changed for that target and it still has 6 UARTs defined. It's likely you made a mistake and flashed the incorrect target.

Whatsuppeter, 2019-11-09:
There is only one target for the PYRODRONEF4 4.1.0, which is the one I chose. 
There are several versions of the PyroDrone Hyperlite F4 FC. Mine is over a year a old, and i believe there have been two or three versions of this FC after mine. Could that be the problem?  Are the BF targets specific to all versions of a certain FC, or do they only work for the latest versions?

etracer65, 2019-11-09:
There is only a single target for this board and it hasn't changed in a long time. If it doesn't work with all revisions of the hardware than that's up to the manufacturer to resolve. Although I doubt that is the issue.

Have you simply tried flashing it again making sure you choose the correct target? Perhaps you mistakingly chose the wrong entry from the drop-down before. Be sure to do a full chip erase.

flatlander150, 2019-11-14:
Very similar problem downloaded latest version of betaflight, and flashed 4.1 and the only port I have now is the usb port.

etracer65, 2019-11-14:
@flatlander150 You accidentally flashed the wrong target. Flash your flight controller again and pick the correct target (and full chip erase).

mikeller, 2019-11-14:
@Whatsuppeter, @flatlander150: Since the manufacturer of this board does not collaborate with us or provide us with engineering samples of their hardware there is very little we can do to investigate / fix this at this point.
@bnn1044: You added the original PYRODRONEF4 target - are you the maintainer for it? If so, can you help here. Also, can you organise some engineering samples for us, or add the Unified Target configuration for this yourself?

flatlander150, 2019-11-14:
I have downloaded the target that 2 different sellers of this item has and they both appear to be the same...reflashed the board and did a full chip erase and the problem still remains the same. I'm not trying to start crap with anyone just trying to get in the air. I have also flashed the board online with recommended target several times. So apparently the manufacturer has listed the wrong target.

flatlander150, 2019-11-14:
I went to the manufacturers web site and found a setup guide that gave cli commands to enter and save that fixed the problem. My board was an iflight good luck to @Whatsuppeter check the manufacturer and see if they have addressed the problem of having unified targets by giving you some cli commands.

bnn1044, 2019-11-14:
> @Whatsuppeter, @flatlander150: Since the manufacturer of this board does not collaborate with us or provide us with engineering samples of their hardware there is very little we can do to [...]

I will add a unified target and test with the new FW. A lot of people running 4.1 on this board. No one report missing port yet. I will get to it. 
Thanks @mikeller

Whatsuppeter, 2019-11-14:
I was able to get the correct PYRODRONEF4 flash for BF 4.1 onto my FC. 
My first attempts was flashing the board from the hex file ( converted to dfu) grabbed from the GitHub site, and not from the beta flight configurator.  This was because I couldn’t get my dfu drivers working. 
Once I got the dfu drivers working properly, I was able to flash my FC board through the targets on BF configurator drop down menu. This flash and/or target worked fine. I’m not sure if the hex files on the GitHub site are outdated or not the same as the BF configurator options, but flashing through the configurator seemed to work out ok.

flatlander150, 2019-11-14:
Good for you, sometimes getting all these things to work together can be
enough to make you go insane.

mikeller, 2019-11-14:
@flatlander150:

> I went to the manufacturers web site and found a setup guide that gave cli commands to enter and save that fixed the problem.

I don't think that this is related, as you are using an entirely different board. But can you please post the link to the CLI commands that you found?

flatlander150, 2019-11-15:
****This** is for the iflight SucceX-D Mini F7 Twin G Stack**
826betaflight/unified-targets
 Code Issues 1 Pull requests 4 Actions Security Insights
unified-targets/configs/default/IFRC-IFF7_TWIN_G_D.config
@johnelliott johnelliott Add IFRC-IFF7_TWIN_G_D
85285d6 14 days ago
137 lines (129 sloc)  2.91 KB
  
# Betaflight / STM32F7X2 (S7X2) 4.0.3 Jun  1 2019 / 11:59:57 (094cfc956) MSP API: 1.41

board_name IFF7_TWIN_G_D
manufacturer_id IFRC

# resources
resource BEEPER 1 C15
resource MOTOR 1 C08
resource MOTOR 2 C06
resource MOTOR 3 C09
resource MOTOR 4 C07
resource MOTOR 5 B06
resource MOTOR 6 B07
resource MOTOR 7 B01
resource MOTOR 8 B00
resource PPM 1 A03
resource LED_STRIP 1 A01
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 C10
resource SERIAL_TX 5 C12
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 C11
resource SERIAL_RX 5 D02
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11
resource LED 1 C04
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 B03
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 B04
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
resource CAMERA_CONTROL 1 A00
resource ADC_BATT 1 C01
resource ADC_RSSI 1 C00
resource ADC_CURR 1 C02
resource PINIO 1 C13
resource PINIO 2 C14
resource PINIO 3 B08
resource FLASH_CS 1 B09
resource OSD_CS 1 B12
resource GYRO_EXTI 1 A08
resource GYRO_EXTI 2 B02
resource GYRO_CS 1 A15
resource GYRO_CS 2 C03

# timer
timer A00 AF2
# pin A00: TIM5 CH1 (AF2)
timer A03 AF3
# pin A03: TIM9 CH2 (AF3)
timer C08 AF3
# pin C08: TIM8 CH3 (AF3)
timer C06 AF3
# pin C06: TIM8 CH1 (AF3)
timer C09 AF3
# pin C09: TIM8 CH4 (AF3)
timer C07 AF3
# pin C07: TIM8 CH2 (AF3)
timer B06 AF2
# pin B06: TIM4 CH1 (AF2)
timer B07 AF2
# pin B07: TIM4 CH2 (AF2)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer B00 AF2
# pin B00: TIM3 CH3 (AF2)
timer A01 AF1
# pin A01: TIM2 CH2 (AF1)

# dma
dma ADC 3 0
# ADC 3: DMA2 Stream 0 Channel 2
dma pin A00 0
# pin A00: DMA1 Stream 2 Channel 6
dma pin C08 1
# pin C08: DMA2 Stream 4 Channel 7
dma pin C06 0
# pin C06: DMA2 Stream 2 Channel 0
dma pin C09 0
# pin C09: DMA2 Stream 7 Channel 7
dma pin C07 1
# pin C07: DMA2 Stream 3 Channel 7
dma pin B06 0
# pin B06: DMA1 Stream 0 Channel 2
dma pin B07 0
# pin B07: DMA1 Stream 3 Channel 2
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dma pin B00 0
# pin B00: DMA1 Stream 7 Channel 5
dma pin A01 0
# pin A01: DMA1 Stream 6 Channel 3

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature SOFTSERIAL
feature OSD

# serial
serial 1 64 115200 57600 0 115200

# master
set gyro_to_use = BOTH
set baro_bustype = I2C
set baro_i2c_device = 2
set mag_bustype = I2C
set mag_i2c_device = 2
set serialrx_provider = SBUS
set adc_device = 3
set blackbox_device = SPIFLASH
set dshot_burst = ON
set motor_pwm_protocol = DSHOT600
set current_meter = ADC
set battery_meter = ADC
set ibata_scale = 100
set beeper_inversion = ON
set beeper_od = OFF
set max7456_spi_bus = 2
set dashboard_i2c_bus = 2
set pinio_config = 129,1,1,1
set pinio_box = 0,255,255,255
set flash_spi_bus = 3
set gyro_1_bustype = SPI
set gyro_1_spibus = 1
set gyro_1_sensor_align = CW270
set gyro_2_spibus = 1
set gyro_2_sensor_align = DEFAULT

closed 2019-12-22 by stale[bot]