betaflight/betaflight #9157: mamba f405 mini mk2: no sbus input

illmillrig, 2019-11-09:
**Describe the bug**
no sbus input on mamba f405 mini mk2

**To Reproduce**
connect RXSR to sbus pad, configure FC for sbus.  don't get receiver signal to FC

**Expected behavior**
an sbus signal into the FC

**Flight controller configuration**
```
# diff

# version
# Betaflight / FURYF4OSD (FY4O) 4.1.0 Oct 16 2019 / 11:37:34 (c37a7c91a) MSP API: 1.42

# start the command batch
batch start

board_name FURYF4OSD

# master
set motor_pwm_protocol = DSHOT600

profile 0

rateprofile 0

# end the command batch
batch end

```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 1
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: BEEPER
A09: FREE
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: FREE
B00: MOTOR 2
B01: MOTOR 3
B02: FREE
B03: FLASH_CS
B04: LED 2
B05: LED 1
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: CAMERA_CONTROL
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: INVERTER 1
C01: ADC_BATT
C02: FREE
C03: FREE
C04: GYRO_EXTI
C05: USB_DETECT
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D02: FREE

Currently active Timers:
-----------------------
TIM1:
    CH3: MOTOR 3
TIM2:
    CH3: MOTOR 4
    CH4: MOTOR 1
TIM3:
    CH3: MOTOR 2
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11:
    CH1: CAMERA_CONTROL
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: MOTOR 4
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: MOTOR 1
DMA1 Stream 7: MOTOR 2
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: MOTOR 3
DMA2 Stream 7: FREE
```

**Setup / Versions**
- Mamba f405 mini mk2
- frsky RXSR receiver (in sbus mode - solid blue led)

illmillrig, 2019-11-09:
I think this FC is busted. im closing

closed 2019-11-09 by illmillrig

wookoouk, 2020-04-17:
Was this the case? because I am having the same issue

illmillrig, 2020-05-10:
> Was this the case? because I am having the same issue

no the FC was fine.  I re-downloaded new firmware from frsky and reflashed the rxsr.  worked fine afterwards.  Which is strange because I had just downloaded and flashed new firmware a few days prior.

mikeller, 2020-05-10:
@illmillrig: Might have been a bad FrSky firmware version - their firmware is a bit hit and miss.

wookoouk, 2020-05-10:
was exactly the same for me, flashing the r-xsr fixed it for me.

Martin Page
WookooUK@gmail.com

illmillrig, 2020-05-10:
> @illmillrig: Might have been a bad FrSky firmware version - their firmware is a bit hit and miss.

I think you're right.  actually, iirc frsky uploaded new firmware 3 days later after the previous upload.  So I think there was a very brief period where the new firmware for the receiver and firmware for the transmitter weren't compatible, and therefore two uploads of the same version firmware within a few days of each other.