betaflight/betaflight #9160: Blackbox configuration not saved

lowki, 2019-11-10:
Hi, I'm using betaflight configurator 10.6.0 on Linux, the board is a Kakute F4V2 flashed with 4.0.6 Firmware.

I can't save the blackbox configuration, I tried through the Betaflight tab and the cli, I tried *No logging* and *Onboard flash* (the one I want to configure), but whatever I do, it always fallback to *Serial Port* after Save and reboot.
All other parameters are saved correctly.

etracer65, 2019-11-10:
Please provide the information required when you opened the issue.

> **Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)
>
> Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

Also provide the output of the CLI command `flash_info`.

Lastly provide a screenshot of the Blackbox tab in the Configurator.

lowki, 2019-11-11:
Here the missing informations:
## Diff
````
# 

# diff

# version
# Betaflight / KAKUTEF4V2 (KTV2) 4.0.6 Sep  1 2019 / 00:43:53 (2a64051a2) MSP API: 1.41

# start the command batch
batch start

board_name KAKUTEF4V2
manufacturer_id 

# name
name humblebee

# resources
resource LED_STRIP 1 NONE
resource SERIAL_TX 11 C08

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature SOFTSERIAL
feature GPS
feature ESC_SENSOR

# beeper
beeper -ON_USB

# beacon

# map

# serial
serial 0 64 115200 57600 0 115200
serial 2 0 115200 57600 0 115200
serial 3 2048 115200 57600 0 115200
serial 4 1024 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1300 2100 0 0
aux 1 1 1 1300 1700 0 0
aux 2 2 1 1700 2100 0 0
aux 3 46 4 1700 2100 0 0
aux 4 13 3 1625 2100 0 0
aux 5 20 0 900 2100 0 0
aux 6 26 2 1300 2100 0 0
aux 7 31 5 1300 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set acc_trim_pitch = 2
set acc_trim_roll = -2
set fpv_mix_degrees = 35
set serialrx_provider = FPORT
set serialrx_inverted = ON
set motor_pwm_protocol = DSHOT1200
set failsafe_procedure = GPS-RESCUE
set current_meter = ADC
set ibata_scale = 1445
set gps_provider = UBLOX
set gps_rescue_descent_dist = 50
set tlm_halfduplex = ON
set osd_tim1 = 2562
set osd_rssi_pos = 2074
set osd_vtx_channel_pos = 2521
set osd_current_pos = 268
set osd_mah_drawn_pos = 2502
set osd_gps_lon_pos = 1
set osd_gps_lat_pos = 13
set osd_home_dir_pos = 2048
set osd_home_dist_pos = 2049
set osd_flight_dist_pos = 6
set osd_avg_cell_voltage_pos = 2496
set osd_battery_usage_pos = 2508
set osd_stat_max_dist = ON
set osd_stat_endbatt = ON
set osd_stat_flight_dist = ON
set vtx_band = 1
set vtx_power = 3
set vtx_freq = 5865

# profile
profile 0

# rateprofile
rateprofile 0

# end the command batch
batch end

````
## Resource show all
````
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: FREE
A02: MOTOR 4
A03: MOTOR 3
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: USB_DETECT
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: FREE
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FLASH_CS
B04: LED 2
B05: LED 1
B06: LED 3
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: FREE
B13: FREE
B14: OSD_CS
B15: INVERTER 3
C00: FREE
C01: FREE
C02: ADC_CURR
C03: ADC_BATT
C04: GYRO_CS 1
C05: GYRO_EXTI
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: FREE
C09: BEEPER
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D02: SERIAL_RX 5

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: MOTOR 4
DMA1 Stream 2: MOTOR 2
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: MOTOR 3
DMA1 Stream 7: MOTOR 1
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
````
## Flash_info
````
flash_info
Flash sectors=256, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=16777216, usedSize=0

#

## Blackbox tab screenshots
 ````
Screenshot before saving: 
[image]

Screenshot after saving:
[image]

etracer65, 2019-11-11:
Try setting it in the CLI as a test:
```
set blackbox_device = SPIFLASH
save
```

lowki, 2019-11-13:
> Try setting it in the CLI as a test:
> 
> ```
> set blackbox_device = SPIFLASH
> save
> ```

This worked.

closed 2019-12-20 by stale[bot]