betaflight/betaflight #9169: Incorrect GPS Altitude

s700ky, 2019-11-12:
**Describe the bug**
Upon Sat lock of 8+ it is still showing my altitude in the OSD as being 730ft

**To Reproduce**
Plug the battery in.

**Expected behavior**
I expect to see my altitude at around 0. My house is 600ft above sea level.

**Flight controller configuration**
```
# diff

# version
# Betaflight / STM32F405 (S405) 4.1.0 Oct 16 2019 / 11:57:16 (c37a7c91a) MSP API: 1.42
# manufacturer_id: MTKS   board_name: MATEKF405STD   custom defaults: YES

# start the command batch
batch start

board_name MATEKF405STD
manufacturer_id MTKS

# resources
resource MOTOR 1 C09
resource MOTOR 2 C08
resource MOTOR 3 C07
resource MOTOR 4 C06

# feature
feature -SOFTSERIAL
feature GPS

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TERA1234

# serial
serial 0 2 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 3 8192 115200 57600 0 115200

# aux
aux 0 0 0 1900 2100 0 0
aux 1 1 3 1900 2100 0 0
aux 2 2 3 1400 1600 0 0
aux 3 46 4 900 1100 0 0
aux 4 13 1 1900 2100 0 0
aux 5 35 2 1900 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 100 200 400 600

# master
set baro_hardware = NONE
set min_check = 1010
set max_check = 1990
set rssi_channel = 16
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set dshot_idle_value = 450
set failsafe_procedure = GPS-RESCUE
set small_angle = 180
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_rescue_angle = 35
set gps_rescue_ground_speed = 1500
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 6
set deadband = 2
set yaw_deadband = 2
set pid_process_denom = 1
set osd_units = IMPERIAL
set osd_warn_batt_not_full = OFF
set osd_warn_rssi = ON
set osd_cap_alarm = 1500
set osd_alt_alarm = 400
set osd_vbat_pos = 449
set osd_rssi_pos = 2106
set osd_tim_2_pos = 2519
set osd_flymode_pos = 2489
set osd_vtx_channel_pos = 2497
set osd_mah_drawn_pos = 2434
set osd_craft_name_pos = 44
set osd_gps_speed_pos = 2273
set osd_gps_lon_pos = 2506
set osd_gps_lat_pos = 2474
set osd_gps_sats_pos = 2101
set osd_home_dir_pos = 2243
set osd_home_dist_pos = 2265
set osd_altitude_pos = 2295
set osd_warnings_pos = 2377
set osd_avg_cell_voltage_pos = 2465
set osd_disarmed_pos = 2283
set osd_flip_arrow_pos = 270
set osd_stat_max_dist = ON
set osd_stat_min_batt = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_max_curr = OFF
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_gps_sats_show_hdop = ON
set vtx_band = 5
set vtx_channel = 5
set vtx_power = 1
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5806
set gyro_1_sensor_align = CW0

profile 0

# profile 0
set acro_trainer_angle_limit = 40

rateprofile 0

# rateprofile 0
set roll_rc_rate = 120
set pitch_rc_rate = 120
set yaw_rc_rate = 120
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 20
set roll_srate = 71
set pitch_srate = 71
set yaw_srate = 66

# end the command batch
batch end
```

Use `resource show all`
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: FREE
A02: FREE
A03: FREE
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: FREE
A14: LED 2
A15: FREE
B00: FREE
B01: FREE
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: LED 1
B10: OSD_CS
B11: FREE
B12: FREE
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: SDCARD_CS
C02: GYRO_CS 1
C03: GYRO_EXTI
C04: ADC_CURR
C05: ADC_BATT
C06: MOTOR 4
C07: MOTOR 3
C08: MOTOR 2
C09: MOTOR 1
C10: SERIAL_TX 3
C11: FREE
C12: FREE
C13: BEEPER
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH1: MOTOR 4
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH2: MOTOR 3
    CH3: MOTOR 2
    CH4: MOTOR 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: TIMUP 8
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

**Setup / Versions**
 - Flight controller MATEK F405ODS;
 - Other components R9MM OTA, IRC TRAMP, BN180 GPS

**Additional context**
Add any other context about the problem here.

IvoFPV, 2019-11-12:
I think it shows above sea level until you arm... Then it goes to above home point i.e. relative, above ground level

mikeller, 2019-11-12:
Duplicate of #8950.

closed 2019-11-12 by mikeller