betaflight/betaflight #9171: Manufacturer ID Request

JanikProphet, 2019-11-13:
Dear BF Development Team,
we would be thankful if you could add our company to the Manufacturer List and assign an ID.

Company: ManazGCS
Website: https://www.shop-mzgcs.de/

Thanks in advance

Janik Prophet from ManazGCS

etracer65, 2019-11-13:
Are you actually producing a custom flight controller? I don't see any evidence of such on your site.

JanikProphet, 2019-11-13:
> Are you actually producing a custom flight controller? I don't see any evidence of such on your site.

Hi @etracer65 we currently only have different kinds of our leds in store. I wanted to setup everything with you guys before offering the flightcontroller online. 
We are still making a few changes to the design but the flight controller will be available before the end of the year. 
Another flight controller specially designed for use in long range setups will be coming shortly after.

closed 2019-11-15 by mikeller