betaflight/betaflight #9172: MATEKF722SE(MTKS) Firmware doesn't have data flash as an option

Gabe-Brown, 2019-11-13:
The MATEKF722SE(MTKS) Firmware for the Matek f722 mini doesn't allow you to use the dataflash for blackbox logging but the legacy firmware works fine. Betaflight doesn't recognise the storage and doesn't have a drop down for it in the blackbox tab.

Steps to reproduce the behavior:
flashing a Matek F722 mini with betaflight 4.1.0 MATEKF722SE(MTKS) Firmware 16-10-2019

**Expected behavior**
A clear and concise description of what you expected to happen.
there to be an option for data flash in blackbox settings and betaflight to recognis the storage.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)
```# version
# Betaflight / MATEKF722SE (MF7S) 4.1.0 Oct 16 2019 / 11:44:08 (c37a7c91a) MSP API: 1.42

# start the command batch
batch start

board_name MATEKF722SE
manufacturer_id MTKS

# resources
resource MOTOR 1 B01
resource MOTOR 2 B04
resource MOTOR 4 B05

# feature
feature GPS
feature ESC_SENSOR

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 0 2 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

# aux
aux 0 0 0 900 1200 0 0
aux 1 1 1 1300 1700 0 0
aux 2 2 1 1700 2100 0 0
aux 3 46 4 1375 2100 0 0
aux 4 13 2 1300 1700 0 0
aux 5 26 3 900 1225 0 0
aux 6 31 3 1850 2100 0 0
aux 7 35 2 1700 2100 0 0

# master
set acc_calibration = 59,43,248
set baro_hardware = NONE
set rssi_channel = 12
set serialrx_provider = CRSF
set blackbox_device = SPIFLASH
set align_board_yaw = 45
set vbat_min_cell_voltage = 300
set vbat_scale = 108
set ibata_scale = 145
set ibata_offset = -3000
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_rescue_angle = 45
set gps_rescue_initial_alt = 70
set gps_rescue_descent_dist = 100
set gps_rescue_min_sats = 6
set gps_rescue_allow_arming_without_fix = ON
set osd_warn_rc_smoothing = OFF
set osd_warn_launch_control = OFF
set osd_rssi_alarm = 80
set osd_alt_alarm = 10000
set osd_tim1 = 3
set osd_tim2 = 1
set osd_vbat_pos = 2433
set osd_rssi_pos = 2167
set osd_tim_1_pos = 2486
set osd_vtx_channel_pos = 2081
set osd_current_pos = 2465
set osd_mah_drawn_pos = 2473
set osd_gps_speed_pos = 2145
set osd_gps_sats_pos = 2103
set osd_home_dir_pos = 2414
set osd_home_dist_pos = 2093
set osd_compass_bar_pos = 2058
set osd_warnings_pos = 14633
set osd_avg_cell_voltage_pos = 2401
set osd_core_temp_pos = 2273
set osd_stat_max_dist = ON
set osd_stat_min_batt = OFF
set osd_stat_battery = ON
set osd_stat_max_alt = ON
set osd_stat_flight_dist = ON
set osd_stat_total_time = ON
set debug_mode = SMARTAUDIO
set vtx_low_power_disarm = ON

profile 2

rateprofile 0

# rateprofile 0
set roll_rc_rate = 84
set pitch_rc_rate = 84
set yaw_rc_rate = 84
set roll_srate = 82
set pitch_srate = 82
set yaw_srate = 82

# end the command batch
batch end```

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.
```# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: ADC_EXT
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: FREE
B00: MOTOR 3
B01: MOTOR 1
B02: GYRO_CS 1
B03: FREE
B04: MOTOR 2
B05: MOTOR 4
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: SERIAL_TX 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: SERIAL_RX 6
C08: PINIO 1
C09: PINIO 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: BEEPER
C14: USB_DETECT
C15: GYRO_CS 2
D00: FREE
D01: FREE
D02: FLASH_CS
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH1: MOTOR 2
    CH2: MOTOR 4
    CH3: MOTOR 3
    CH4: MOTOR 1
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE```

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
Matek F722 mini 20x20mm from Unmanned Tech UK
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

IvoFPV, 2019-11-13:
>flashing a Matek F722 **mini** with betaflight 4.1.0 MATEKF722**SE**(MTKS) 
[image]

Gabe-Brown, 2019-11-13:
> flashing a Matek F722 mini with betaflight 4.1.0 MATEKF722SE(MTKS<

The matek website states the target is MATEKF722SE so that's what I used, there isn't any information on what that target is for that i could find.

IvoFPV, 2019-11-13:
Yes it will work on old legacy targets which will eventually be removed. But with unified targets you need to use the target I posted above, that's why it exists as MINI and that is the correct target.

etracer65, 2019-11-13:
In Betaflight 4.1 a new target was added specifically for the MATEKF722MINI. Use that. The MATEKF722SE target now only supports the sdcard version. Prior to BF4.1 you can use the MATEKF722SE target.

mikeller, 2019-11-13:
@MATEKSYS:

> The matek website states the target is MATEKF722SE so that's what I used, there isn't any information on what that target is for that i could find.

Can you please update the documentation?

MATEKSYS, 2019-11-14:
if users click "Firmware" button in F722-mini page, the details are available here. http://www.mateksys.com/?p=5159#tab-id-7

Gabe-Brown, 2019-11-14:
OK, thanks for correcting me guys, I was looking at the FC product page which still had the old target.

closed 2019-12-21 by stale[bot]