betaflight/betaflight #9173: RSSI Reading with no Aux channel on Crossfire WRONG

activeh2, 2019-11-14:
When using crossfire rssi with no aux channel it shows low readings around 40-60 which are wrong.

Using micro tx with crossfire nano receiver
Everyrhing updated to latest Versions.

activeh2, 2019-11-14:
`Entering CLI Mode, type 'exit' to return, or 

# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.1.0 Oct 16 2019 / 11:58:45 (c37a7c91a) MSP API: 1.42
# manufacturer_id: DIAT   board_name: MAMBAF722   custom defaults: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MAMBAF722
manufacturer_id DIAT
mcu_id 005100233237510e31313935
signature 

# name: justLook

# feature
feature -RX_PARALLEL_PWM

# beacon
beacon RX_LOST
beacon RX_SET

# map
map RETA1234

# serial
serial 0 0 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 4 64 115200 57600 0 115200

# aux
aux 0 0 1 1700 2100 0 0
aux 1 1 0 1900 2100 0 0
aux 2 27 2 1300 1700 0 0
aux 3 13 2 1925 2100 0 0
aux 4 35 0 1300 1700 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 4
vtxtable powervalues 14 20 26 30
vtxtable powerlabels 25 100 400 900

# master
set gyro_lowpass2_hz = 275
set dyn_lpf_gyro_min_hz = 220
set dyn_lpf_gyro_max_hz = 550
set acc_calibration = -91,11,-404
set serialrx_provider = CRSF
set dshot_bidir = ON
set bat_capacity = 1500
set ibata_scale = 277
set pid_process_denom = 1
set osd_vbat_pos = 2123
set osd_rssi_pos = 2445
set osd_current_pos = 2419
set osd_mah_drawn_pos = 2451
set osd_craft_name_pos = 2474
set osd_warnings_pos = 14441
set osd_avg_cell_voltage_pos = 2438
set osd_disarmed_pos = 2186
set vtx_band = 5
set vtx_channel = 7
set vtx_power = 4
set vtx_freq = 5880
set gyro_1_align_yaw = 1800
set name = justLook

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 77
set dyn_lpf_dterm_max_hz = 187
set dterm_lowpass2_hz = 165
set iterm_relax_type = GYRO
set p_pitch = 41
set p_roll = 38

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# `

activeh2, 2019-11-14:
Many other guys have this Problem too..

How to reproduce:

Flash newest Betaflight 4.1 on your Quad, bind with crossfire nano receiver, set CRSF protocol in Betaflight and set aux channel to none like in Joshua Bardwells Video [Video here.](https://www.youtube.com/watch?v=suAqRh2foq8&)

test RSSI with your quad on the workbench (2-3meters away and it drops to like 60-70.)

mikeller, 2019-11-14:
@activeh2: This is the RSS value that is reported by the CRSF RX - support for this was added in #8180. If you do not like the scaling of the values, you can always adjust them with `rssi_scale` and `rssi_offset` to the range you would like to have.

activeh2, 2019-11-14:
thank you very much, maybe this should be added to the Infos cause many People are asking for it and are confused

mikeller, 2019-11-14:
@activeh2: Good point. @laz2wiringpi: #8180 is your pull request - do you want to add some documentation for it?

laz2wiringpi, 2019-11-16:
@mikeller I had planed to do so but life sort of got in the way. :(

mikeller, 2019-11-16:
@laz2wiringpi: It does that...

closed 2019-12-23 by stale[bot]