betaflight/betaflight #9174: Servo Tilt set up motor one stops working

HABTomas, 2019-11-14:
**Describe the bug**
om Matek 
setting up servo tilt on DALRCF405 board with Betaflight 4.1. 
When I configure  resource servo 1 B01 instead of resource motor 6 B01  , servo starts to work but motor one stops working..
something I do wrong?? or is it a bug??
# resource
resource BEEPER 1 C13
resource MOTOR 1 B00
resource MOTOR 2 C06
resource MOTOR 3 A10
resource MOTOR 4 A08
resource MOTOR 5 C08
resource MOTOR 7 C07
resource MOTOR 8 C09
resource SERVO 1 B01
resource PPM 1 B15

**To Reproduce**
tried several times alvais the same, activata the Servo tilt funstion  adress AUX 3 to servo zero and cnage resource B01  from motor 6 to Servo 1
**Expected behavior**
I want the servo move and all motors also :o)

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up. servo connected to E6  motors E1 to E4

**Additional context**
# diff

# version
# Betaflight / STM32F405 (S405) 4.1.0 Oct 16 2019 / 11:57:16 (c37a7c91a) MSP API: 1.42
# manufacturer_id: DALR   board_name: DALRCF405   custom defaults: YES

# start the command batch
batch start

board_name DALRCF405
manufacturer_id DALR

# resources
resource MOTOR 6 NONE
resource SERVO 1 B01

# servo
servo 0 1200 1800 1500 -50 6
servo 1 1000 2000 1500 100 6

# feature
feature SERVO_TILT

# serial
serial 0 64 115200 57600 0 115200
serial 4 1 115200 57600 0 115200

# aux
aux 0 0 0 900 1175 0 0
aux 1 1 1 1300 1700 0 0
aux 2 2 1 1700 2100 0 0
aux 3 13 3 900 1300 0 0
aux 4 32 2 1300 1700 0 0

# master
set mag_hardware = NONE
set min_throttle = 1070
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 480
set yaw_motors_reversed = ON
set channel_forwarding_start = 6
set deadband = 1
set yaw_deadband = 1
set pid_process_denom = 1

profile 0

rateprofile 0

# rateprofile 0
set roll_srate = 60
set pitch_srate = 60
set yaw_srate = 60

# end the command batch
batch end

HABTomas, 2019-11-15:
hello 
what info is missing?? sorry I don´t know.
this id connected to DJI FPV system and DJI radio , input thru S-bus. 
servo is Hitec HS5070
best regards
Tomas

mikeller, 2019-11-15:
> Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

HABTomas, 2019-11-15:
hello do you mean like this?
# Building AutoComplete Cache ... Done!
# 
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: CAMERA_CONTROL
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 4
A09: FREE
A10: MOTOR 3
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: OSD_CS
B00: MOTOR 1
B01: FREE
B02: FREE
B03: SPI_SCK 1
B04: FREE
B05: SPI_MOSI 3
B06: FREE
B07: SERIAL_RX 1
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: FREE
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: SPI_MOSI 2
C04: GYRO_EXTI
C05: FREE
C06: MOTOR 2
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SERIAL_TX 5
C13: BEEPER
C14: LED 1
C15: FREE
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: MOTOR 4
    CH3: MOTOR 3
TIM2:
    CH1: CAMERA_CONTROL
TIM3:
    CH3: MOTOR 1
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: MOTOR 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: TIMUP 8
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: TIMUP 1
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

HABTomas, 2019-11-15:
right now I have changed it so the motor works , adres servo two to B01 that makes the motor one work but not the Servo .
this is how it looks with the servo working but not motor 1
# Building AutoComplete Cache ... Done!
# 
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: CAMERA_CONTROL
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 4
A09: FREE
A10: MOTOR 3
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: OSD_CS
B00: MOTOR 1
B01: SERVO 1
B02: FREE
B03: SPI_SCK 1
B04: FREE
B05: SPI_MOSI 3
B06: FREE
B07: SERIAL_RX 1
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: FREE
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: SPI_MOSI 2
C04: GYRO_EXTI
C05: FREE
C06: MOTOR 2
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SERIAL_TX 5
C13: BEEPER
C14: LED 1
C15: FREE
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: MOTOR 4
    CH3: MOTOR 3
TIM2:
    CH1: CAMERA_CONTROL
TIM3:
    CH3: MOTOR 1
    CH4: SERVO 1
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: MOTOR 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: TIMUP 8
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: TIMUP 1
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

mikeller, 2019-11-15:
@HABTomas:

Bingo!

And there's your problem:

>TIM3:
CH3: MOTOR 1
CH4: SERVO 1

Having a motor and a servo on the same timer won't work - you'll have to move your servo to a pin that isn't sharing a timer with a motor, or change the timer for the servo pin (`timer b1 list` for a list of options, then `timer b1 <option> to configure).

HABTomas, 2019-11-15:
hi Thanks
now it loks like this  but the servo is not moving
# Building AutoComplete Cache ... Done!
# 
# timer b1 list
# AF1: TIM1 CH3N
# AF2: TIM3 CH4
# AF3: TIM8 CH3N

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: CAMERA_CONTROL
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 4
A09: FREE
A10: MOTOR 3
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: OSD_CS
B00: MOTOR 1
B01: SERVO 1
B02: FREE
B03: SPI_SCK 1
B04: FREE
B05: SPI_MOSI 3
B06: FREE
B07: SERIAL_RX 1
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: FREE
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: SPI_MOSI 2
C04: GYRO_EXTI
C05: FREE
C06: MOTOR 2
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SERIAL_TX 5
C13: BEEPER
C14: LED 1
C15: FREE
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: MOTOR 4
    CH3: MOTOR 3
TIM2:
    CH1: CAMERA_CONTROL
TIM3:
    CH3: MOTOR 1
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: MOTOR 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: TIMUP 8
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: TIMUP 1
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

HABTomas, 2019-11-15:
there is no servo on timer now, how do I ad one??

mikeller, 2019-11-15:
@HABTomas: Looks like you managed to delete the timer for your servo. But it does not look good for it on B1 anyway:

>timer b1 list
AF1: TIM1 CH3N
AF2: TIM3 CH4
AF3: TIM8 CH3N

>TIM1:
CH1: MOTOR 4
CH3: MOTOR 3
TIM2:
CH1: CAMERA_CONTROL
TIM3:
CH3: MOTOR 1
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
CH1: MOTOR 2

=> All the timers that are available for B1 are used by at least one motor.

But: `timer c6 list` 

>`# AF2: TIM3 CH1`
`# AF3: TIM8 CH1`

What you could try:

`timer c6 af2` => move motor 2 onto timer 3

`timer b1 af3` => move the servo onto the now freed timer 8

HABTomas, 2019-11-15:
now it is Working THANKS!!!!!!!!!!!!!!!!!!!!!!

HABTomas, 2019-12-01:
Now the servo tilt is not working again , wanted to test RPM filters and activated them , that made two motors not to start  after armin had to rearm several times before they started to work and the servo stopet moving , I updatetd the software with latest firmvare with old settings and the servo 
 don´t move . flashed it again and started from zero , motors are runing godd now but servo is not working.  did all the setupp around the servo tilt , servo one and two is moving in motor tab , used resource to remap the servo instead of motor 6 and tested also instead of LED.
nothing works! :o( .  what do I do wrong?? sorry for taking up your time but the servo tilt is acualy working realy god on a racequad and it is totaly flyable with that function active. Great for landing also. 

diff gives me this
board_name DALRCF405
manufacturer_id DALR

# resources
resource SERVO 2 B06
resource LED_STRIP 1 NONE

# servo
servo 0 1000 2000 1500 100 6
servo 1 1000 2000 1500 100 6

# feature
feature SERVO_TILT

# serial
serial 0 64 115200 57600 0 115200
serial 4 1 115200 57600 0 115200

# aux
aux 0 0 0 950 1050 0 0
aux 1 1 1 1450 1550 0 0
aux 2 2 1 1950 2050 0 0
aux 3 8 2 900 2075 0 0
aux 4 13 3 950 1050 0 0

# master
set gyro_sync_denom = 2
set mag_hardware = NONE
set fpv_mix_degrees = 14
set min_throttle = 1070
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT300
set motor_pwm_rate = 480
set yaw_motors_reversed = ON
set channel_forwarding_start = 6
set pid_process_denom = 1

profile 0

rateprofile 0

# rateprofile 0
set roll_srate = 60
set pitch_srate = 60
set yaw_srate = 60

# end the command batch
batch en

resource show all
Give me this, don´t understad why there is not 4 motor timers.
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: CAMERA_CONTROL
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 4
A09: FREE
A10: MOTOR 3
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: OSD_CS
B00: MOTOR 1
B01: FREE
B02: FREE
B03: SPI_SCK 1
B04: FREE
B05: SPI_MOSI 3
B06: FREE
B07: SERIAL_RX 1
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: FREE
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: SPI_MOSI 2
C04: GYRO_EXTI
C05: FREE
C06: MOTOR 2
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SERIAL_TX 5
C13: BEEPER
C14: LED 1
C15: FREE
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH1: CAMERA_CONTROL
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
    CH2: DSHOT_BITBANG 3
    CH3: DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: DSHOT_BITBANG 3
DMA2 Stream 4: DSHOT_BITBANG 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

mikeller, 2019-12-02:
@HABTomas:

> resource SERVO 2 B06

This won't work, as servos are initialised from 1 onwards, stopping with the first servo that isn't defined. So defining a resource for servo 2 but none for servo 1 means that servo 2 is never initialised.

HABTomas, 2019-12-02:
hi I have tested with servo one before, now I have changed it to ene again , but it is still not working .
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: CAMERA_CONTROL
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 4
A09: FREE
A10: MOTOR 3
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: OSD_CS
B00: MOTOR 1
B01: FREE
B02: FREE
B03: SPI_SCK 1
B04: FREE
B05: SPI_MOSI 3
B06: SERVO 1
B07: SERIAL_RX 1
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: FREE
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: SPI_MOSI 2
C04: GYRO_EXTI
C05: FREE
C06: MOTOR 2
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SERIAL_TX 5
C13: BEEPER
C14: LED 1
C15: FREE
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH1: CAMERA_CONTROL
TIM3: FREE
TIM4:
    CH1: SERVO 1
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
    CH2: DSHOT_BITBANG 3
    CH3: DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: DSHOT_BITBANG 3
DMA2 Stream 4: DSHOT_BITBANG 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

jflyper, 2019-12-02:
We need to see your config diff.

HABTomas, 2019-12-02:
hi 
here it is , 
# diff

# version
# Betaflight / STM32F405 (S405) 4.1.1 Nov 15 2019 / 12:54:53 (1e5e3d369) MSP API: 1.42
# manufacturer_id: DALR   board_name: DALRCF405   custom defaults: YES

# start the command batch
batch start

board_name DALRCF405
manufacturer_id DALR

# resources
resource MOTOR 6 NONE
resource SERVO 1 B06
resource SERVO 2 B01
resource LED_STRIP 1 NONE

# servo
servo 0 1000 2000 1500 100 6
servo 1 1000 2000 1500 100 6

# feature
feature SERVO_TILT

# serial
serial 0 64 115200 57600 0 115200
serial 4 1 115200 57600 0 115200

# aux
aux 0 0 0 950 1050 0 0
aux 1 1 1 1450 1550 0 0
aux 2 2 1 1950 2050 0 0
aux 3 8 2 1300 1825 0 0
aux 4 13 3 950 1050 0 0

# master
set gyro_sync_denom = 2
set gyro_lowpass2_hz = 325
set dyn_lpf_gyro_min_hz = 260
set dyn_lpf_gyro_max_hz = 650
set mag_hardware = NONE
set fpv_mix_degrees = 14
set blackbox_p_ratio = 0
set min_throttle = 1070
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT300
set motor_pwm_rate = 480
set yaw_motors_reversed = ON
set channel_forwarding_start = 6
set pid_process_denom = 1

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 91
set dyn_lpf_dterm_max_hz = 221
set dterm_lowpass2_hz = 195

rateprofile 0

# rateprofile 0
set roll_srate = 60
set pitch_srate = 60
set yaw_srate = 60

# end the command batch
batch end

jflyper, 2019-12-02:
@HABTomas I think you had a configuration procedure problem.

Try again starting from another full chip erase flash of the `DALRCF405 (DALR)` target and paste your config diff above.

I've done it, and got correct timers:
```
# timer show

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH1: CAMERA_CONTROL
TIM3:
    CH4: SERVO 2
TIM4:
    CH1: SERVO 1
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
    CH2: DSHOT_BITBANG 3
    CH3: DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE
```

And please remember to enclose your paste from your CLI with three back ticks on the line by them selves to avoid false formatting by markdown.

HABTomas, 2019-12-02:
hi made it again from scrach , activated servo _tilt, usese sero one and two on servotab with AUX 3.
in motortab servo one and two is moving , go to resource and change motor 6 to servo one.
servo is not moving . any ideas? servo tested , 5v is there .
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: CAMERA_CONTROL
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 4
A09: FREE
A10: MOTOR 3
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: OSD_CS
B00: MOTOR 1
B01: SERVO 1
B02: FREE
B03: SPI_SCK 1
B04: FREE
B05: SPI_MOSI 3
B06: FREE
B07: SERIAL_RX 1
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: FREE
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: SPI_MOSI 2
C04: GYRO_EXTI
C05: FREE
C06: MOTOR 2
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SERIAL_TX 5
C13: BEEPER
C14: LED 1
C15: FREE
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH1: CAMERA_CONTROL
TIM3:
    CH4: SERVO 1
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
    CH2: DSHOT_BITBANG 3
    CH3: DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: DSHOT_BITBANG 3
DMA2 Stream 4: DSHOT_BITBANG 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# resource
resource BEEPER 1 C13
resource MOTOR 1 B00
resource MOTOR 2 C06
resource MOTOR 3 A10
resource MOTOR 4 A08
resource MOTOR 5 C08
resource MOTOR 7 C07
resource MOTOR 8 C09
resource SERVO 1 B01
resource PPM 1 B15
resource LED_STRIP 1 B06
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_RX 1 B07
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource I2C_SCL 1 B08
resource I2C_SDA 1 B09
resource LED 1 C14
resource SPI_SCK 1 B03
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 C03
resource SPI_MOSI 3 B05
resource ESCSERIAL 1 A03
resource CAMERA_CONTROL 1 A05
resource ADC_BATT 1 C02
resource ADC_RSSI 1 A00
resource ADC_CURR 1 C01
resource FLASH_CS 1 B12
resource OSD_CS 1 A15
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 A04

# version
# Betaflight / STM32F405 (S405) 4.1.1 Nov 15 2019 / 12:54:53 (1e5e3d369) MSP API: 1.42
# manufacturer_id: DALR   board_name: DALRCF405   custom defaults: YES

# start the command batch
batch start

board_name DALRCF405
manufacturer_id DALR

# resources
resource MOTOR 6 NONE
resource SERVO 1 B01

# servo
servo 0 1000 2000 1500 100 6
servo 1 1000 2000 1500 100 6

# feature
feature SERVO_TILT

# serial
serial 0 64 115200 57600 0 115200
serial 4 1 115200 57600 0 115200

# aux
aux 0 0 0 925 1050 0 0
aux 1 1 1 1300 1700 0 0
aux 2 2 1 1825 2100 0 0
aux 3 8 2 1300 1700 0 0
aux 4 13 3 900 1175 0 0

# master
set gyro_sync_denom = 2
set mag_hardware = NONE
set baro_hardware = NONE
set fpv_mix_degrees = 15
set min_throttle = 1000
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_pwm_rate = 16000
set pid_process_denom = 1

profile 0

rateprofile 0

# end the command batch
batch end

jflyper, 2019-12-02:
I don't understand what you are doing. The complete config you've pasted already has servos 1 and 2 configured on PB6 and PB1 respectively, and you seem to have overwritten the servo 2 config on PB1 with servo 1 resulting only servo 1 on PB1.

HABTomas, 2019-12-03:
hello 
I try to do the servo_tilt. like this.
started with reset of the board. 
Activate servo _tilt,  
Activate servo one  in servo tab with AUX 3 chanell
in motortab servo one and two is moving ,
In  resource I  change motor 6 on B01 to servo 1 on B01
that is all.
I only want one servo to move.
loks like this in settings and the servo on motor 6 out is not moving.
thansk for your help!!!!

# diff

# version
# Betaflight / STM32F405 (S405) 4.1.1 Nov 15 2019 / 12:54:53 (1e5e3d369) MSP API: 1.42
# manufacturer_id: DALR   board_name: DALRCF405   custom defaults: YES

# start the command batch
batch start

board_name DALRCF405
manufacturer_id DALR

# resources
resource MOTOR 6 NONE
resource SERVO 1 B01

# servo
servo 0 1000 2000 1500 100 6

# feature
feature SERVO_TILT

# serial
serial 0 64 115200 57600 0 115200
serial 4 1 115200 57600 0 115200

# aux
aux 0 0 0 925 1050 0 0
aux 1 1 1 1300 1700 0 0
aux 2 2 1 1825 2100 0 0
aux 3 8 2 1300 1700 0 0
aux 4 13 3 900 1175 0 0

# master
set gyro_sync_denom = 2
set mag_hardware = NONE
set baro_hardware = NONE
set fpv_mix_degrees = 15
set min_throttle = 1000
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_pwm_rate = 16000
set yaw_motors_reversed = ON
set pid_process_denom = 1

profile 0

rateprofile 0

# end the command batch
batch end

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: CAMERA_CONTROL
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 4
A09: FREE
A10: MOTOR 3
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: OSD_CS
B00: MOTOR 1
B01: SERVO 1
B02: FREE
B03: SPI_SCK 1
B04: FREE
B05: SPI_MOSI 3
B06: FREE
B07: SERIAL_RX 1
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: FREE
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: SPI_MOSI 2
C04: GYRO_EXTI
C05: FREE
C06: MOTOR 2
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SERIAL_TX 5
C13: BEEPER
C14: LED 1
C15: FREE
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH1: CAMERA_CONTROL
TIM3:
    CH4: SERVO 1
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
    CH2: DSHOT_BITBANG 3
    CH3: DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: DSHOT_BITBANG 3
DMA2 Stream 4: DSHOT_BITBANG 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

closed 2020-01-09 by stale[bot]