betaflight/betaflight #9176: Incorrect RC rate when using a TBS Crossfire Micro RX and TBS Unify EVO with CSFR

bizmar, 2019-11-15:
**Describe the bug**
The detected active RC packet interval for TBS Crossfire at 150hz should be 6.6ms, but it's being detected at 5ms or even faster. Causing the active RC input and derivative filtering cutoff to be set as high as 100hz, making the setpoint very jagged. 

**To Reproduce**
This configuration of connected devices causes the detected CSFR packet rate to be wrong. The Crossfire Micro RX is relaying telemetry data from the FC, through outputs 3 and 4 to the connected Unify EVO, so the Unify can display its own OSD with VBAT, RSSI...  
[image]
(the two FCs represent both sides of one actual FC)

**Expected behavior**
The active RC rate is read as 6.6ms (150hz mode) or 20ms (50hz mode)

**Setup / Versions**
MATEKF722SE
BF 4.1.0 and 4.0.5
TBS Crossfire Micro TX
TBS Unify EVO VTX

**Additional context**
I'm reporting this issue on behalf of a friend, who was in touch with TBS, here is an excerpt of their reply which has lead me to open the issue here. 

_"This looks more like a betaflight issue than a TBS issue. a lower number in RC rate means more frames. the evo emits some frames as well which are forwarded to all CRSF ports which is the FC in this case. It seems like betaflight counts all the frames and not just the RC ones."_
The whole exchange is [here](https://app.sparkmailapp.com/web-share/L22wTbY9mmUI2TcOxVBgE9UmKRim2oPw66JDeiQq).

Other things that were tried: Reset and refresh everything, swap wires and position on the quad, turning off resource intensive features and lowering the Gyro/PID rate. Non of those had any effect. 

Log from BF 4.0.5 with debug mode `rc_smootihng_rate` that demonstrates the issue in flight. 
[Tim_LOG00022.BFL.zip](https://github.com/betaflight/betaflight/files/3852741/Tim_LOG00022.BFL.zip)

```
After factory reset and update to 3.23 TX & RX

Output 1: CRSF TX -> FC
Output 2: CRSF RX <- FC
Output 3: CRSF RX <- EVO
Output 4: CRSF TX -> EVO
Output 5: BST SDA - unconnected 
Output 6: BST SCL - unconnected 

# rc_smoothing_info
# RC Smoothing Type: FILTER
# Detected RX frame rate: 3.455ms
# Input filter type: BIQUAD
# Active input cutoff: 130hz (auto)
# Derivative filter type: BIQUAD
# Active derivative cutoff: 130hz (auto)

# rc_smoothing_info
# RC Smoothing Type: FILTER
# Detected RX frame rate: 4.93ms
# Input filter type: BIQUAD
# Active input cutoff: 110hz (auto)
# Derivative filter type: BIQUAD
# Active derivative cutoff: 110hz (auto)

# rc_smoothing_info
# RC Smoothing Type: FILTER
# Detected RX frame rate: 4.93ms
# Input filter type: BIQUAD
# Active input cutoff: 110hz (auto)
# Derivative filter type: BIQUAD
# Active derivative cutoff: 110hz (auto)

# rc_smoothing_info
# RC Smoothing Type: FILTER
# Detected RX frame rate: 3.834ms
# Input filter type: BIQUAD
# Active input cutoff: 117hz (auto)
# Derivative filter type: BIQUAD
# Active derivative cutoff: 117hz (auto)

[... 233 lines cut ...]
set d_pitch = 46
set p_roll = 50
set d_roll = 42
set p_yaw = 36
set d_min_roll = 24
set d_min_pitch = 26

rateprofile 1

# rateprofile 1
set roll_rc_rate = 158
set pitch_rc_rate = 158
set yaw_rc_rate = 140
set roll_expo = 43
set pitch_expo = 40
set yaw_expo = 10
set roll_srate = 58
set pitch_srate = 50
set yaw_srate = 54

# end the command batch
batch end

# 
```

etracer65, 2019-11-15:
I don't have the equipment to test, but I see something that's probably the cause. Can you try a build from this: https://github.com/etracer65/betaflight/tree/crsf_fix_rcdata_framerate

bizmar, 2019-11-17:
Same diff as above

# rc_smoothing_info
# RC Smoothing Type: FILTER
# Detected RX frame rate: 5.143ms
# Input filter type: BIQUAD
# Active input cutoff: 87hz (auto)
# Derivative filter type: BIQUAD
# Active derivative cutoff: 87hz (auto)

Anytihng else you would like us to try?

etracer65, 2019-11-17:
Well while we work to get hardware to test it's kind of like taking shots in the dark... All I can do is patch "possible" causes without knowing if those situations are actually happening. I pushed another commit to the branch above to try. Make sure you do a refresh before building to ensure you get the additional commit.

bizmar, 2019-11-17:
That wasn't it.

etracer65, 2019-11-17:
Well then I'm out of ideas until I can get some hardware for debugging. Based on the changes I made in that test branch I don't see any way for the problem to happen unless the Crossfire equipment was in fact sending extra RC channel packets outside the 6.67ms refresh cycle.

bizmar, 2019-11-17:
I checked again to make sure we had the right version, 72cf68c.

etracer65, 2019-11-17:
Are you sure you refreshed your local copy of my branch? Otherwise you wouldn't get the latest commit. You wouldn't be able to tell just from compiling.

etracer65, 2019-11-17:
Never mind, I see that the hash is correct.

etracer65, 2019-11-23:
I was able to duplicate this and forwarded the information on to TBS. When the receiver is set to also use CRSF on channels 3/4 then it begins duplicating every channel data frame and that's what's causing the incorrect timing. Actually the timing is correct - it's just that there are duplicate packets. Turning off CRSF on channel 3/4 of the receiver stops the problem, but then you can't connect it to the Unify Evo as desired.

etracer65, 2019-11-27:
TBS has identified and fixed the problem and it will be included in the 3.24 firmware update. If you have access to the alpha firmware builds you can try it out now. I tested and confirmed it fixed the problem.

DutchFPV, 2019-11-27:
Can I ask how this problem manifests itself in flight?  I believe I'm having a similar issue.  In fact I think I might be the 'other issue' that TBS mentions in your email.  I've been back and forth with Mike at TBS for a month + over a similar issue.  I'm experiencing random lag/latency/lack of sync in CF controls.  Its affecting 2 quads, one running BF 3.6 and one running BF 4.0.  It started with the update to BF 3.6 and is worse with 4.0.   I won't go into the full detail in case I need to bump this to a new thread, but I and TBS have more than convinced ourselves that its not my radio or Crossfire module (both can control sims without issue).  The affected drones share no common components besides CF and BF (the FCs are different and running different BF versions) but the issue is affecting both in the exact same way.  Here's a link to a clip showing the behavior.  Is this what you're experiencing or should I start a new thread?
https://youtu.be/jyXIcg_7giI

etracer65, 2019-11-27:
@DutchFPV This problem related to this issue would cause the RC channel smoothing to not work as well as it should, leading to more noise in feedforward. It's not clear from your video what you're trying to show, but it's doubtful that this problem would be visible in fpv flight footage. But feel free to try a nightly BF4.2 build and the prerelease Crossfire firmware to see if it improves your problems.

DutchFPV, 2019-11-28:
Thanks.  Will do.  I appear to be getting the opposite effect.  Like 'over-smoothing' for lack of a better term.  Makes the quad go into this Matrix-style slo-mo.  What you see in the video is me making fast, aggressive stick inputs, and getting slow motion outputs.  Its bizarre.  I'll try the updates and start a new thread if the problem persists.   Appreciate the info.

closed 2020-01-04 by stale[bot]

bizmar, 2020-01-11:
CF 3.24 beta was released today and it fixes the “double” RC packet issue with a unify EVO connected over CRSF.

# rc_smoothing_info
# RC Smoothing Type: FILTER
# Detected RX frame rate: 6.670ms
# Input filter type: BIQUAD
# Active input cutoff: 67hz (auto)
# Derivative filter type: BIQUAD
# Active derivative cutoff: 67hz (auto)

# rc_smoothing_info
# RC Smoothing Type: FILTER
# Detected RX frame rate: 6.670ms
# Input filter type: BIQUAD
# Active input cutoff: 67hz (auto)
# Derivative filter type: BIQUAD
# Active derivative cutoff: 67hz (auto)

# rc_smoothing_info
# RC Smoothing Type: FILTER
# Detected RX frame rate: 6.670ms
# Input filter type: BIQUAD
# Active input cutoff: 67hz (auto)
# Derivative filter type: BIQUAD
# Active derivative cutoff: 67hz (auto)

# rc_smoothing_info
# RC Smoothing Type: FILTER
# Detected RX frame rate: 6.670ms
# Input filter type: BIQUAD
# Active input cutoff: 67hz (auto)
# Derivative filter type: BIQUAD
# Active derivative cutoff: 67hz (auto)

# rc_smoothing_info
# RC Smoothing Type: FILTER
# Detected RX frame rate: 6.670ms
# Input filter type: BIQUAD
# Active input cutoff: 67hz (auto)
# Derivative filter type: BIQUAD
# Active derivative cutoff: 67hz (auto)

# rc_smoothing_info
# RC Smoothing Type: FILTER
# Detected RX frame rate: 6.670ms
# Input filter type: BIQUAD
# Active input cutoff: 67hz (auto)
# Derivative filter type: BIQUAD
# Active derivative cutoff: 67hz (auto)

# rc_smoothing_info
# RC Smoothing Type: FILTER
# Detected RX frame rate: 6.670ms
# Input filter type: BIQUAD
# Active input cutoff: 67hz (auto)
# Derivative filter type: BIQUAD
# Active derivative cutoff: 67hz (auto)

# rc_smoothing_info
# RC Smoothing Type: FILTER
# Detected RX frame rate: 6.670ms
# Input filter type: BIQUAD
# Active input cutoff: 67hz (auto)
# Derivative filter type: BIQUAD
# Active derivative cutoff: 67hz (auto)

# rc_smoothing_info
# RC Smoothing Type: FILTER
# Detected RX frame rate: 6.670ms
# Input filter type: BIQUAD
# Active input cutoff: 67hz (auto)
# Derivative filter type: BIQUAD
# Active derivative cutoff: 67hz (auto)