betaflight/betaflight #9180: Dynamic Idle dont work with 3D ESC setup

3DRacer, 2019-11-16:
Setup is BF 4.1 with BLheli_32  32.7 with RPM filtering.
ESC´s are set on bidirectional soft.
Motors stutter at min throttle stick position, even whwn 3D is switched off  over switch.
Set idle_min_rpm = 14 makes no differenses to idle_min_rpm = 100. 
Dshot_idle_value = 1

etracer65, 2019-11-16:
Not supported for 3D modes.

3DRacer, 2019-11-16:
Hopefully it will be a feature, also in 3D mode.

etracer65, 2019-11-16:
And it should work fine while you're in switched 3D mode (normal throttle range). Just looks like you've used inappropriate settings. You should read the documentation.

https://github.com/betaflight/betaflight/wiki/Tuning-Dynamic-Idle

etracer65, 2019-11-16:
But if you switch into real 3D mode then the behavior will probably be unexpected.

etracer65, 2019-11-16:
I looked a little more at the code. If the 3D feature is enabled then dynamic idle doesn't do anything. The reason your motors were "stuttering" was because you set the DSHOT idle to 1% which is too low.

3DRacer, 2019-11-16:
Thanks for answer. I set dhot_idle only for test so low. I have read the documentation.  But found no evidence that it is not effective in 3D. Again, I hope it will be a feature, available also with 3D setup

etracer65, 2019-11-16:
I updated the wiki page noting that the feature doesn't work with 3D enabled.

3DRacer, 2019-12-03:
Thanks for updating the wiki. But is that the right way? Is there any possibility to enable the feature in 3D also? For some moves, I like to turn off the 3D Mode for a short time. Would be fine, when than the dynamic Idle activated for maximum hangtime. Please make a feature request from this.

etracer65, 2019-12-03:
It would depend on somebody interested in working on it. Currently none of the active dev team are 3D pilots.

3DRacer, 2019-12-03:
I would be happy if I could help. I am not a programmer. But very experienced 3d heli pilot. All my copter, from 2 "to 5" are able to fly 3D. For critical maneuvers in a small space, it is undesirable if the motors change direction. That's why, please.

etracer65, 2019-12-03:
I tagged this as a feature request. But remember that's no guarantee that it will be worked on. It will take somebody interested in implementing this.

3DRacer, 2019-12-03:
Thank you so mutch. It would be great...

KenImhof, 2020-01-23:
@3DRacer, I use the idle up throttle curve to select 3D mode just like a helicopter.  Normal throttle curve is something like 50, 60, 70,80,100 and idle up (3D mode) is; 0, 25, 50, 75, 100.

I am also using bidirectional DSHOT (telemetry), to synchronize motors and increase ramp-up power for reversals, which greatly improves the reversal transition and I believe it is just as good as a collective heli.  Without this update my micro 3D quads would just desync and crash and transitions on my 8-inch 3D quad was very sloppy.  Both are now perfect with the DSHOT telemetry synchronization reversals.

blckmn, 2026-09-26:
We're closing this as part of a clean-up of issues more than two years old. Sorry it sat here so long without a resolution, that's on us. Betaflight has changed a lot since this was opened, so if it's still relevant on the latest release, please open a new issue with current details and link back here.

closed 2026-09-26 by blckmn

3DRacer, 2026-09-28:
That’s a real shame. Unfortunately, I haven’t found a satisfactory solution for it yet, so I still have to keep my copters strictly separate—the fun 3D copters and the standard freestyle copters with GPS. It’s a pity that the two can’t work together. So, things will just have to stay the way they are. Unfortunately.