betaflight/betaflight #9192: Kakute F7 unified target

swampmonster, 2019-11-19:
**Describe the bug**
The unified target for KAKUTEF7 seems broken

**To Reproduce**
Steps to reproduce the behavior:
- Flash either the 4.1.1 or the 4.1.0 unified target for KAKUTEF7

**Expected behavior**
Firmware boots and you're able to configure it but when flashed it doesn't boot up properly and has to be reflashed. Confirmed that the legacy target works but unified does not.

**Flight controller configuration**
N/A

**Setup / Versions**
 - Kakute F7 V1.5 bought directly from holybro

Asizon, 2019-11-19:
F7 based flight controllers need to manually reconnect when flashing is done and red led light

swampmonster, 2019-11-19:
> F7 based flight controllers need to manually reconnect when flashing is done and finish with steady red led light

I'll give that a shot and report back

swampmonster, 2019-11-19:
> > F7 based flight controllers need to manually reconnect when flashing is done and finish with steady red led light
> 
> I'll give that a shot and report back

So i flashed it and then unplugged and replugged the usb and let it sit for a few minutes and eventuellt it came back and registered on the com port. Never did see a red light but it seems to work now. Thank you very much for the help!

swampmonster, 2019-11-19:
Btw, is this documented somewhere (i couldn't find anything) and if it's not perhaps it should be.

mikeller, 2019-11-19:
@swampmonster:

> So i flashed it and then unplugged and replugged the usb and let it sit for a few minutes and eventuellt it came back and registered on the com port

If this takes minutes then this suggests that some component is defective and takes extremely long time to initialise - your board is probably defective.

swampmonster, 2019-11-19:
> @swampmonster:
>
>
> If this takes minutes then this suggests that some component is defective and takes extremely long time to initialise - your board is probably defective.

I went and did something else so it could well have been shorter. I can do a proper test and actually watch it if there's interest.

swampmonster, 2019-11-20:
Closing this since it’s solved and doesn’t need to clog up the tracker. Thanks again for the assistance!

closed 2019-11-20 by swampmonster