betaflight/betaflight #9194: 4.1.1 GPS issue

anakin323, 2019-11-19:
Dear Betaflight dev team,

I got the GPS issue with 4.1.1
It fixed position very fast, but after several minutes, it down to 4 or 5 sats and hold at that for long time, and GPS rescue N/A warning, cannot use GPS resuce. I test on both Kakute F7 and Mamba F722S with Radiolink TS100, it gave the same results
On 3.5.7, it worked without any issues. Hope the dev fix this issue. I love the RPM filter, and i don't wanna roll back to the previous.

Thank you.

https://tinyurl.com/vyy4p3l
https://tinyurl.com/ru2lkrn
https://tinyurl.com/syogktw

etracer65, 2019-11-19:
The Betaflight firmware version does not affect the number of satellites acquired by the GPS module. That's completely under the control of the GPS module and it simply reports the number to Betaflight.

Typically GPS reception is affected by electrical and radio frequency noise generated by the components of the quad like cameras (high def and fpv), vtx, receiver, and even the flight controller. Also greatly affected by the placement of the GPS module and making sure it has a clear view of the sky. The GPS receiver has a small antenna trying to receive an extremely weak signal from satellites that are 20,000km away. Having RF transmitters and interference close by can make this extremely difficult. People can often demonstrate this for themselves on the bench. Try testing the number of satellites received when the quad is fully powered. Then unplug as many components as possible and try to have just the flight controller and GPS module active and see how many more satellites are acquired. Even better if powered by USB instead fo the battery as the on-board power supplies on many flight controllers generate significant RF interference.

This has been discussed before and it always turns out that there's no difference between Betaflight versions when testing in a controlled way. And there's no reason why there should since it's completely under the control of the GPS module.

anakin323, 2019-11-19:
I tested on the bench with FC and the GPS module only, no more another component, the same setup on both 3.5.7 and 4.1.1. 
Here is the video on 4.1.1. https://youtu.be/X9FCHArfFoQ
 GPS sats flickering between 4 and 14. On 3.5.7 it seem to be very solid as attatch file on my post.

etracer65, 2019-11-19:
Can you report what shows on your OSD?

anakin323, 2019-11-19:
> Can you report what shows on your OSD?

This is what i got. https://www.youtube.com/watch?v=3ekHA1oLuB8&feature=youtu.be

etracer65, 2019-11-20:
Please provide the information requested when you opened the issue.

> **Flight controller configuration**
> Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

>Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

etracer65, 2019-11-20:
And while waiting for your information...

I tested 2 different flight controllers with different GPS modules and couldn't replicate this problem. In each case the number of satellites was completely stable and showed no problems. Also we've not seen any other reports like this.

For the satellite count to change that quickly and randomly I would suspect a wiring problem or possible a failing GPS module. The behavior of initially working and then beginning to misbehave after a time period also sounds like a hardware problem with the GPS module acting up as it heats up. Or possibly the 5V source supplying the GPS module is not stable or becomes unstable as it heats up.

As tests I would try:
- Check your wiring. Look for loose plugs, broken or crimped wires, etc.
- Try a different GPS module if you can.
- Try going bad to 3.5.7 and bench test to see if the problem is still happening in the GPS tab.

anakin323, 2019-11-20:
I am at the office now. I will diff after work. 
Roll back to 3.5.7 and have a flight test in this weeked is on my way. 
Thanks for your support.

etracer65, 2019-11-20:
Well since you could demonstrate the problem in the Configurator with the satellite count rapidly changing value you could simply start with a bench test on 3.5.7 to see if the problem is still happening.

anakin323, 2019-11-20:
When i flashed back to 3.5.7 and this weird behavior has gone. Satellites counted very solid around 12-17. I will make a video after work.

etracer65, 2019-11-21:
The only thing I can think of is there was a change in BF4.1 to select a different GPS "profile" as part of the UBLOX protocol that's more appropriate for our use (allows faster initial fix and matches our use case better). I suppose it's possible that your GPS module doesn't properly support the real UBLOX protocol completely. Do you have another GPS module you can try? What GPS module are you using? Again I can't replicate this problem with multiple GPS modules and there have been no other reports like this.

etracer65, 2019-11-21:
You could try testing with BF4.0 and if it doesn't have the problem that would further point to your module not liking the UBLOX profile change for 4.1.

anakin323, 2019-11-21:
**Flight controller configuration**
```# 

# diff

# version
# Betaflight / MAMBAF722 (MBF7) 4.1.0 Oct 16 2019 / 11:42:41 (c37a7c91a) MSP API: 1.42

# start the command batch
batch start

board_name MAMBAF722

# feature
feature -SOFTSERIAL
feature GPS

# map
map TAER1234

# serial
serial 0 0 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 2 8192 115200 57600 0 115200
serial 4 2 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

# aux
aux 0 0 0 900 1200 0 0
aux 1 1 2 1300 1700 0 0
aux 2 13 1 1700 2100 0 0
aux 3 26 0 900 1200 0 0
aux 4 39 4 1800 2100 0 0
aux 5 46 2 1775 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 100 200 400 600

# rxfail
rxfail 0 s 1500
rxfail 1 s 1500
rxfail 2 s 1500
rxfail 3 s 1350
rxfail 6 s 1500

# master
set gyro_lowpass2_hz = 325
set dyn_notch_width_percent = 0
set dyn_notch_q = 230
set dyn_notch_min_hz = 90
set dyn_lpf_gyro_min_hz = 260
set dyn_lpf_gyro_max_hz = 650
set acc_calibration = -123,-12,405
set rssi_channel = 8
set serialrx_provider = CRSF
set blackbox_device = NONE
set dshot_idle_value = 400
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set failsafe_switch_mode = STAGE2
set failsafe_procedure = GPS-RESCUE
set current_meter = ESC
set small_angle = 180
set gps_provider = UBLOX
set gps_rescue_angle = 45
set gps_rescue_initial_alt = 35
set gps_rescue_descent_dist = 50
set gps_rescue_ground_speed = 1000
set gps_rescue_min_sats = 6
set gps_rescue_min_dth = 70
set gps_rescue_allow_arming_without_fix = ON
set deadband = 3
set yaw_deadband = 3
set pid_process_denom = 2
set osd_rssi_alarm = 85
set osd_cap_alarm = 700
set osd_alt_alarm = 50
set osd_vbat_pos = 2305
set osd_rssi_pos = 2049
set osd_throttle_pos = 2337
set osd_current_pos = 34
set osd_mah_drawn_pos = 311
set osd_gps_speed_pos = 2328
set osd_gps_lon_pos = 2544
set osd_gps_lat_pos = 2529
set osd_gps_sats_pos = 2073
set osd_home_dir_pos = 2060
set osd_home_dist_pos = 2068
set osd_altitude_pos = 2358
set osd_warnings_pos = 2442
set osd_stat_max_dist = ON
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 1
set vtx_power = 5
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5658
set gyro_1_sensor_align = CW0FLIP
set gyro_1_align_pitch = 1800
set gyro_1_align_yaw = 0

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 98
set dyn_lpf_dterm_max_hz = 238
set dterm_lowpass2_hz = 210
set vbat_pid_gain = ON
set anti_gravity_gain = 10000
set iterm_relax_cutoff = 10
set p_pitch = 63
set i_pitch = 80
set d_pitch = 27
set f_pitch = 100
set p_roll = 55
set i_roll = 73
set d_roll = 28
set f_roll = 85
set p_yaw = 35
set i_yaw = 80
set d_min_roll = 0
set d_min_pitch = 0
set d_min_boost_gain = 32

rateprofile 0

# rateprofile 0
set roll_rc_rate = 120
set pitch_rc_rate = 120
set yaw_rc_rate = 175
set roll_srate = 72
set pitch_srate = 69
set yaw_srate = 40
set tpa_breakpoint = 1450
set tpa_mode = PD

# end the command batch
batch end

# ```

```A00: FREE
A01: FREE
A02: LED 1
A03: FREE
A04: SDCARD_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: FREE
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FREE
B04: FREE
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: SERIAL_TX 3
B11: SERIAL_RX 3
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: ADC_CURR
C03: ADC_BATT
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: SDCARD_DETECT
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: BEEPER
E00: FREE
E01: GYRO_EXTI
E02: SPI_SCK 4
E03: FREE
E04: GYRO_CS 1
E05: SPI_MISO 4
E06: SPI_MOSI 4
E07: FREE
E08: FREE
E09: MOTOR 3
E10: FREE
E11: MOTOR 4
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: MOTOR 3
    CH2: MOTOR 4
TIM2: FREE
TIM3:
    CH3: MOTOR 1
    CH4: MOTOR 2
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE```

anakin323, 2019-11-21:
> You could try testing with BF4.0 and if it doesn't have the problem that would further point to your module not liking the UBLOX profile change for 4.1.

1. 
This is what i test with 3.5.7 https://www.youtube.com/watch?v=5WvrYiasMik&feature=youtu.be

2.
4.0.6 https://www.youtube.com/watch?v=unQWXv7Tat0&feature=youtu.be 

3.
4.1.1 https://www.youtube.com/watch?v=UfIhrnxe0Vs&feature=youtu.be

etracer65, 2019-11-21:
@TonyBlit Any thoughts on this? I'm suspecting that the GPS module (Radiolink TS100) used is not happy with the change to use pedestrian mode for the initial fix and then switch to airborne mode introduced in #8758. Problem only seems to happen with 4.1. Note the videos a couple of comments in that show the satellite count rapidly oscillating from low to normal counts.

TonyBlit, 2019-11-21:
It's quite weird.. I've checked and Radiolink TS100 also uses the same M8030 module as all the Beitian family, so maybe that particular module has an old firmware? I don't have a TS100 in order to test what is going on, and there is no easy way to log the ublox messages in the current code (I had to hack something on my own while developing to check what the module was answering, 
and in what order). I'd be happy to help but not sure how.

anakin323, 2019-11-22:
> It's quite weird.. I've checked and Radiolink TS100 also uses the same M8030 module as all the Beitian family, so maybe that particular module has an old firmware? I don't have a TS100 in order to [...]

My radiolink ts100 works great, but not suitable with 4.1.x. Could you give me some advice or i need to change to beitian family.

TonyBlit, 2019-11-22:
For the time being, switching to beitian (or similar) would be your best bet. If a lot of other people complains about having this issue on other devices, I guess the pedestrian-airborne switch will have to be optionally disabled in a future version of BF.

anakin323, 2019-11-22:
> For the time being, switching to beitian (or similar) would be your best bet. If a lot of other people complains about having this issue on other devices, I guess the pedestrian-airborne switch will have to be optionally disabled in a future version of BF.

I have borrow my friend a BN-880 and test with BF 4.1.1, the same result, only 4-6 sats counted.
Here is the video: https://www.youtube.com/watch?v=XOW4PEY645Y&feature=youtu.be

etracer65, 2019-11-22:
To be clear, we're not discussing or addressing "how many" satellites you get. Only the behavior you documented where the satellite count was rapidly alternating between a high value and low value. If this other GPS is not doing that then it confirms that the other module has some issue with the pedestrian operational mode.

TonyBlit, 2019-11-22:
It's so strange that ucenter shows so many sats while BF reports only 4-6. Does this happen every time you plug the battery? I mean, have you tried 4 or 5 times completely unplugging the battery each time and got the same 4-6 sats? Can you repeat the same exact test with BN-880 and BF 4.0.x instead of 4.1.x ?

TonyBlit, 2019-11-22:
Please, disregard my previous email. I've done a quick test looking at the BF configurator with a BN-880 in my city (super bad reception). The list of sats is quite big, maybe 12 or so, but sat count is about 6 because the reception is so bad. So your last video is totally fine, it just says you're having super bad reception for whatever reason.

To be sure you're getting a good amount of sats, please:
- Go to a place where you can see ALL the sky (no mountains, no big trees, no buildings, etc)
- Orient the GPS so the ceramic part has the most sky visibility possible, ie: at the very top of the quad.
- The less cloudy the sky, the better.

anakin323, 2019-11-22:
> Please, disregard my previous email. I've done a quick test looking at the BF configurator with a BN-880 in my city (super bad reception). The list of sats is quite big, maybe 12 or so, but sat [...]

I am going to have a flight with bn880 this morning. Keep you posted.

anakin323, 2019-11-23:
> Please, disregard my previous email. I've done a quick test looking at the BF configurator with a BN-880 in my city (super bad reception). The list of sats is quite big, maybe 12 or so, but sat [...]
After plug in, it fixes very fast to 16
It still change from high number of sats to low, you can check at 3:22. It changes from 7 to 18 and back to 5 and stable at 7 till the end.

TonyBlit, 2019-11-23:
Which BF version have you used for this last test, 4.1.x? Can you do the exact same test with exact same hardware in the exact same location but with BF 4.0.x?

anakin323, 2019-11-23:
> Which BF version have you used for this last test, 4.1.x? Can you do the exact same test with exact same hardware in the exact same location but with BF 4.0.x?

4.1.0 https://youtu.be/DKbH1o4NmIU
4.0.6

anakin323, 2019-11-23:
@TonyBlit 
This is the DVR of 4.0.6. https://www.youtube.com/watch?v=pI_gF2-2A8M&feature=youtu.be
it happened at 1:29 ; 3:47 4:23

TonyBlit, 2019-11-23:
Thanks for doing these tests, they prove your issue is not related to a change introduced in 4.1.x (as it also happens on 4.0.x), so unfortunately I'm out of ideas of what's going on in your case. I have experienced these kind of "blackouts" very very rarely and I was told they were due to solar storms, but I highly doubt you're experiencing this (current Kp index is low as of today). You're flying in a big city, so maybe you're experiencing some kind of interference. I'd try flying in a non-urban area to see if that changes.

anakin323, 2019-11-24:
> Thanks for doing these tests, they prove your issue is not related to a change introduced in 4.1.x (as it also happens on 4.0.x), so unfortunately I'm out of ideas of what's going on in your case. I [...]

I have checked the ROM firmware of my BN880, it is the latest 3.01. 
The BN-220 has just arrived, i will do a test soon

 https://youtu.be/j7e7RciVhkY
This guy test gps rescue with 4.1.0, the satellites change from high to 7 but stable at 7.

etracer65, 2019-11-24:
Try testing on the bench with as few components active as possible. Like only with the flight controller and GPS powered. Other components emitting RF noise can affect the GPS receiver and prevent it from receiving satellite. This is particularly true if you have receiver or VTX antennas close to the module. But even things like FPV cameras or GoPros can also emit RF interference. The hybrid FPV/HD cameras are particularly bad with ones like the Caddx Tarsier or Runcam split making GPS reception extremely poor. Also there have been cases where the on-board voltage regulation on flight controllers have created particularly bad interference. The symptom there is that if powered by USB then everything works well but when running off the battery you get fewer satellites because the on-board voltage regulators are generating RF noise.

TonyBlit, 2019-11-25:
Yesterday I was flying with some friends, we had a mixture of BF versions and GPS Modules. At some point of the day, a few of the quads started to get trouble locking sats, and we even experienced this high to low sat count bouncing. As I wasn't expecting this, I didn't paid enough attention so I could find a pattern, but I just wanted to share that there is something going wrong, not sure what.

anakin323, 2019-11-25:
> Try testing on the bench with as few components active as possible. Like only with the flight controller and GPS powered. Other components emitting RF noise can affect the GPS receiver and prevent [...]

When i do a test on the betaflight configurator, i only power the fc and GPS module by USB, so the interference from VTX or Crossfire module is zero.
My friend reported that his Matek M8Q satellites count is stable at 12-15 with 4.1.0, at the same time, same location.

@TonyBlit 
I have test my hardware with inav software, the same FC and GPS module, in inav it is very stable, 16-18 all the time, no lose, at the same time, same location at my home.

mikeller, 2019-11-25:
@TonyBlit: What you describe sounds a lot like interference from a nearby RF source - GPS signals are weak, so it does not take a lot to block them out with something that emits in the vicinity of the receiver. Even dedicated GPS jammers are readily available these days: https://www.jammer-store.com/gp5000-car-use-gps-jammer-blocker.html

TonyBlit, 2019-11-25:
We were literally in the middle of nowhere when this happened, with no houses or electrical poles at sight. But anyway, my point is the behaviour matched what anakin was reporting, so it may not be just a coincidence. However, we have very little data, and sometimes contradictory so we can't draw any conclusion yet. I'll do further testing next weekend.

vlstep3, 2019-12-01:
How Did you turn on I2C on mamba f722?

anakin323, 2019-12-02:
> anakin323
> How Did you turn on I2C on mamba f722?

i don't use I2C, just UART for GPS.

TorbanDallas, 2019-12-19:
I’m having exactly the same issue with all of my quads with Bf4.1.1. I have tested different quads with different BN 180 GPS and 880 Gps Modules. I’m always losing sats. The sats are dropping from around 16 down to 4 or 5 and I’m getting the rescue Mode N/A massage shown in my osd.
Even on a quad that was running without issues on BF 3.5.7 now with 4.1.1 this issue occurs. 
I never had a solid GPS signal after around 30 flights with three different quads on BF 4.1.1

TonyBlit, 2019-12-19:
I haven't been able to fly in a few weeks, so I sadly don't have any additional info. However, Torban's feedback seems solid enough to confirm the previous reports.
I'll do some tests with my debug code enabled to see if I can reproduce the issue and hopefully find anything in the code causing it. Otherwise, I'll be sending a PR to revert the pedestrian change.

chichoamado, 2019-12-20:
Hello guys, i´m from Argentina. And i have the same problem and i think was just for me, but i realized not.
With the same gear, same field to fly, BF 3.5 get solid 12 to 15 sats. With BF 4.1 get little less, 11, but in fligth, drop to 5 to 7. 
I guess this hapen when i install rpm filter. ¿Can be? i going tto check if disabling rpm filter could be the problem. by the way, i have an arris x200 with the GPS beitian BN 220, is an F4, and the GPS mounted in the front rigth arm. but with BF 3,5 works fine.

TorbanDallas, 2019-12-20:
regarding to the feedback I have got in the comments of my video ( https://www.youtube.com/watch?v=57y1dUwL-uw ) apparently, are lot of pilots did experience this Issue with BF4.1

chichoamado, 2019-12-20:
I'm using jesc license for the rpm because my esc support blheli_s.But if there changes in the gps mode, maybe there was the issue too.Hope they can fix it.
-------- Mensaje original --------De: Torban Dallas <notifications@github.com> Fecha: 20/12/19  08:06  (GMT-03:00) A: betaflight/betaflight <betaflight@noreply.github.com> CC: chichoamado <chichinbobot@gmail.com>, Comment <comment@noreply.github.com> Asunto: Re: [betaflight/betaflight] 4.1.1 GPS issue (#9194) regarding to the feedback I have got in the comments of my video ( https://www.youtube.com/watch?v=57y1dUwL-uw ) apparently, are lot of pilots did experience this Issue with BF4.1

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/betaflight/betaflight/issues/9194?email_source=notifications\u0026email_token=ANPE5K4WY65WPKF2M5WHWUTQZSRLJA5CNFSM4JPDGEP2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEHMUPPY#issuecomment-567887807",
"url": "https://github.com/betaflight/betaflight/issues/9194?email_source=notifications\u0026email_token=ANPE5K4WY65WPKF2M5WHWUTQZSRLJA5CNFSM4JPDGEP2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEHMUPPY#issuecomment-567887807",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

TonyBlit, 2019-12-21:
@chichoamado this does not have anything to do with rpm filter, only with a change introduced in BF 4.1
I've seen a bug in the code that could potentially trigger an issue like this, but I need to do several flights to confirm. Today I have only been able to bench test, and haven't observed any dip.

TorbanDallas, 2019-12-21:
@TonyBlit I did another test flight today, I have changed the GPS Rescue Min Sats from 8 to 6 and also gps_rescue_use_mag from ON (default) to OFF ... but I have lost satellites again ... hopefully, you can fix the issue :)

TonyBlit, 2019-12-21:
The change in BF 4.1 is unrelated to any other feature, so fiddling with options is not going to help, unfortunately.
I've added yet more code to debug this issue tomorrow on the field, let's see how it goes. :)

chichoamado, 2019-12-21:
I'm using min_sat in 5 for rescue mode, and arming_without_fix ON

GiulianoDeMicheli, 2019-12-22:
Me too. Solid 14/15 satscon arm. 

> I’m having exactly the same issue with all of my quads with Bf4.1.1. I have tested different quads with different BN 180 GPS and 880 Gps Modules. I’m always losing sats. The sats are dropping from [...]

TonyBlit, 2019-12-23:
I did a bunch of flights yesterday and:
- I can confirm this problem exists in 4.1.x, but it's not clear how to exactly reproduce it. I have seen it when flying (not in bench so far), and not in every flight.
- This problem seems to be due to an improvement I introduced in 4.1 to allow faster sat lock
- The code for this improvement had a small mistake, but despite my several flights with debugging info, I haven't been able to prove this mistake is causing the sat count dip, or if the improvement in itself is causing it. It's so hard to debug a GPS when flying :-)
- The only thing I know at this point is this improvement is causing trouble than it's worth, and I don't have a clear understanding of the exact cause, so I will revert my improvement and as soon as it gets merged (hopefully in 4.1.3) we should get back to the behaviour of 4.0.x or 3.5.x.
- I'll keep working on safer ways to reduce the sat lock time

chichoamado, 2019-12-23:
You are awesome. Thanks for your time. I can wait a minute to the first fix if layer can use return home safely. Really, i appreciatte  ery much your commitment and the all other developers.

AirBreak69, 2019-12-23:
I became aware of this thread lately and was surprised of the code change to "pedestrian model" which IMO does not fit our application.
The high accelerations need a higher dynamic model. But we have to accept the drawbacks of higher TTFFs and less GPS signal stability.

Some thoughts about the dynamic behaviour of (ublox) GPS receivers...

What makes GPS problems hard to catch:
- GPS calculation is NOT a precise and deterministic algorithm. Each satellite constellation, multipath and changing partial reception is handled by more or less optimal strategies.
- GPS recievers are full of magic tricks, heuristical approaches, prediction and application specific options.
- Normally the patch antennas in the small receivers would need a proper "ground plane" (metal plate of at least 10cm diameter) to work optimal. Quads do not have this.
- Electro-magnetic interferences from motors, ESCs, CPU and telemetry do not help the receiver in his delicate job.

TTFF:
- After a "cold start" (buffer battery is empty and the satellite course descriptions are invalid) it takes approx. 12.5min after detecting the first (GPS-) satellite until the "almanac" is build.
- Somewhere inside this 13min+ intervall the receiver has identified enough satellites (4-5) to calculate a first valid position.
- So a final number of "satellites in view" can take up to 20min!
- On a restart within the same geographic area, low time gap from the last start and valid "almanac" memory (eg. "warmstart") the "TTFF" may be some seconds.

Satellite usage:
- If not changed by specific commands ("Main Talker ID") the receiver focusses on GPS satellite system first. GLONASS, Galileo etc. are only used for improvement of the solution, if a FIX could be reached with GPS satellites before.
- Speeds and accelerations higher than the selected model is designed for may force the receiver to take satellites out of the calculation as they do not fit to the predicted course. It starts this by ommiting the improving satellite systems.
- So a drop to 5-7 sats may indicate that the receiver had to reduce to GPS only due to fitting problems by too fast location or constellation changes.
- The displayed number of sats are the "used sats". The number of visible/recepted sats may be higher.

Computational reserves:
The "NAV rate" (data output frequency) limits the computation time and the computational accuracy/stability. Some typical data for the ublox M8 generation from my experience:
- 1Hz will work in any constellation, but is not really suitable for autonomous navigation. RTH would be very steppy.
- 5Hz should work with up to 3 active satellite systems. A good compromise for RTH. This is setup currently.
- 10Hz may be stable for a GPS only setup.
Probably we have to reduce to 3Hz in the future because the increasing number of satellites may overload the receiver. In addition it is limited to 32 tracking channels.
The M8 generation receivers come their limits nowadays.

Dynamics:
"Faster" dynamic models give lower accuracy and stability but generally have to fit the application (dynamics of flying). That are roughly: 3D movement, 5G peak but 1G average accelerations, 10...140km/h.
- The "pedestrian dynamic model" assumes slow speeds and low accelerations in 2D movement. It collects and filters the satellite tracking over several seconds.
- The "Airborne 1G dynamic model" assumes medium speeds and accelerations in 3D movement.
- The "Airborne 4G dynamic model" allows high speeds and accelerations. TTFF may be high and calculation quality lower. Not 100% flawless.

The code of the last year used the highest dynamic but least stable "Airborne 4G". There is a intermediate 2G model available. I privately used it before. It seemed to be a bit more stable.
I had problems with "pedestrian" in an iNAV setup some years ago (overshooting RTH on raised RTH speeds). Low dynamic modes introduce notable latencies.

A better approach to handle the TTFF vs. dynamics dilemma would probably be:
- Boot up in "pedestrian"
- Switch to "Airborne 2G" just before lift off.

mikeller, 2019-12-23:
Thanks @AirBreak69, very good writeup!

One addendum is that the battery supporting the state memory is under dimensioned (or just plain crap) in many cheap Chinese modules, so cold starts seem to be a common occurrence even after being powered down for only a short while.

TonyBlit, 2019-12-23:
@AirBreak69 the approach of booting in pedestrian and switching to Airborne (4G in this case) is EXACTLY what I introduced in this PR https://github.com/betaflight/betaflight/pull/8758 which made into 4.1.x. As you say, it's very very difficult to understand the sat count, but on a big scale (with lots of users testing) you can start to see differences. The same way there was a general consensus that in pedestrian mode the TTFF was lower, now it seems that the dynamic approach I introduced is causing some weird behaviour.
Last but not least, ublox has a way to download some helper data that improves the TTFF dramatically, but it's only available for selected vendors/projects. I tried to contact them but got no answer.

chichoamado, 2019-12-26:
Thanks @AirBreak69, your explanation it´s great.
But despite all the dificult that supose make work a GPS in a quad, working great in BF 3.5.7, and not so good in 4.1
I have a quad, BF 3.5.7 and install a Beitian bn220 on it to use rescue mode, and works great. When update to 4.1, get the "rescue n/a" message in the osd, so it´s make me fly without confidence. So why i put a gps in my quad if i can´t use it?
by the way, in BF 3.5 the time to fix 7 or 8 sats was less than a minute the first fligth of the day, because when i cahnge batterys, the adquisition was nearly instantaneously.
Now, with BF 4.1 i can wait 3 minutes and have 5 sats with lucky. So neither is improving the fix time.

TorbanDallas, 2020-01-05:
Tested Betaflight 4.1.3 #21 Maintenance today - same Issue. 
https://youtu.be/CuT7UecYvoc
GPS Issue occurs on:  02:05 , 02:22 , 03:52 , 06:30

chichoamado, 2020-01-05:
Since last time i post here, ia change the location of the gps from the rigth front arm to the rigth rear arm, and because it's little more far from the go pro i think it's better. And yes. Got 10 sats and have  o issues in fligth. But i don't have the 12 or 16 sats that i have on BF 3.5.

TonyBlit, 2020-01-06:
@TorbanDallas can you describe how you have built or where you have got the 4.1.3 build to be sure it contains the fix?

TorbanDallas, 2020-01-06:
@TonyBlit I have selected the Maintenance Version as you can see at the beginning of my video.  ( https://youtu.be/CuT7UecYvoc )

TonyBlit, 2020-01-06:
Thanks, my revert was merged on 24 dec, and I see the build you used is from 26 dec so it should be there. I don't have a solid way of reproducing the issue, so I have to admit I'm going a bit blind here, and it's really bad news the revert hasn't fixed the issue.
My next step then will be creating a 4.1 build with the exact same GPS code from 4.0, or even from 3.5, just to be sure this is not coming from anywhere else. But I won't be committing anything else to the maintenance branch until I find the root cause and I can be sure it's completely fixed (which means a lot of testing). Anyone willing to help with the testing just let me know. :)

TorbanDallas, 2020-01-06:
@TonyBlit I'm willing to go on testing. The Code from BF 3.5.7 was fine, earlier version had issues with the sanity check. But maybe (hopefully) your changes are not affecting the Rescue Mode

TonyBlit, 2020-01-06:
@TorbanDallas let's switch to email to discuss the testing: tony.cabello at gmail

oliverherdener, 2020-01-06:
I am happy to test as well. I have 2 quads with the same issue. One with a BN180 and one with a BN220.

anakin323, 2020-01-06:
I am building a new quad with kakute f7 v1.5 and BN180. I will join in testing too.

[130 more replies]

closed 2024-05-12 by haslinghuis