betaflight/betaflight #9195: mag_declination: out of range error (CLI)

FMarcelino, 2019-11-19:
The mag_declination on the area I leave is -21.12. When I set this value on BF configurator 10.6.0, it accept it normally, however on CLI, it shows an out-of-range error. The value on CLI is -21120, what make the system understand I am out of -18000 and +18000. Then I decided to edit this value on CLI interface, changing it to -02112. Now there is no error msg anymore, however, back to BFconfig, the value displayed is -2.11.

The GPS is working normally, indicating precisely my location as well as the "arrow" indicating home position on the OSD screen. 

I am not sure if it is just a wrong indication at BFconfig interface. I did not try a GPS Rescue because I don´t know what is the real problem...

GPS module BN-220
FC Matek F722mini
Firmware Flashed 4.1.0
Target MATEKF722MINI (MTKS)
No BARO module installed
No Compass module installed

In other FC (NOX-F4), the mag_declination setting is not accepted at all. I can set any value via BFconfig 10.6.0 and save. After reboot the board, the value returns to zero...

CLI:
set align_mag = DEFAULT
set mag_align_roll = 0
set mag_align_pitch = 0
set mag_align_yaw = 0
set mag_bustype = I2C
set mag_i2c_device = 1
set mag_i2c_address = 0
set mag_spi_device = 0
set mag_hardware = NONE
´´´set mag_declination = -21620
###ERROR: CORRUPTED CONFIG: mag_declination = -21620###```
set mag_calibration = 0,0,0

Thanks.

etracer65, 2019-11-19:
> GPS module BN-220
> FC Matek F722mini
> Firmware Flashed 4.1.0
> Target MATEKF722MINI (MTKS)
> No BARO module installed
> **No Compass module installed**

If you don't have a compass (magnetometer) connected then why are you trying to set the declination? That only applies to the compass and won't do anything otherwise.

> In other FC (NOX-F4), the mag_declination setting is not accepted at all. I can set any value via BFconfig 10.6.0 and save. After reboot the board, the value returns to zero...

You're likely using the "legacy target" instead of the newer unified target - "NOX (AIRB)". The legacy target does not have support for a compass included. But again, unless you actually have a magnetometer connected it's irrelevant.

FMarcelino, 2019-11-19:
Thanks for the explanation. I thought the mag_declination was also needed by GPS Rescue function...

closed 2019-11-19 by FMarcelino

FMarcelino, 2019-11-22:
Bruce/Support Team,

Regarding to FC-NOX, mine was flashed with FW 4.1.0.
CLI version command returns:
# Betaflight / NOX (NOX1) 4.1.0 Oct 16 2019 / 11:45:54 (c37a7c91a) MSP API: 1.42

My hardware was bought from Banggood and is a NOXe-F4.
It does a have a BARO in it and it is working (so far).
Should I really reflash it with NOX(AIRB)?
Below is the board picture, just in case....
[cid:835221af-a1cb-49f7-ba64-60cb301b55f9]

Many thanks in advance

FMarcelino, 2019-11-22:
CLI version command returns: NOX(NOX1) as board identity
________________________________
From: Fabio Marcelino <bbxxya@hotmail.com>
Sent: Thursday, November 21, 2019 10:58 PM
To: betaflight/betaflight <reply@reply.github.com>
Subject: Re: [betaflight/betaflight] mag_declination: out of range error (CLI) (#9195)

Bruce/Support Team,

Regarding to FC-NOX, mine was flashed with FW 4.1.0.
CLI version command returns:
# Betaflight / NOX (NOX1) 4.1.0 Oct 16 2019 / 11:45:54 (c37a7c91a) MSP API: 1.42

My hardware was bought from Banggood and is a NOXe-F4.
It does a have a BARO in it and it is working (so far).
Should I really reflash it with NOX(AIRB)?
Below is the board picture, just in case....
[cid:835221af-a1cb-49f7-ba64-60cb301b55f9]

Many thanks in advance