betaflight/betaflight #9199: smart audio communication issue on betaflight 4.1.1

Martian180, 2019-11-19:
**Describe the bug**
After updating to betaflight 4.1.1 could not change channels and bands via smart audio. The power selection works, and pit mode can be entered and left but the vtx doesn't change channnel.
A manual vtx table setup or a standard one does not matter.

 **To Reproduce**
Setup which worked fine on 4.1.0 and offers problems described above. The diff-all was exactly the same on 4.1.1. as on 4.1.0 

**Expected behavior**
The smart audio menu shows ---- in stead of the channel and band. Power selection is displayed fine. When changing bands and channels all of the channels made in the vtx table show but the vtx does not change when you select "set" in the menu.

**Flight controller configuration**
'''diff all

# version
# Betaflight / OMNIBUSF4SD (OBSD) 4.1.0 Oct 16 2019 / 11:48:35 (c37a7c91a) MSP API: 1.42

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name OMNIBUSF4SD
mcu_id 00270057424d500d20383654
signature 

# name: MEGABEE v2

# feature
feature -RX_PARALLEL_PWM
feature -AIRMODE
feature RX_SERIAL
feature TELEMETRY

# beeper
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 2 8192 115200 57600 0 115200

# aux
aux 0 0 0 1800 2100 0 0
aux 1 1 2 900 1200 0 0
aux 2 2 2 1300 1700 0 0
aux 3 13 1 1800 2100 0 0
aux 4 28 2 1800 2100 0 0
aux 5 35 4 1800 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860    0
vtxtable band 4 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 5 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 4
vtxtable powervalues 25 100 200 500
vtxtable powerlabels 25 100 200 500

# master
set gyro_sync_denom = 2
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 90
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = CRSF
set dshot_idle_value = 450
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set motor_poles = 12
set bat_capacity = 1550
set vbat_warning_cell_voltage = 340
set small_angle = 180
set pid_process_denom = 1
set osd_vbat_pos = 2497
set osd_rssi_pos = 2074
set osd_link_quality_pos = 2049
set osd_tim_2_pos = 2517
set osd_vtx_channel_pos = 2484
set osd_warnings_pos = 14792
set osd_avg_cell_voltage_pos = 2465
set osd_stat_max_spd = OFF
set osd_stat_min_batt = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_min_rssi_dbm = ON
set vtx_band = 4
set vtx_channel = 7
set vtx_power = 1
set vtx_freq = 5880
set name = MEGABEE v2

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# '''

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

'''# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 3
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A15: OSD_CS
B00: MOTOR 1
B01: MOTOR 2
B03: PREINIT 2
B04: BEEPER
B05: LED 1
B06: FREE
B07: SDCARD_DETECT
B08: FREE
B09: FREE
B10: SERIAL_TX 3
B11: FREE
B12: SDCARD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: USB_DETECT
C06: FREE
C07: FREE
C08: INVERTER 6
C09: INVERTER 3
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
D02: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
    CH3: DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: ADC 2
DMA2 Stream 4: DSHOT_BITBANG 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
'''
Write your command here. Press Tab for AutoComplete.

**Setup / Versions**
Iflight succex stack with Iflight vtx
Iflight megabee v2 bought from banggood, it was a bind and fly so it's build by Iflight specs.

**Additional context**
When using the exact same setup everything works great on betaflight 4.1.0

etracer65, 2019-11-20:
You have your configuration set for IRC Tramp protocol - not SmartAudio. SmartAudio is different than IRC Tramp so determine what is required by your particular VTX and configure the UART and the `vtxtable` appropriately. Also in the OSD be sure to use the correct menu. VTX SA` is for SmartAudio and `VTX TR` is for Tramp.

Martian180, 2019-11-20:
It's because my lack of technical English I guess,  I know it's IRC tramp protocol,  so is my vtx. With these settings it works fine on beta 4.1.0 , just not on 4.1.1

Outlook voor Android downloaden<https://aka.ms/ghei36>

bonkas, 2019-11-21:
I am experiencing a similar issue.

I have the Rush Tank Ultimate VTX

Set protocol to TBS SmartAudio as required.

I can adjust the Channel via stick input or LUA Script but the power level is unable to be changed, it is stuck on 25mw.

Here is diff all of my config:

`# version
# Betaflight / STM32F7X2 (S7X2) 4.1.0 Oct 16 2019 / 11:58:45 (c37a7c91a) MSP API: 1.42
# manufacturer_id: DIAT   board_name: MAMBAF722   custom defaults: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MAMBAF722
manufacturer_id DIAT
mcu_id 0033002f3237510731313935
signature 

# name: Karearea

# feature
feature -RX_PARALLEL_PWM
feature GPS

# serial
serial 0 0 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 3 0 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 2 1 1300 1700 0 0
aux 3 13 2 1600 2100 0 0
aux 4 35 4 1700 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5999
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740 5584 5621
vtxtable powerlevels 4
vtxtable powervalues 14 23 27 29
vtxtable powerlabels 25 200 500 800

# master
set serialrx_provider = CRSF
set dshot_idle_value = 300
set dshot_bidir = ON
set small_angle = 180
set gps_provider = UBLOX
set pid_process_denom = 2
set osd_vbat_pos = 2497
set osd_rssi_pos = 6201
set osd_link_quality_pos = 6233
set osd_tim_1_pos = 6582
set osd_flymode_pos = 6573
set osd_vtx_channel_pos = 6177
set osd_current_pos = 6561
set osd_mah_drawn_pos = 6616
set osd_craft_name_pos = 6603
set osd_gps_speed_pos = 4193
set osd_gps_sats_pos = 4161
set osd_stat_min_link_quality = ON
set vtx_band = 1
set vtx_channel = 8
set vtx_power = 4
set vtx_low_power_disarm = ON
set vtx_freq = 5725
set gyro_1_align_yaw = 1800
set name = Karearea

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save`

etracer65, 2019-11-21:
@bonkas You have a configuration problem. That VTX is a SA2.0 device and you've used the wrong `vtxtable` settings. Read the documentation here: https://github.com/betaflight/betaflight/wiki/VTX-tables

Also you have `vtx_low_power_disarm` enabled so the power will not change while disarmed.

bonkas, 2019-11-21:
Thank you very much for your reply @etracer65 

I thought that might have been the case, I did try the settings for SA 2.0 and 2.1 as I wasn't 100% sure because neither were working and since it was changing channels I didn't think it was the tables. 

Watching videos online all used the power values instead of 1,2,3,4 etc which lead me down that path which was wrong, but I did try both.

I spent some time on this after I posted and I actually found the problem, I had some (I guess) unsupported frequencies in the table, once I removed these everything started to work.

mikeller, 2019-11-23:
@bonkas: Do you mind posting the output of CLI `diff` of what is now working for you? It would be interesting to see what frequencies were not supported by your hardware.

bonkas, 2019-12-11:
Sorry for the late reply. Here is a diff from my current working setup:

# version
# Betaflight / STM32F7X2 (S7X2) 4.1.1 Nov 15 2019 / 12:57:05 (1e5e3d369) MSP API: 1.42
# manufacturer_id: DIAT   board_name: MAMBAF722   custom defaults: YES

# start the command batch
batch start

board_name MAMBAF722
manufacturer_id DIAT

# name: Karearea

# feature
feature -RX_PARALLEL_PWM
feature GPS

# serial
serial 0 0 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 3 0 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 2 1 1300 1700 0 0
aux 3 46 3 1600 2100 0 0
aux 4 13 2 1600 2100 0 0
aux 5 13 3 1600 2100 0 0
aux 6 35 4 1700 2100 0 0

# adjrange
adjrange 0 0 3 900 1300 29 3 0 0
adjrange 1 0 3 1300 1700 29 3 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 500 800

# rxfail
rxfail 5 s 1500

# master
set acc_calibration = -80,72,-505
set min_check = 1005
set serialrx_provider = CRSF
set dshot_idle_value = 300
set dshot_bidir = ON
set failsafe_procedure = GPS-RESCUE
set small_angle = 180
set gps_provider = UBLOX
set gps_rescue_angle = 36
set gps_rescue_ground_speed = 1500
set gps_rescue_throttle_min = 1200
set gps_rescue_min_sats = 6
set gps_rescue_allow_arming_without_fix = ON
set pid_process_denom = 2
set osd_vbat_pos = 14785
set osd_rssi_pos = 14393
set osd_link_quality_pos = 14425
set osd_tim_1_pos = 14774
set osd_flymode_pos = 14765
set osd_vtx_channel_pos = 14369
set osd_current_pos = 14753
set osd_mah_drawn_pos = 14808
set osd_craft_name_pos = 14795
set osd_gps_speed_pos = 12385
set osd_gps_lon_pos = 12363
set osd_gps_lat_pos = 12331
set osd_gps_sats_pos = 12353
set osd_home_dir_pos = 12417
set osd_home_dist_pos = 12418
set osd_altitude_pos = 14721
set osd_stat_min_link_quality = ON
set osd_profile = 2
set vtx_band = 1
set vtx_channel = 8
set vtx_power = 4
set vtx_low_power_disarm = ON
set vtx_freq = 5725
set gyro_1_align_yaw = 1800
set name = Karearea

profile 0

rateprofile 0

# end the command batch
batch end

closed 2019-12-11 by Martian180