betaflight/betaflight #9200: BF4.2.0 - arming only with activated accelerometer

TheIsotopes, 2019-11-19:
**Describe the bug**
since `build #1963` i can not arming with deactivated accelerometer ... on bfc arming disable flags: Angle is displayed ... after enabling accelerometer the angle flag disappeared and i can now arming

imo the bug seems to exist since `build #1960`

**Flight controller configuration**
```
# Betaflight / SPRACINGH7NANO (SP7N) 4.2.0 Nov 19 2019 / 19:45:55 (881a25698) MSP API: 1.43

# start the command batch
batch start

board_name SPRACINGH7NANO

# name: Xhover

# feature
feature -TELEMETRY
feature -RSSI_ADC
feature -OSD
feature -TRANSPONDER

# beeper
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -ON_USB

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 1 0 115200 57600 0 115200

# led
led 0 9,2::AW:0
led 1 9,3::AW:0
led 2 9,4::AW:0
led 3 9,5::AW:0
led 4 6,5::AW:0
led 5 6,4::AW:0
led 6 6,3::AW:0
led 7 6,2::AW:0

# mode_color
mode_color 6 1 3

# aux
aux 0 0 0 1600 2000 0 0
aux 1 13 1 1000 1200 0 0
aux 2 26 0 1800 2000 0 0
aux 3 31 0 1200 1400 0 0
aux 4 36 1 1800 2000 0 0

# master
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 90
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = SUMD
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set bat_capacity = 1550
set use_cbat_alerts = ON
set cbat_alert_percent = 20
set vbat_duration_for_warning = 50
set vbat_duration_for_critical = 20
set ibata_scale = 184
set pid_process_denom = 1
set name = Xhover

# profile 0
set anti_gravity_gain = 3500
set feedforward_transition = 30
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 10
set ff_interpolate_sp = ON

# rateprofile 0
set roll_rc_rate = 135
set pitch_rc_rate = 135
set yaw_rc_rate = 120
set roll_expo = 15
set pitch_expo = 15
set roll_srate = 75
set pitch_srate = 75

# end the command batch
batch end
```

etracer65, 2019-11-19:
We've decided that you should be using the accelerometer.

Just kidding, it's a bug introduced with some other changes to `small_angle` handling.

mikeller, 2019-11-19:
Fixed by #9188.

etracer65, 2019-11-19:
Fixed in #9201

etracer65, 2019-11-19:
Oops, never mind.

etracer65, 2019-11-19:
Wait, not oops. 😁 

@mikeller - This is a different issue. See #9201

mikeller, 2019-11-20:
D-oh!

PR #9201 (2019-11-21): Fix small angle check when ACC is disabled or not defined
Fixes #9200 

Related to #9185
closed 2019-11-21 by mikeller