betaflight/betaflight #9204: cant change vtx power level

m013, 2019-11-20:
**Describe the bug**
vtx power level is stuck at 25mw (tried it with low power disarm on and off)

**To Reproduce**
try to change it via osd.

**Expected behavior**
to be able to adjust power levels

**Flight controller configuration**
```
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.1.1 Nov 15 2019 / 12:57:05 (1e5e3d369) MSP API: 1.42
# manufacturer_id: MTKS   board_name: MATEKF722MINI   custom defaults: YES

# start the command batch
batch start

board_name MATEKF722MINI
manufacturer_id MTKS

# name: FLOSSTYLE

# resources
resource MOTOR 1 B01
resource MOTOR 2 B00
resource MOTOR 3 B05
resource MOTOR 4 B04

# feature
feature -AIRMODE
feature GPS
feature ESC_SENSOR

# beeper
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -ON_USB
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# map
map AERT1234

# serial
serial 0 2 115200 57600 0 115200
serial 2 8192 115200 57600 0 115200
serial 3 16384 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

# aux
aux 0 0 0 1800 2100 0 0
aux 1 1 1 1375 1625 0 0
aux 2 46 2 1800 2100 0 0
aux 3 13 3 1375 1600 0 0
aux 4 19 4 1975 2100 0 0
aux 5 26 3 900 1125 0 0
aux 6 28 1 1875 2100 0 0
aux 7 33 5 1925 2100 0 0
aux 8 35 1 900 1100 0 0
aux 9 39 6 1250 1450 0 0

# rxrange
rxrange 0 987 2010
rxrange 1 987 2010
rxrange 2 987 2010
rxrange 3 988 2011

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 3
vtxtable powervalues 25 200 500
vtxtable powerlabels 25 200 500

# rxfail
rxfail 5 s 1500
rxfail 7 s 1500

# master
set gyro_lowpass_hz = 0
set gyro_lowpass2_hz = 0
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_lpf_gyro_min_hz = 0
set acc_trim_pitch = -14
set acc_trim_roll = -4
set acc_calibration = -27,73,138
set min_check = 1010
set max_check = 1990
set rssi_scale = 200
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set blackbox_p_ratio = 64
set min_throttle = 1010
set dshot_idle_value = 450
set dshot_bidir = ON
set failsafe_delay = 20
set failsafe_off_delay = 30
set failsafe_throttle = 1150
set failsafe_throttle_low_delay = 150
set failsafe_procedure = GPS-RESCUE
set align_board_yaw = -90
set ibata_scale = 400
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_rescue_angle = 40
set gps_rescue_descent_dist = 150
set gps_rescue_ground_speed = 1400
set gps_rescue_throttle_min = 1170
set gps_rescue_throttle_max = 1400
set gps_rescue_throttle_hover = 1270
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 6
set gps_rescue_min_dth = 180
set gps_rescue_allow_arming_without_fix = ON
set gps_rescue_use_mag = OFF
set pid_process_denom = 1
set thrust_linear = 20
set osd_warn_batt_not_full = OFF
set osd_warn_esc_fail = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_launch_control = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_warn_rssi = ON
set osd_rssi_alarm = 50
set osd_cap_alarm = 1200
set osd_tim2 = 1281
set osd_vbat_pos = 419
set osd_rssi_pos = 14712
set osd_link_quality_pos = 14710
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2389
set osd_remaining_time_estimate_pos = 470
set osd_flymode_pos = 471
set osd_throttle_pos = 8579
set osd_vtx_channel_pos = 2114
set osd_ah_pos = 200
set osd_current_pos = 2370
set osd_mah_drawn_pos = 2444
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 14742
set osd_gps_lon_pos = 2064
set osd_gps_lat_pos = 2051
set osd_gps_sats_pos = 2083
set osd_home_dir_pos = 2102
set osd_home_dist_pos = 2103
set osd_compass_bar_pos = 266
set osd_altitude_pos = 6530
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 8225
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 14698
set osd_avg_cell_voltage_pos = 14691
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 489
set osd_disarmed_pos = 14635
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 14422
set osd_esc_rpm_pos = 83
set osd_flip_arrow_pos = 6350
set osd_stick_overlay_left_pos = 195
set osd_stick_overlay_right_pos = 211
set osd_stat_rtc_date_time = ON
set osd_stat_max_dist = ON
set osd_stat_battery = ON
set osd_stat_max_alt = ON
set osd_stat_max_fft = ON
set debug_mode = GYRO_SCALED
set scheduler_optimize_rate = ON
set vtx_band = 3
set vtx_channel = 5
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5885
set name = FLOSSTYLE

profile 0

# profile 0
set dyn_lpf_dterm_max_hz = 180
set dterm_lowpass_type = BIQUAD
set dterm_lowpass2_hz = 0
set vbat_pid_gain = ON
set crash_time = 650
set crash_delay = 10
set iterm_relax = RPY
set iterm_relax_cutoff = 35
set throttle_boost = 3
set p_pitch = 68
set d_pitch = 35
set f_pitch = 65
set p_roll = 58
set i_roll = 80
set d_roll = 32
set f_roll = 65
set p_yaw = 40
set i_yaw = 0
set d_yaw = 20
set angle_level_strength = 70
set horizon_level_strength = 70
set level_limit = 35
set abs_control_gain = 4
set use_integrated_yaw = ON
set d_min_roll = 19
set d_min_yaw = 12
set d_min_boost_gain = 35
set ff_boost = 20

rateprofile 0

# rateprofile 0
set roll_rc_rate = 131
set pitch_rc_rate = 131
set yaw_rc_rate = 136
set roll_srate = 62
set pitch_srate = 62
set yaw_srate = 62
set tpa_rate = 80
set tpa_breakpoint = 1750

# end the command batch
batch end
```
**Setup / Versions**
 - matek 722 mini
 - speedybee tx500 

**Additional context**
i can enter in low power pitmode and change channels and bands

etracer65, 2019-11-20:
You have:
```
set vtx_low_power_disarm = ON
```
This means the VTX will stay at the lowest defined power level until you arm. On arming it will switch to the selected power level. If you want to test changing power levels then turn this option off.

m013, 2019-11-20:
> 
> 
> You have:
> 
> ```
> set vtx_low_power_disarm = ON
> ```
> 
> This means the VTX will stay at the lowest defined power level until you arm. On arming it will switch to the selected power level. If you want to test changing power levels then turn this option off.

as mentioned above i double checked this option. more over ,  vtx channel is enabled in osd - the power level doesn't change when i arm the quad.

closed 2019-11-20 by m013