betaflight/betaflight #9210: 180 degree servo only has 90 degree range with servo limit from 500us to 2500us

alanypf, 2019-11-22:
Hi all,
I am using the servo_tile feature to control 2 more servos with my kakute F7. The issue is that even the servo limit in the servo pane is 500 to 2500, I still didn't get 180 range. The servo can achieve 180 degrees with the Arduino library. 
Did I miss anything to get the setup correctly?
Thank you for your kind help.

alanypf, 2019-11-22:
It seems that the command above 2000 and below 1000 never reached the servo.

mikeller, 2019-11-22:
@alanypf: Servos are not a focus area for Betaflight, so it is well possible that the 1000 to 2000 range is hard coded as it seems to be the range used by a majority of servos on the market.

Will have to investigate.

closed 2019-12-29 by stale[bot]