betaflight/betaflight #9213: SmartAudio 2.0 custom bands not working

stylesuxx, 2019-11-22:
**Describe the bug**
I am using the factory bands and one custom band (IMD6) on my rush tank race VTX (SmartAudio 2.0). Switching the factory bands works without any issues. But I can not switch to a frequency on my custom band.

This does not only happen with this specific VTX, I have a couple of SmartAudio 2.0 devices that behave the same way.

Also when I change the factory bands to custom I can no longer switch channels.

With my SmartAudio 2.1 devices this works without any issue. It seems to me as if 2.0 does not support non factory channels at all?

**To Reproduce**
Try to use a custom band with SmartAudio 2.0.

**Expected behavior**
I expect custom bands to work with SmartAudio 2.0.

**Flight controller configuration**
```
# version
# Betaflight / STM32F745 (S745) 4.1.0 Oct  6 2019 / 06:48:19 (c6452a55c) MSP API: 1.42
# manufacturer_id: HBRO   board_name: KAKUTEF7   custom defaults: YES

# start the command batch
batch start

board_name KAKUTEF7
manufacturer_id HBRO

# name: stylesuxx

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# map
map TAER1234

# serial
serial 0 2048 115200 57600 0 115200
serial 1 16384 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# aux
aux 0 0 2 900 1300 0 0
aux 1 13 1 1300 2100 0 0
aux 2 35 0 1700 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 50 200 500

# master
set gyro_sync_denom = 2
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set pid_process_denom = 1
set osd_rssi_pos = 2105
set osd_tim_2_pos = 2519
set osd_crosshairs_pos = 2253
set osd_craft_name_pos = 2507
set osd_warnings_pos = 14601
set osd_avg_cell_voltage_pos = 2487
set debug_mode = SMARTAUDIO
set scheduler_optimize_rate = ON
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 3
set vtx_freq = 5917
set vcd_video_system = PAL
set gyro_1_align_yaw = 2700
set name = stylesuxx

profile 0

rateprofile 0

# rateprofile 0
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80
set tpa_rate = 80
set tpa_breakpoint = 1750

# end the command batch
batch end
```

**Setup / Versions**
 - Betaflight 4.1.0
 - Multiple SmartAudio 2.0 enabled video transmitters

etracer65, 2019-11-22:
Contact the manufacturer to see if that VTX supports the custom frequencies you're trying to use. Just because the `vtxtable` is configured for a particular frequency doesn't mean the VTX will accept that.

stylesuxx, 2019-11-22:
I see - just to make sure I understand you correctly - although the ```set frequency``` command is part of the SmartAudio specs manufacturers choose not to implement it?

I will ask the manufacturer - I was just wondering because it happens on 3 different SmartAudio 2.0 devices. From your experience, is it common that custom frequencies do not work on SmartAudio 2.0?

etracer65, 2019-11-22:
> I see - just to make sure I understand you correctly - although the set frequency command is part of the SmartAudio specs manufacturers choose not to implement it?

Whether the VTX switches to the requested frequency is up to the implementation by the manufacturer. Some VTX may refuse any frequency that's not part of their normal band/channels. Other's may "lock out" certain frequencies. And yes, the manufacturer could even choose to not implement frequency-based setting and only support the band/channel mode. Just because it's possible with the SA specification doesn't mean the manufacturer implemented it.

stylesuxx, 2019-11-22:
I understand. Thank you for this information. Since the frequencies are all available on the factory bands and also work there I assume the manufacturers chose not to implement the frequency based setting. Will contact the manufacturers.

closed 2019-11-22 by stylesuxx

etracer65, 2019-11-23:
I just tested on a SA2.0 device and the custom IMD6 band worked just fine. Have you actually checked to see if your VTX is transmitting on the custom channel when you change it? Many VTX don't have a way to display the "custom" information so will just keep displaying the last band/channel on the LEDs even though it has actually changed frequency. Also you can see the frequency change in the OSD menu.

stylesuxx, 2019-11-24:
Thank you for checking it out. Yes I set the channels through the OSD and checked through the goggles. When selecting a channel from the IMD6 band it would just stay on the last channel that I set from one of the factory bands.

I contacted the manufacturers and am waiting for their reply - may I ask on which VTX you checked it? I had the theory that maybe some manufacturers did not implement the frequency setting part of SmartAudio since until now there was no easy way for people to set custom frequencies...