betaflight/betaflight #9223: radix-li 8k/8k loop time not saving

Bigjmf, 2019-11-24:
I am trying to set my loop times on Radix-LI to 8k/8k and it will not work keeps going back to 4k/4k on BF 4.1. on 4.04 I was running 8k/8k. 

board_name RADIX
manufacturer_id 

# name: -

# resources
resource BEEPER 1 NONE
resource MOTOR 1 A02
resource MOTOR 2 A03
resource MOTOR 3 A10
resource MOTOR 4 A15
resource MOTOR 5 C08
resource MOTOR 6 B00
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B14
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 NONE
resource SERIAL_TX 1 B06
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_RX 1 B07
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 C05
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource LED 1 A04
resource LED 2 NONE
resource LED 3 NONE
resource RX_BIND 1 NONE
resource RX_BIND_PLUG 1 NONE
resource SPI_SCK 1 A05
resource SPI_SCK 2 NONE
resource SPI_SCK 3 B03
resource SPI_MISO 1 A06
resource SPI_MISO 2 NONE
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 NONE
resource SPI_MOSI 3 C12
resource ESCSERIAL 1 NONE
[... 928 lines cut ...]
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 65
set tpa_breakpoint = 1250
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end

**Important: This is not the right place to get help with configuration issues. Your issue will be closed without further comment.**

Please get help from other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

etracer65, 2019-11-24:
BrainFPV has their own fork of Betaflight and we don’t provide support for those boards. Contact BrainFPV for support.

Bigjmf, 2019-11-24:
I thought so just wasn't sure

Thank you.

mikeller, 2019-11-24:
Lol @no-response, go home, you're drunk.

closed 2019-11-24 by mikeller