betaflight/betaflight #9232: BF4.2.0 - fc bricked after cli command defaults and save

TheIsotopes, 2019-11-26:
**Describe the bug**
since `build #1972` all my fc's with stm32h7 will bricked after using latest firmware 

**To Reproduce**
flashing firmware higher as `build #1971` go into cli and use the commands `defaults` or `save`
the fc tries to reboot and re-initialize, but it does not succeed and after a while he goes into the dfu mode and is bricked

@jflyper I don't know if it's #9215 but can you check this

**Flight controller configuration**
defaults

**Setup / Versions**
SPRACINGH7NANO and SPRACINGH7EXTREME

mikeller, 2019-11-26:
Confirmed that it is indeed e1244af09ae3979d8ed4d9a3add9bd24c5b3f5a9 that is causing this.

The good news is that this can easily be fixed by undefining `USE_DISPLAYPORT_MSP_VENDOR_SPECIFIC`.

@jflyper: Unless you specifically need to test on H7, I would recommend that we only define `USE_DISPLAYPORT_MSP_VENDOR_SPECIFIC` for boards / architectures that you are testing with.

jflyper, 2019-11-27:
Lemme look into this.

jflyper, 2019-11-27:
Hmm... Doesn't reproduce with NUCLEOH743 and a prototype H743 board. The later has a 7456 chip. May be H750 specific???

jflyper, 2019-11-27:
@TheIsotopes Can do you CLI `status` and report what `Config size` and `Max available config` are?
First with last working version, and next, if possible, with the problematic version.

mikeller, 2019-11-27:
@jflyper: I did some more testing with the RAM based NUCLEOH743 image, and found that it actually wedges earlier, at a12ab13462cea5b7ab4610180364d9abca73018f, in a different way (the init blinks work), so we are potentially looking at a bug that is triggered by the in-memory layout of the code.

jflyper, 2019-11-27:
Oh? I was suspecting config size growing beyond allocated size on the flash and eating into the next section (which I assume a firmware section) and effectively tampering firmware hash value.

mikeller, 2019-11-27:
@jflyper: Config size looks like a possible smoking gun for a compounding problem - after e1244af NUCLEOH743_RAMBASED does not even get to the init blinks - looks like we are having multiple problems.

jflyper, 2019-11-27:
While you are at it, How about shrinking the vendor init string size down so that config size is less than 4096 and see what happens.

mikeller, 2019-11-27:
That's what I was going to do when I discovered that NUCLEOH743_RAMBASED isn't actually working properly even before the vendor init blob is added.
As it is now I'd be taking potshots as I wouldn't be able to confirm that it works with the reduced size, need to fix the first problem first.

TheIsotopes, 2019-11-28:
> Can do you CLI status and report what Config size and Max available config are?
> First with last working version, and next, if possible, with the problematic version.

@jflyper 
Both version have the same output `Config size: 3682, Max available config: 131072`

jflyper, 2019-11-28:
> Both version have the same output `Config size: 3682, Max available config: 131072`

Wth! This is unexpected, in two ways: same size and smaller than 4K...

mikeller, 2019-11-28:
@jflyper: Could the recent changes in compiler play into this? Maybe we should test building newer firmware versions for H7 with the 7.3.1 toolchain?

mikeller, 2019-12-28:
@TheIsotopes: Can you test with the current tip of `master` to see if this is still an issue?

TheIsotopes, 2020-01-02:
@mikeller with latest `master` all is ok

mikeller, 2020-01-16:
@TheIsotopes: Can you please close this then?

closed 2020-01-17 by TheIsotopes