betaflight/betaflight #9233: GPS Rescue not working with 3D Mode enabled

flamingofpv, 2019-11-26:
When i enable GPS Rescue while 3D Mode is on, it doesn't get applied. I set reasonable values for Throttle, but can't get both working. I tried on 3 different Quads, all with the TBS M8 GPS Module on BF 4.1.1

Steps to reproduce the behavior:
Enable both, 3D Mode and GPS Rescue.

Setup / Versions
 - Matek F722SE
 - TBS M8 GPS Module

diff:
 ```
# version
# Betaflight / STM32F7X2 (S7X2) 4.1.1 Nov 15 2019 / 12:57:05 (1e5e3d369) MSP API: 1.42
# manufacturer_id: MTKS   board_name: MATEKF722SE   custom defaults: YES

# start the command batch
batch start

board_name MATEKF722SE
manufacturer_id MTKS

# resources
resource PWM 2 NONE
resource SERIAL_RX 4 NONE
resource CAMERA_CONTROL 1 A01

# feature
feature GPS

# beacon
beacon RX_SET

# serial
serial 0 2 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200

# aux
aux 0 0 0 900 1300 0 0
aux 1 2 1 1300 1700 0 0
aux 2 46 1 1700 2100 0 0
aux 3 13 3 1700 2100 0 0
aux 4 35 2 1700 2100 0 0

# rxrange
rxrange 3 1500 2000

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 14 20 26 30
vtxtable powerlabels 25 100 400 1W

# rxfail
rxfail 3 s 1500
rxfail 5 s 2000

# master
set gyro_lowpass2_hz = 500
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_lpf_gyro_min_hz = 400
set dyn_lpf_gyro_max_hz = 1000
set acc_calibration = 100,40,92
set rssi_scale = 200
set rc_smoothing_derivative_hz = 20
set rc_smoothing_derivative_type = PT1
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set blackbox_p_ratio = 64
set min_throttle = 1070
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_rate = 480
set failsafe_delay = 7
set failsafe_switch_mode = STAGE2
set failsafe_procedure = GPS-RESCUE
set current_meter = ESC
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_rescue_descent_dist = 50
set gps_rescue_min_sats = 5
set gps_rescue_allow_arming_without_fix = ON
set deadband = 2
set pid_process_denom = 1
set osd_vbat_pos = 385
set osd_rssi_pos = 2504
set osd_rssi_dbm_pos = 140
set osd_tim_1_pos = 385
set osd_tim_2_pos = 2465
set osd_vtx_channel_pos = 67
set osd_gps_speed_pos = 2167
set osd_gps_sats_pos = 2471
set osd_home_dir_pos = 2081
set osd_home_dist_pos = 2135
set osd_flight_dist_pos = 87
set osd_altitude_pos = 2103
set osd_warnings_pos = 14802
set osd_avg_cell_voltage_pos = 2497
set osd_disarmed_pos = 2250
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 5
set vtx_power = 4
set vtx_freq = 5806
set pinio_config = 129,1,1,1
set gyro_1_align_pitch = 1800
set gyro_1_align_yaw = 1800
set gyro_2_align_yaw = 900

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 140
set dyn_lpf_dterm_max_hz = 340
set dterm_lowpass2_hz = 300
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 5
set p_pitch = 68
set d_pitch = 57
set p_roll = 64
set d_roll = 53
set p_yaw = 45
set d_min_roll = 30
set d_min_pitch = 33

rateprofile 0

# rateprofile 0
set roll_rc_rate = 120
set pitch_rc_rate = 120
set yaw_rc_rate = 120
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 20
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 75

# end the command batch
batch end
 ```

etracer65, 2019-11-26:
That is correct. GPS Rescue is not designed to support 3D mode.

closed 2019-11-26 by etracer65

chimera15, 2023-10-25:
Then it should stop it in flight from activating, not in the base settings cause I have it disabled by default, if I want to fly normally up a mountain and have rth on, but fly back upside down and choose to have it disabled that's fine but you're preventing me from flying out and possibly losing signal going up the mountain from safely flying up the mountain in normal mode.