betaflight/betaflight #9257: OSD battery precision with 1s and 2s

vodka-bears, 2019-12-03:
**Is your feature request related to a problem? Please describe.**
I'm really unhappy with this: https://github.com/betaflight/betaflight/commit/5d30b3f437e683fce7177c225e556cb7a3bc18e5
This commit message says **Show only 3 _significant_ digits in vbat OSD element**, but with 1s and 2s it displays only two. I fly 1s tiny kwads and it bugs me that it always displays zero in the last digit (e.g. 3.70v).

**Describe the solution you'd like**
If there's 1s or 2s (or vbat is less than 10.0v) display hundredths (e.g. 3.74v). Otherwise display only tenths (14.2v)

**Describe alternatives you've considered**
Add a setting on OSD vbat precision

**Additional context**
If there's no clear refusal from the betaflight team members within 2 weeks, I'll write a pull request. This issue is just to make sure that I don't do a worthless work.

vodka-bears, 2019-12-03:
Also we all don't usually fly models which have vbat changing from more than 10v to less or back in flight. Even if they do this is probably an emergency. So I think we should display 9.93v and 10.1v. If it changes in flight the dot should move so there are only 3 significant numbers left.

etracer65, 2019-12-03:
See discussion here: https://github.com/betaflight/betaflight/pull/8854

mikeller, 2019-12-03:
@vodka-bears:

> Also we all don't usually fly models which have vbat changing from more than 10v to less or back in flight. Even if they do this is probably an emergency. So I think we should display 9.93v and 10.1v. If it changes in flight the dot should move so there are only 3 significant digits left.

Not sure who the 'we' is you are talking about. But since Betaflight supports users using 3S batteries this definitely does not apply to Betaflight users. Maybe the 'we' you are referring to should go and find or develop a solution that is better suited for your needs.

vodka-bears, 2019-12-03:
@mikeller maybe I picked the wrong words but I'm sure there are a lot of Betaflight (whom I don't personally know) users who use it with 1S or 2S. And I _assume_ that it's not only me for whom a zero in the last OSD VBat line digit matches a definition of hack more than what I've proposed in the discussion of #8854 . Also I didn't mean to be toxic/impolite/intrusive, sorry for my maybe emotional topic. If there's still something to discuss let's do it in #8854

vodka-bears, 2019-12-04:
Workaround for 1S: replace **Battery voltage** with **Battery average cell voltage**. I don't have any 2S hardware, so that's all for now.

closed 2019-12-04 by vodka-bears