betaflight/betaflight #9260: LQ in OSD is 199 but shoud be 99

1ofko6, 2019-12-04:
Here is some bug with non erasing 1st number place while LQ is < 100

1ofko6, 2019-12-04:
Betaflight 4.1.1 OSD setting while using crossfire

etracer65, 2019-12-04:
CRSF link quality ranges from 0 to 300. There is no problem.

mikeller, 2019-12-04:
Please read the text when opening issues on GitHub, and follow the instructions.

1ofko6, 2019-12-05:
Well, it works while transmitter mode (RFMD) = 2
but while I´m going for long range (50Hz mode), in osd stays "1" and than 99 is okay
but "1" should not be there

bizmar, 2019-12-05:
300-200 is 150hz mode, 200-100 is 50hz mode, and 100-0 is 4hz mode. (explanation is probably off by one)

I haven't ever seen 4hz mode, but 50 and 150 work like they should.

closed 2019-12-05 by mikeller

rFlightFPV, 2020-02-15:
There seems to be a misunderstanding here on how RFMD modes work in Crossfire. I flashed BF 4.1.3 to one of my quads, using CSRF LQ. The display makes no sense as it is now, let me explain:

Crossfire has 3 RF modes:
150Hz, named RFMD 2
50Hz, named RFMD 1
4 Hz, named RFMD 0

**There is no Crossfire mode where all three modes will be used.** The normal mode, which most of use, used RFMD 2 and RMFD 1, never RFMD 0. If you want to use RFMD 0, you will have to explicitly set this on your module, called the "forced telemetry" mode. 4 Hz is very slow and as per TBS themselves only recommended (quoting the TBS manual) "to only use this RF profile (Forced telemetry mode) when in assisted flight modes (GPS, position hold,waypoint navigation)", clearly not something betaflight was designed for. People that use RFMD 0 for ultra longrange need to use iNAV, because BF per design lacks GPS assisted flight modes. Note that in RFMD 0 (again quoting the TBS manual) "Only 4 new servopositions per second is sent over the link but it will give you 4 times more range compared to the“Normal update-rate” profile." This makes clear that this mode is useless for quads not flying in a GPS assisted flight mode.

What we have now in BF is that LQ shows 300 down to 200 for RFMD2, then 200 down to 100 for RFMD 1. We will never see anything below 100 as RFMD 0 will never be entered as I expained below. I have been flying Crossfire for years and long before it became so popular. The TBS OSD used to display LQ 300 down to 100 for RFMD2 and 100 down to 0 for RFMF 1. There is a common understanding that you don't need to care about LQ when numbers are in the range between 300 and 100, only in RFMD 1, when you drift towards 80 you will need to start monitoring your LQ, at 70 you will need to turn around as a failsafe isn't far anymore. TBS recently removed the very nervous values on RFMD2, as people often misinterpreted this, thinking LQ was degrading fast as numbers dropped towards 100 which intuitively is far away from 300. Instead of constantly changing numbers TBS now only shows 100% (or 99%, can't recall) as long as the link mode is RFMD2. This makes a lot of sense, as the system will first degrade to RFMD1 and take another long degrading of the link quality until you will run into issues. You cannot possibly have an LQ issue when in RFMD2. TBS therefore agreed that there is no information contained in 300 down to 100, as in that range per definition LQ is perfectly fine.

Back to Betaflight: Before the introduction of LQ we used to assign CRSF LQ to say AUX4 and use that as RSSI channel in the configurator. This showed 99% in the OSD for the entire range of RFMD2, and only started to drop when in RFMD1. As with the TBS OSD, if values started to go towards 80 it was considered a "warning" the LQ is getting low, values around 70 considered "critical".

With the current BF LQ implementation we have multiple issues:
- unnecessarily nervous number going from 300 to 200
- when entering RFMD 1, we see 199, which historically was a value nobody ever saw when in RFMD 1. This is the reason this thread was started, as the OP was expecting 99 in RFMD1 to start with. My first thought was also "oh, they forgot to blank out the '1' when it switched to RFMD1".

With the current situation we would need to set our LQ warning level to 180, and the critical level to 170, this makes no sense at all. Everyone not knowing this and expecting to see an LQ of 80 before monitoring LQ will failsafe for sure because their quad will drop out of the sky when LQ hits 100, which in reality is LQ 0. This is dangerous.

May I kindly ask you to rework this, my favorite implementation would be as TBS does it:

1. LQ should show 100% as long as the RF mode is RFMD 2 (99% was actually better as it uses one digits less and frees up space on the OSD)
2. For RMFD1, please scale it from 99 down to 0 so that commonly accepted LQ thresholds are valid in BF as well.