betaflight/betaflight #9270: 6 Positions switch issue with positions 1 & 6 using a Spektrum receiver

livibleu69, 2019-12-08:
Hi,

When using a 6 positions button to select flight rates, positions 1 and 6 cannot be selected when using a Spektrum receiver while it works fine when using a Crossfire or Frsky one...
It seems that positions 1 & 2 are considered as position 2 only, and positions 5 & 6 as position 5 only.

I have just added a rule in the Adjustments tab to use AUX2 to apply Rate profile selection whatever is the position of AUX1, and turn "rate_6pos_switch" ON in CLI.

```
# version
# Betaflight / STM32F411 (S411) 4.1.1 Nov 15 2019 / 12:55:17 (1e5e3d369) MSP API: 1.42
# manufacturer_id: HAMO   board_name: CRAZYBEEF4DX   custom defaults: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name CRAZYBEEF4DX
manufacturer_id HAMO
mcu_id 003400473437510e33353431
signature 

# name: Trashcan

# feature
feature -AIRMODE
feature MOTOR_STOP
feature TELEMETRY
feature LED_STRIP

# beeper
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 0 2048 115200 57600 0 115200

# led
led 0 4,7::CO:10
led 1 5,7::CO:10
led 2 6,7::A:2
led 3 7,7::CO:10
led 4 8,7::CO:10

# mode_color
mode_color 6 1 2

# aux
aux 0 0 0 1100 1625 0 0
aux 1 2 0 1100 1200 0 0
aux 2 13 0 1725 1800 0 0
aux 3 28 0 1375 1475 0 0
aux 4 35 0 1500 1625 0 0

# adjrange
adjrange 0 0 0 1150 1850 12 1 0 0

# rxrange
rxrange 0 1156 1843
rxrange 1 1156 1843
rxrange 2 1156 1843
rxrange 3 1156 1843

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5665 5885 5905 5905 5905
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 3
vtxtable powervalues 14 20 23
vtxtable powerlabels 25 100 200

# master
set gyro_sync_denom = 2
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_min_hz = 100
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1030
set max_check = 1970
set rssi_channel = 8
set blackbox_p_ratio = 64
set blackbox_device = NONE
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set vbat_max_cell_voltage = 440
set vbat_min_cell_voltage = 310
set vbat_warning_cell_voltage = 330
set vbat_scale = 109
set ibata_scale = 1200
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set osd_warn_batt_not_full = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_warn_rssi = ON
set osd_warn_link_quality = ON
set osd_rssi_alarm = 40
set osd_rssi_pos = 14369
set osd_tim_2_pos = 14380
set osd_flymode_pos = 14744
set osd_throttle_pos = 14712
set osd_crosshairs_pos = 2253
set osd_ah_sbar_pos = 4302
set osd_ah_pos = 4174
set osd_mah_drawn_pos = 14424
set osd_craft_name_pos = 14730
set osd_avg_cell_voltage_pos = 14391
set osd_stat_max_spd = OFF
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_g_force = ON
set osd_stat_min_link_quality = ON
set debug_mode = GYRO_SCALED
set rate_6pos_switch = ON
set vtx_band = 2
set vtx_channel = 7
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5847
set gyro_1_align_yaw = 900
set name = Trashcan

profile 0

profile 1

# profile 1
set vbat_pid_gain = ON
set anti_gravity_gain = 7000
set p_pitch = 70
set i_pitch = 25
set d_pitch = 90
set f_pitch = 200
set p_roll = 70
set i_roll = 25
set d_roll = 90
set f_roll = 200
set p_yaw = 90
set f_yaw = 0
set angle_level_strength = 100
set horizon_level_strength = 100
set level_limit = 85
set d_min_roll = 80
set d_min_pitch = 80
set d_min_boost_gain = 30
set d_min_advance = 0
set ff_spike_limit = 255
set ff_boost = 50

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 192
set pitch_rc_rate = 192
set yaw_rc_rate = 192
set roll_expo = 25
set pitch_expo = 25
set yaw_expo = 25
set roll_srate = 63
set pitch_srate = 63
set yaw_srate = 63

rateprofile 1

# rateprofile 1
set roll_rc_rate = 192
set pitch_rc_rate = 192
set yaw_rc_rate = 192
set roll_expo = 25
set pitch_expo = 25
set yaw_expo = 25
set roll_srate = 63
set pitch_srate = 63
set yaw_srate = 63

rateprofile 2

# rateprofile 2
set roll_rc_rate = 156
set pitch_rc_rate = 156
set yaw_rc_rate = 156
set roll_expo = 25
set pitch_expo = 25
set yaw_expo = 25

rateprofile 3

# rateprofile 3
set roll_rc_rate = 130
set pitch_rc_rate = 130
set yaw_rc_rate = 130
set roll_expo = 25
set pitch_expo = 25
set yaw_expo = 25
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 75

rateprofile 4

# rateprofile 4
set roll_rc_rate = 104
set pitch_rc_rate = 104
set yaw_rc_rate = 104
set roll_expo = 25
set pitch_expo = 25
set yaw_expo = 25
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80

rateprofile 5

# rateprofile 5
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 20
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 75

# restore original rateprofile selection
rateprofile 1

```

mikeller, 2019-12-08:
@livibleu69: Spektrum uses a non-default (i.e. not 1000 to 2000) value range for RC channels by default. The easiest way to make this work is to change the configuration of your TX to use 1000 to 2000, or alternatively by setting `rx_min_usec` and `rx_min_usec` to the correct values in Betaflight.

livibleu69, 2019-12-08:
I have noticed that AUX with Spektrum have 1156 and 1843 instead of 1000 and 2000. However RXRANGE only works with channel 0 to 3.
I am using Jumper T16 with multi protocol and cannot (or at least do not know how to) change TX configuration...
On another hand I have just tried to change rx_min_usec and rx_max_usec but could not solve the problem...

etracer65, 2019-12-08:
You're going to have to figure out how to set your endpoints in the radio to range from 1000-2000 for the aux channel.

livibleu69, 2019-12-08:
That is obvious but how? One can only reduce end points (from 100% to 95%)! 
Am I the first to face this issue?

mikeller, 2019-12-08:
@livibleu69: Not sure if you are the first one, but since this is not a Betaflight problem but one concerning the multiprotocol module, you are much more likely to find others in https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues.

etracer65, 2019-12-08:
@livibleu69 

> That is obvious but how? One can only reduce end points (from 100% to 95%)!

With OpenTx you go to the model setup page (1st page) and enable "Extended limits". Then you can set the endpoints beyond +-100%.

But none of this is Betaflight related. And whether the multi-protocol module works this way I can't tell you.

livibleu69, 2019-12-08:
Thank a lot etracer65 it works perfectly !

closed 2019-12-08 by livibleu69