betaflight/betaflight #9279: Link Quality showing as ( instead of LQ in OSD

evilmerlin, 2019-12-10:
1. Describe your problem;
Link Quality in the OSD shows up with a ( instead of LQ. So instead of say LQ300, it is (300. Changing different fonts does not change the symbol.

2. Include ways to reproduce the problem;
Select LQ in the OSD tab.

3. Provide information about your flightcontroller and other components including how they are connected/wired;
This issue was seen in the following flight controllers.
CL_Racing F7 V2/V2.1
Aikon F4
iFlight F7 TwinG

4. Add the used configuration and firmware version;
Issue was observed in both 4.1.1 and nightly #1999

5. Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); 

'''# version
# Betaflight / STM32F7X2 (S7X2) 4.2.0 Dec  5 2019 / 01:00:06 (b18478f43) MSP API: 1.43
# manufacturer_id: IFRC   board_name: IFF7_TWIN_G   custom defaults: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name IFF7_TWIN_G
manufacturer_id IFRC
mcu_id 003a00233237510331313935
signature 

# name: Alvin

# resources
resource SERIAL_RX 4 NONE
resource PINIO 1 C11

# feature
feature TELEMETRY

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# map
map TAER1234

# aux
aux 0 0 0 1850 2100 0 0
aux 1 35 1 900 1125 0 0
aux 2 49 2 1825 2100 0 0

# adjrange
adjrange 0 0 3 900 2100 12 3 0 0

# master
set gyro_lowpass_hz = 225
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set acc_calibration = 76,17,29,1
set mag_hardware = NONE
set baro_hardware = NONE
set rc_smoothing_input_hz = 40
set rc_smoothing_derivative_hz = 20
set rc_smoothing_input_type = PT1
set rc_smoothing_derivative_type = PT1
set serialrx_provider = CRSF
set min_throttle = 1070
set dshot_bidir = ON
set dshot_bitbang = ON
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT300
set motor_pwm_rate = 480
set vbat_max_cell_voltage = 440
set ibata_scale = 206
set small_angle = 180
set pid_process_denom = 2
set thrust_linear = 25
set transient_throttle_limit = 10
set osd_warn_arming_disable = OFF
set osd_warn_batt_not_full = OFF
set osd_warn_batt_warning = OFF
set osd_warn_batt_critical = OFF
set osd_warn_visual_beeper = OFF
set osd_warn_esc_fail = OFF
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_warn_link_quality = ON
set osd_cap_alarm = 1400
set osd_tim1 = 2561
set osd_rssi_pos = 452
set osd_link_quality_pos = 2423
set osd_tim_1_pos = 6530
set osd_throttle_pos = 2402
set osd_current_pos = 2369
set osd_mah_drawn_pos = 6507
set osd_craft_name_pos = 6539
set osd_warnings_pos = 329
set osd_avg_cell_voltage_pos = 2455
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_min_batt = OFF
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_max_curr = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set debug_mode = GYRO_SCALED
set pinio_box = 40,41,42,43
set gyro_2_align_yaw = 900
set name = Alvin

profile 0

# profile 0
set iterm_relax = RPY
set iterm_relax_cutoff = 10
set throttle_boost = 0
set p_pitch = 60
set i_pitch = 117
set d_pitch = 43
set f_pitch = 180
set p_roll = 66
set i_roll = 111
set d_roll = 48
set f_roll = 180
set p_yaw = 47
set i_yaw = 117
set f_yaw = 180
set d_min_roll = 31
set d_min_pitch = 29
set d_min_boost_gain = 35
set d_min_advance = 0
set auto_profile_cell_count = 6
set launch_control_mode = PITCHONLY
set launch_angle_limit = 48
set ff_interpolate_sp = AVERAGED_3

profile 1

# profile 1
set iterm_relax = RPY
set iterm_relax_cutoff = 10
set throttle_boost = 0
set p_pitch = 60
set i_pitch = 117
set d_pitch = 43
set f_pitch = 180
set p_roll = 66
set i_roll = 111
set d_roll = 48
set f_roll = 180
set p_yaw = 47
set i_yaw = 117
set f_yaw = 180
set d_min_roll = 31
set d_min_pitch = 29
set d_min_boost_gain = 35
set d_min_advance = 0
set auto_profile_cell_count = 6
set launch_control_mode = PITCHONLY
set launch_angle_limit = 48
set ff_interpolate_sp = AVERAGED_3

profile 2

# profile 2
set iterm_relax = RPY
set iterm_relax_cutoff = 10
set throttle_boost = 0
set p_pitch = 60
set i_pitch = 117
set d_pitch = 43
set f_pitch = 180
set p_roll = 66
set i_roll = 111
set d_roll = 48
set f_roll = 180
set p_yaw = 47
set i_yaw = 117
set f_yaw = 180
set d_min_roll = 31
set d_min_pitch = 29
set d_min_boost_gain = 35
set d_min_advance = 0
set auto_profile_cell_count = 6
set launch_control_mode = PITCHONLY
set launch_angle_limit = 48
set ff_interpolate_sp = AVERAGED_3

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_srate = 60
set pitch_srate = 60
set yaw_srate = 60
set tpa_rate = 75
set tpa_breakpoint = 1400
set throttle_limit_type = SCALE

rateprofile 1

# rateprofile 1
set roll_srate = 60
set pitch_srate = 60
set yaw_srate = 60
set throttle_limit_type = SCALE
set throttle_limit_percent = 80

rateprofile 2

# rateprofile 2
set roll_srate = 60
set pitch_srate = 60
set yaw_srate = 60
set throttle_limit_type = SCALE
set throttle_limit_percent = 56

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 1

# save configuration
save'''

mikeller, 2019-12-10:
@evilmerlin: What version of configurator are you using? If it is older than 10.6.0 then you will have to update it, as the OSD fonts to be used with Betaflight 4.1 come packaged with configurator 10.6.0.

evilmerlin, 2019-12-10:
I am running 10.6.0. Let me try redownloading and installing it again.

evilmerlin, 2019-12-10:
I have redownloaded and reinstalled 10.6.0. The configurator shows LQ but the actual OSD shows (. Reflashed Betaflight (latest nightly), copied my diff all over and the OSD still shows (.

mikeller, 2019-12-10:
@evilmerlin: Did you actually re-upload the font with configurator 10.6.0?

evilmerlin, 2019-12-10:
Yup! I actually did it a couple of times. Each time the lq still shows up as (. I tried it with both the default and impact fonts.

etracer65, 2019-12-10:
Make sure you have the battery connected (PROPS REMOVED!) when uploading the font. Some flight controllers don’t power the OSD chip properly on USB.

evilmerlin, 2019-12-10:
Yup! Tried that too. Connect USB first then battery then upload font. Select uploaded font in font manager and saved.

mikeller, 2019-12-10:
@evilmerlin: Not sure how you are uploading the font, but this is where you need to do it:

[image]

Tested with an AIKONF4 and your configuration, got the brace. Then uploaded the 'default' font (no battery connected), and I got this:

[image]

evilmerlin, 2019-12-10:
Finally got it working! For some of the other FCs I had to power the quad with a lipo first before plugging the USB and uploading the font.
Thanks for the help!

closed 2019-12-10 by evilmerlin