betaflight/betaflight #9280: Support for non vaporware graphic OSD.

ModdQuad, 2019-12-10:
I have been getting great results with my MCU based OSD.  The secret is to use a chip to 2 stage pipeline with predictable IRQ timing.

Its SPI write compatible with the MAX7456 register set, so I was able to make a video using vanilla BF4.1 and wiring directly to a MAX7456 chip's SPI port.
Video is available at https://youtube.com/video/sXBh9YauKeg/

The initial register set summary and a schematic is available on my git hub page https://github.com/ModdQuad/ModdOSD

A minimosd size development board is designed and getting quoted for manufacture.
It looks like I will also be developing a FC, as I was not able to find any firm commitments within my time frame, and the minim board the reduces risks of putting a firmware engineer in charge of a pick and place enough to justify a more complicated FC design.  Rather just do firmware though.

I plan on open sourcing the firmware once it is feature complete and cleaned up a bit (Q1 2020), with FC development (Q2 2020).

I welcome any comments regarding my proposed changes to the MAX7456 registers or any feature requests or concerns.

Some additional project information at https://www.moddquad.com/p/hardware.html

jflyper, 2019-12-10:
@ModdQuad And I believe you wanted to add SPI pass through or something similar, to update the OSD firmware, right?

jflyper, 2019-12-10:
@ModdQuad Btw, how are you dealing with read side SPI commands? I believe it is extremely hard for a software implementation of an SPI device to read a command and respond in a next cycle.

ModdQuad, 2019-12-10:
There is actually a https://github.com/glegrain/STM32-SPI-Bootloader-host github project I used to test a spi pass though, I am not too worried about that, it can be added in software later.  The betaflight code is easy enough to pull and make (10 minutes),  the part that seems foreign to me is the configuration settings for various FC.  I will eventually need flag to enable graphics etc.  Also is there an open source FC design to follow?  Backup plan is an omnibus clone.

ModdQuad, 2019-12-10:
That's why I only claim write compatibility.   The SPI is all DMA driven into a 1K buffer as NTSC lines (comparator irq) get drawn at 15.626KHz.

ModdQuad, 2019-12-10:
I can run using a UART instead of SPI if reading is a requirement.  Using SPI you can probably read all zeros or ones for 1 bit of status.

hydra, 2019-12-11:
@ModdQuad heh cool!  It's so interesting seeing all the OSD systems.  I've had my own design running since June 2019.  Does your software currently support any non-character based drawing as the BF team are currently looking at this space?

ModdQuad, 2019-12-11:
It currently supports MoveTo, LineTo, BoxTo, FrameTo, and Text commands to a specific graphics object.  Each command is 32 bits, and the SPI registers 0x40-0x7f are free to send up to 16 commands.   The commands are executed by setting a bit in the EXE register.  All graphics are double buffered allow you to send as many commands as needed before updating the display.

A Graphic Object can be displayed at character position,  multiple graphic object can be opened  but is limited to about 1/4-1/3 of the screen because of available memory.

The graphics layer is mixed with the text layer,  the mixer registers sets colors and controls how two layers interact when overlapping.

I am not sure if the SPI port has enough bandwidth for this kind of interface, but it is there.  The preferred method for an artificial horizon is to send the OSD the gyro values and let the OSD software draw the graphics.  This would involve a betaflight change, so its going to be one of the last things I do.

See the proposed register set specification for more information
https://github.com/ModdQuad/ModdOSD/tree/master/Firmware/docs

ModdQuad, 2019-12-11:
There is a hole in that if you send a bunch draw very large box draw commands, the SPI buffer could overflow without any warnings.  I will probably use the 1 bit of status, described above, as form of flow control for EXE commands.

Easy4Racing, 2019-12-12:
Great work, fellow !！Happy to see some new things , Im digging "pixel OSD" these days.
But I want to know clearly what advantages of ModdOSD ?
I guss they are all character OSD not pixel osd.

ModdQuad, 2019-12-12:
Not sure is 'pixel osd' a project , there seems to be a lot of attempts in the space.    
There is just enough memory to run a 2 bit bitmap but little else, so i went a different route.  May revisit for pong/Sparkfun crowd.
Advantages are;
Mine was designed to be cheap (<$2 bom), have a small PCB footprint, and easy to swap with a MAX7456 with no or minimal software changes required.

Easy4Racing, 2019-12-12:
Thks. 
Some manufactures have developed FC which support pixel OSD...
If you have interest, mail to me.I'll reply U some links.

ModdQuad, 2019-12-13:
Closing this issue tomorrow.  This place is for bug tracking not for whatever this it.

I do want to report an issue I noticed during my development.  While not likely, or really a bug, it can happen.  I noticed it only because I am running the spi signal over long wires.

If the MAX7456 chip somehow gets into auto increment mode because of noisy spi signal or glitch.  It remains in this mode indefinitely because the 0xFF escape code will never be sent during normal operation.   This can be mitigated by issuing a NOP read to register 0xFF every once in a while.

ModdQuad, 2019-12-13:
Gatekeepers here, routing around.

closed 2019-12-13 by ModdQuad