betaflight/betaflight #9283: Decouple antigravity multiplier from integral gain

fujin, 2019-12-10:
**Is your feature request related to a problem? Please describe.**
In #9128, we manipulate the antigravity multiplier based on previous changes to integral term defaults. We might instead make antigravity additive rather than multiplicative such that the values can be tuned independently while controlling the ratio.

**Describe the solution you'd like**
AG multiplier decoupled from I-term gain defaults.

**Describe alternatives you've considered**
A curve of some sort instead of fixed multiplier or addition.

IllusionFpv, 2020-01-18:
I'm working on this https://github.com/IllusionFpv/betaflight/tree/ag-decoupled-from-iterm

mikeller, 2020-02-01:
@fujin: I guess with #9389 merged this can now be closed?

fujin, 2020-02-01:
Added in #9389, closing

closed 2020-02-01 by fujin