betaflight/betaflight #9285: Remove support for Proshot

sskaug, 2019-12-11:
In BLHeli_32, we plan to remove support for Proshot in future revisions.
There are at least three reasons for this:
- Proshot is not much used, and can readily be replaced by Dshot
- Proshot signaling is very timing sensitive and not as robust as Dshot
- Removing it saves flash space and reduces complexity in BLHeli_32

mikeller, 2019-12-11:
@sskaug: To my understanding ProShot was only ever working poorly in BLHeli_32 due to hardware limitations?
This probably means that not many users will have used ProShot with BLHeli_32 anyway, and I do not know how the fact that you are removing support for ProShot from BLHeli_32 would change anything in the motivation for Betaflight to support ProShot.

@TonyBazz: By any chance, are you planning to remove support for ProShot from your ESC firmware now that BLHeli_32 is doing the same?

sskaug, 2019-12-12:
I think in most cases Proshot work well with BLHeli_32. But ESCs generally do not use a crystal and the internal clock has an accuracy of up to almost 4% for temperature extremes. I think if we make an error budget analysis of the 16 timing steps of Proshot with 4% clock frequency error, it will fail.
Dshot is more robust and I can't see what additional benefit Proshot brings over Dshot.

ledvinap, 2019-12-12:
@sskaug : 
- Proshot can use adaptive timing (there are pulses of known length, this can be used; there are possible improvements in framing to make it easier)
- Proshot can transfer more information per edge then DShot. This is useful when resources are associated with edge = using timer (or even interrupt) for data output/input. This conceptual difference in IMO largely overlooked ...

sskaug, 2019-12-12:
Indeed, proshot has more info in fewer edges. Which puts more info on the time axis. With "15" being 23 cycles, there is only a timing difference of 4.5% down to 22 cycles. This generally works, but is not sufficient to guarantee performance over temperature etc with the MCU internal oscillators we use.

ledvinap, 2019-12-12:
@sskaug : RC oscillator is usually tunable. Proshot packet length is alway constant (or falling edge distance), so you can use this to adjust RC frequency. And as a side effect there will be exact timing for motor control (probably not important) and serial telemetry ...

On FC side with timers used for output, proshot is more efficient (less time taken by DMA, smaller buffer space).
Proshot also guarantees minimum pulse width, which will allow interrupt-based implementation with smaller interrupt latency requirements and  considerably smaller CPU utilization (could be useful on ESC side).

But it is true that dshot works too and supporting both is nuisance ..

adrianmiriuta, 2019-12-23:
I think PROSHOT's variant of a transport  layer protocol is superior to DSHOT.

I develop a GPL v3 ESC firmware (https://github.com/3x8/nostromo)
It uses only PROSHOT and (PWM for thrust tests only)

1. higher Information content (higher speed, less delay, on same packet size)
2. lower CPU usage for decoding, smaller memory footprint.
3. master-slave Clock calibration is possible
4. the fact that it uses DSHOT data layer conventions minimizes the support cost.

adrianmiriuta, 2019-12-25:
@sskaug 
 > Proshot signaling is very timing sensitive and not as robust as Dshot

I disagree

with my firmware
PROSHOT CRC error rate under real flight conditions is less then 3/million 
(CLRACINGF7, WRAITH32V2, 4KHz PID loop)

closed 2020-01-31 by stale[bot]