betaflight/betaflight #9286: Unstable connection to configurator, FC disconnects almost immediately

acidg, 2019-12-13:
Connection to Betaflight Configurator is unstable. The FC disconnects itself after some seconds or won't connect at all.

Steps to reproduce the behavior:
1. With Betaflight Configurator open, connect FC to USB, click "Connect" as soon as the port shows up (or use auto-connect feature).
2. Connection will (sometimes) be established, configuration is showing up, virtual copter moves (slower than usual) according to movements of the FC.
3. After a very short time (1-3 seconds), the FC disconnects itself, log shows "successfully disconnected"
4. Clicking connect shows "port successfully opened with ID...", but nothing is happening.

**Expected behavior**
FC should stay connected to Betaflight configurator.

**Flight controller configuration**
Can not `diff`, since connection is not stable enough.

**Setup / Versions**
 - Flight controller Mamba F405 MK2 2019 from Banggood
 - Tried Betaflight 4.1.1, 4.0.3, 3.5.7
 - Betaflight Configurator 10.6.0
 - Nothing else is connected to the FC

jflyper, 2019-12-13:
Can you describe what the "usual" environment/setup is?

acidg, 2019-12-13:
I am using the latest Arch Linux. FC is connected to the USB port of my laptop. I started the latest configurator unsing sudo, since accessing the serial port was an issue otherwise. What else do you need?

tenten8401, 2019-12-13:
@acidg Try disabling ModemManager and close Ultimaker Cura if you have it open. ModemManager is what did it for me, since it tries to probe anything with a serial port.

jflyper, 2019-12-13:
The "usual" is as in "virtual copter moves (slower than usual)" in your problem description. If the FC still behaves "usual", then you can rule out the dud FC case.

If you are using Linux, I would check if there is a service watching/scanning serial ports, like modem services and such.

acidg, 2019-12-15:
Problem was Ultimaker Cura running in the background. It's working flawlessly now. Thanks a lot for the quick help!

closed 2019-12-15 by acidg

gak4u, 2021-05-30:
> @acidg Try disabling ModemManager and close Ultimaker Cura if you have it open. ModemManager is what did it for me, since it tries to probe anything with a serial port.
> `systemctl disable --now ModemManager`

Worked as soon as closed Cura. You are a lifesaver.