betaflight/betaflight #9294: Add ability to change VTX power from control Switch in flight

Improg, 2019-12-17:
Please, add changing of VTX power in flight, when armed, using a switch. Changing VTX power via smart audio in flight will help avoid interference when the model is next to other pilots.

etracer65, 2019-12-17:
You can already do this. For example the following will configure a 3-position switch on Aux3 to switch power levels 1/2/3.
```
vtx 0 2 0 0 1 900 1200
vtx 1 2 0 0 2 1300 1700
vtx 2 2 0 0 3 1800 2100
```
This can also be used to switch to specific band/channels based on a switch but since the band/channel in the example above are 0 it won't change channels - only power. See the `help` for the `vtx` command.
```
vtx - vtx channels on switch
	<index> <aux_channel> <vtx_band> <vtx_channel> <vtx_power> <start_range> <end_range>
```
```
```
**NOTE: There's currently a bug in that you can't use the highest configured power level as the parameter.** So if you configured 3 power levels in the `vtxtable` then the 3rd line above will give a parsing error. This will be fixed in the next maintenance release. See #9295

chichoamado, 2020-01-07:
@etracert65. I'm asking for the same adjusment tab in infligth adjustment. I don't know that can be done via cli.
Can you explain in your example witch means every number? I do not see the aux channel 3 in your example. Please help. Thank you.

etracer65, 2020-01-07:
@chichoamado The information is listed above and in the output from `help` in the CLI.

```
<index> <aux_channel> <vtx_band> <vtx_channel> <vtx_power> <start_range> <end_range>
```
The aux channel parameter is number starting at 0 so 0 = aux1, 1 = aux2, etc.  So the 2 in this case is aux3.

SchonRichtig, 2020-01-07:
> Can you explain in your example witch means every number? I do not see the aux channel 3 in your example. Please help. Thank you.

[(https://www.youtube.com/watch?v=JREW4BdnwwU)]

See the Video discription for examples

chichoamado, 2020-01-07:
Thank you both. Si helpfull.

closed 2020-02-13 by stale[bot]