betaflight/betaflight #9299: Runcam Hybrid (style) cameras

ltwin8, 2019-12-20:
The RunCam Hybrid has two separate UART-Ports, one for the Digital HD Part, one for the Analog OSD Part.

Is it possible to add support to use both ports simultaneously (on 2 separate parts of course)

etracer65, 2019-12-20:
Would take a pretty significant rewrite. Suggest it to RunCam - they wrote the original code to support their cameras.

ltwin8, 2019-12-20:
So enable both simultaneously is not possible? for example Runcam device checked on uart_1 and uart_2?

etracer65, 2019-12-20:
I didn't say it was impossible. I said it would take significant work. If you want this feature feel free to work on the code and submit your solution. This is after all an open source project.

closed 2020-01-26 by stale[bot]