betaflight/betaflight #9300: Dec 12 Nightly: OSD RSSI() telemetry value wrong by factor of 2 - showing 50 instead of 100 when close

RipperDrone, 2019-12-21:
**Bug**
As title says

**To Reproduce**
flash Dec 12th Nightly, include RSSI in OSD display

**Expected behavior**
show same value as in versions prior to this Nightly (OSD parameter grouping seems to have made the difference...)

**Flight controller configuration**
# manufacturer_id: IFRC   board_name: IFF7_TWIN_G_D   custom defaults: YES

**Setup / Versions**
FC: iFlight Twin_G D F7, RX = R-XSR with latest FPORT flash from FrSky web page, other components don't matter I suppose

**Additional context**
can easily flip-flop bug on and off btw Dec 12 Nightly and previous versions

McGiverGim, 2019-12-22:
Please add all the info asked when you open the issue.

The RSSI value reported by the latest r-xsr firmware has changed, I don't know if it's a bug or a feature from frsky. Have you installed the latest r-xsr firmware?

RipperDrone, 2019-12-22:
@McGiverGim yes, pls read my post, it's been stated there. Latest FPORT flash.

RipperDrone, 2019-12-22:
@etracer65 what do u mean? I only skipped the diff, but explained why: Bug can easily be reproduced by 

1. flashing BF Dec 12th Nightly
2. loading defaults
3. using FPORT protocol and 
4. enabling RSSI in OSD

That's it.

@McGiverGim In case FrSky has changed the reported value to half, this might be the explanation. However, would be very confusing to use a max value of 50 instead of 100, isn't it??

RipperDrone, 2019-12-22:
On their webpage, FrSky don't state anything aout altered RSSI scaling...
[image]

RipperDrone, 2019-12-22:
Plus: Flashing BF 4.1 runs fine, shows 100 correctly as max RSSI. BF 4.2 Dec 12th Nightly shows 50. Seems to me this can't be related to FrSky scaling then, must be up to BF...

RipperDrone, 2019-12-22:
Here's the diff, even though all info needed to clearly reproduce the bug was posted earlier. 

```# version
# Betaflight / STM32F7X2 (S7X2) 4.1.1 Nov 15 2019 / 12:57:05 (1e5e3d369) MSP API: 1.42
# manufacturer_id: IFRC   board_name: IFF7_TWIN_G_D   custom defaults: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name IFF7_TWIN_G_D
manufacturer_id IFRC
mcu_id 002600303238511336353235
signature 

# name: RFK R5

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 1 8192 115200 57600 0 115200
serial 4 1024 115200 57600 0 115200

# aux
aux 0 0 0 1300 1700 0 0
aux 1 1 1 1300 1700 0 0
aux 2 2 1 1700 2100 0 0
aux 3 27 0 1700 2100 0 0
aux 4 13 0 1700 2100 0 0
aux 5 35 2 900 1300 0 0
aux 6 49 0 1300 1700 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 LOWBANDL L CUSTOM  5362 5399 5436 5473 5510 5547 5584 5621
vtxtable powerlevels 4
vtxtable powervalues 25 100 200 400
vtxtable powerlabels 25 100 200 400

# master
set gyro_lowpass_hz = 150
set gyro_lowpass2_hz = 150
set dyn_lpf_gyro_min_hz = 0
set dyn_lpf_gyro_max_hz = 425
set acc_calibration = 35,2,24
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1030
set max_check = 1970
set max_aux_channels = 4
set serialrx_provider = FPORT
set airmode_start_throttle_percent = 13
set serialrx_halfduplex = ON
set blackbox_p_ratio = 64
set dshot_idle_value = 350
set dshot_burst = OFF
set dshot_bidir = ON
set dshot_bitbang = ON
set failsafe_off_delay = 25
set failsafe_throttle = 1070
set failsafe_switch_mode = KILL
set failsafe_throttle_low_delay = 50
set failsafe_procedure = AUTO-LAND
set vbat_max_cell_voltage = 450
set ibata_offset = -1900
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set osd_warn_core_temp = OFF
set osd_cap_alarm = 500
set osd_vbat_pos = 2081
set osd_rssi_pos = 2088
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2113
set osd_flymode_pos = 461
set osd_anti_gravity_pos = 385
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_ah_pos = 200
set osd_current_pos = 2102
set osd_mah_drawn_pos = 2136
set osd_craft_name_pos = 2093
set osd_gps_speed_pos = 34
set osd_gps_lon_pos = 498
set osd_gps_lat_pos = 466
set osd_gps_sats_pos = 1
set osd_home_dir_pos = 100
set osd_home_dist_pos = 1
set osd_compass_bar_pos = 266
set osd_altitude_pos = 33
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 391
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2378
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 20
set osd_disarmed_pos = 2411
set osd_nheading_pos = 311
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set debug_mode = GYRO_SCALED
set vtx_band = 3
set vtx_channel = 2
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5685
set camera_control_internal_resistance = 270
set camera_control_button_resistance = 450,270,150,75,15
set gyro_1_align_yaw = 2700
set gyro_2_sensor_align = CW0
set name = RFK R5

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 60
set dterm_lowpass2_hz = 100
set p_pitch = 23
set i_pitch = 55
set d_pitch = 24
set f_pitch = 30
set p_roll = 22
set i_roll = 50
set d_roll = 22
set f_roll = 30
set p_yaw = 19
set i_yaw = 60
set f_yaw = 30
set d_min_roll = 0
set d_min_pitch = 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 120
set pitch_rc_rate = 120
set yaw_rc_rate = 120
set tpa_rate = 70
set tpa_breakpoint = 1350

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save```

RipperDrone, 2019-12-22:
Suspicion is that Unified Targets might have changed the RSSI scaling value for OSD RSSI value calculation?

etracer65, 2019-12-22:
I just tested with the latest build (2008) and everything is normal. Using a XSR with FPort firmware.

> Suspicion is that Unified Targets might have changed the RSSI scaling value for OSD RSSI value calculation?

No. That unified target config hasn't been updated since it was originally submitted on 31-Oct.

etracer65, 2019-12-22:
Also your `diff` is from 4.1.1 - not from the build where you said you have the problem. I'm not sure why you persist with making it difficult to get the requested information. Whether you think it relevant or not it's the information we require.

RipperDrone, 2019-12-22:
@etracer65 - this is how I saved it. It is exactly the diff I am copy-pasting into 4.1.2 flash. Ok, will again flash up and share the 4.1.2, but it will be the defaults plus the posted diff, nothing else.

My RX is R-XSR so maybe there's a difference in the ACCST flashes from FrSky? Anyway, don't get why scaling is correct in BF4.1.1 and incorrect in 4.1.2. Only RSSI related commits I can see is OSD PG change (compared them line by line, everything looks good) and Unified Targets... :-(

RipperDrone, 2019-12-22:
```# version
# Betaflight / STM32F7X2 (S7X2) 4.1.2 Dec 22 2019 / 02:21:16 (89051e256) MSP API: 1.42
# manufacturer_id: IFRC   board_name: IFF7_TWIN_G_D   custom defaults: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name IFF7_TWIN_G_D
manufacturer_id IFRC
mcu_id 002600303238511336353235
signature 

# name: RFK R5

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 1 8192 115200 57600 0 115200
serial 4 1024 115200 57600 0 115200

# aux
aux 0 0 0 1300 1700 0 0
aux 1 1 1 1300 1700 0 0
aux 2 2 1 1700 2100 0 0
aux 3 27 0 1700 2100 0 0
aux 4 13 0 1700 2100 0 0
aux 5 35 2 900 1300 0 0
aux 6 49 0 1300 1700 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 LOWBANDL L CUSTOM  5362 5399 5436 5473 5510 5547 5584 5621
vtxtable powerlevels 4
vtxtable powervalues 25 100 200 400
vtxtable powerlabels 25 100 200 400

# master
set gyro_lowpass_hz = 150
set gyro_lowpass2_hz = 150
set dyn_lpf_gyro_min_hz = 0
set dyn_lpf_gyro_max_hz = 425
set acc_calibration = 35,2,24
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1030
set max_check = 1970
set max_aux_channels = 4
set serialrx_provider = FPORT
set airmode_start_throttle_percent = 13
set serialrx_halfduplex = ON
set blackbox_p_ratio = 64
set dshot_idle_value = 350
set dshot_burst = OFF
set dshot_bidir = ON
set dshot_bitbang = ON
set failsafe_off_delay = 25
set failsafe_throttle = 1070
set failsafe_switch_mode = KILL
set failsafe_throttle_low_delay = 50
set failsafe_procedure = AUTO-LAND
set vbat_max_cell_voltage = 450
set ibata_offset = -1900
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set osd_warn_core_temp = OFF
set osd_cap_alarm = 500
set osd_vbat_pos = 2081
set osd_rssi_pos = 2088
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2113
set osd_flymode_pos = 461
set osd_anti_gravity_pos = 385
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_ah_pos = 200
set osd_current_pos = 2102
set osd_mah_drawn_pos = 2136
set osd_craft_name_pos = 2093
set osd_gps_speed_pos = 34
set osd_gps_lon_pos = 498
set osd_gps_lat_pos = 466
set osd_gps_sats_pos = 1
set osd_home_dir_pos = 100
set osd_home_dist_pos = 1
set osd_compass_bar_pos = 266
set osd_altitude_pos = 33
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 391
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2378
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 20
set osd_disarmed_pos = 2411
set osd_nheading_pos = 311
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set debug_mode = GYRO_SCALED
set vtx_band = 3
set vtx_channel = 2
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5685
set camera_control_internal_resistance = 270
set camera_control_button_resistance = 450,270,150,75,15
set gyro_1_align_yaw = 2700
set gyro_2_sensor_align = CW0
set name = RFK R5

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 60
set dterm_lowpass2_hz = 100
set p_pitch = 23
set i_pitch = 55
set d_pitch = 24
set f_pitch = 30
set p_roll = 22
set i_roll = 50
set d_roll = 22
set f_roll = 30
set p_yaw = 19
set i_yaw = 60
set f_yaw = 30
set d_min_roll = 0
set d_min_pitch = 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 120
set pitch_rc_rate = 120
set yaw_rc_rate = 120
set tpa_rate = 70
set tpa_breakpoint = 1350

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save

RipperDrone, 2019-12-22:
How can it scale differently in lua telemetry script (shwing 100 max correctly) visa OSD in goggles (showing 50)? Is there any soft scaling that has been changed? Strange to believe this is linked to XSR vs R-XSR or FrSky firmware sided... 🤔

RipperDrone, 2019-12-22:
Have tried #2008, no luck.

RipperDrone, 2019-12-22:
checked RSSI_scaling parameter in both versions, it's 100 as supposed...

klutvott123, 2019-12-22:
Could it somehow be showing the percentage instead of the dBm value? Configurator shows the percentage and it's exactly half of the dBm

RipperDrone, 2019-12-22:
@klutvott123 Are we talking about the same glitch?:

- I put RSSI_value on the display, NOT RSSI_dbm. So I expect percentagem thus 100 max. Instead, it's showing 50 on BF OSD, but 100 on lua telemetry script on my Taranis.

Seems vice versa compared to what you are suggesting?

klutvott123, 2019-12-22:
@RipperDrone Just saying that the configurator shows 50 when my tx shows 100. I would assume it's the same value that you have in your OSD.
I have 4.1.0 with R-XSR latest FPORT firmware on one of my quads. It shows 50 in the configurator and 100 on my transmitter. Flashed it to 4.1.2 and it's the same. Flashed it to 4.0.0 and it's the same.

Then I tried with another quad that has R-XSR with older firmware. Flashed to 4.1.2 and it's fine. 

Then I tried with the first quad again downgrading the firmware to 190311 and it's fine again.
Sure looks like a frsky feature to me. Not sure why it works for you with 191128 and earlier versions of BF.

RipperDrone, 2019-12-22:
I see now. Wow, this is getting complex: You get 50 on 4.1.0 and 4.0.0 with latest FPORT FW on R-XSR when I get 100. What else could we try - is your radio FW OTX 2.3.3? Only difference I see btw our setups, isn't it...?

And again - I can toggle btw 50 and 100 in OSD display by flashing 4.1.1 <-> 4.1.2 so still believe this can be a FrSky issue only. Only difference is BF version...?

etracer65, 2019-12-22:
Nothing changed in 4.1.2 that would affect this.

RipperDrone, 2019-12-22:
@etracer65 There must be something - I am keeping every single component and all settings the same, just toggling btw BF4.1.1 and 4.1.2 switches the bug off and on. So there has to be an influence...
@klutvott123 mentioned that one RSSI value might be half of the other btw. dBm and 'value' (percentage) RSSI? Could it be s/t about the order of OSD elements got mixed up somehow btw BF versions so OSD toggles btw percentage and dBm scale RSSI btw versions 4.1.1 and 4.1.2, under the same OSD element ID?

klutvott123, 2019-12-22:
@RipperDrone Yes, 2.3.3. Like @etracer65 says there we're no changes in BF that would cause this to happen. The value shown in the configurator should be independent of the osd changes anyway.

The only difference I can see is that with 190311 the rssi won't go higher than 102(1023).
With 191128 it clearly goes higher. Maybe it's messing with the scaling somehow.

RipperDrone, 2019-12-22:
@klutvott123 Ok, thank you for confirming. For my better understanding, how can the RSSI value shown on RC via telemetry script be any different from the value shown in OSD in general? Shouldn't it be the same basic variable used for both? Would liek to understand better where / how any scaling happens.

RipperDrone, 2019-12-22:
Found on FPVInfo: 
[image]

klutvott123, 2019-12-23:
@RipperDrone rx.c/fport.c is where it happens. Looks perfectly fine to me.
The RSSI you see in opentx is sent as telemetry from the RX. The one you see in betaflight is received as part of a control frame. I'm not sure if it's added by the tx or just tacked on by the rx before passing it on to the FC. Looks to me like the value we're receiving is actually half of what it "should" be for some reason.

etracer65, 2019-12-23:
It's related to the most recent R-XSR firmware. For some reason Frsky changed the scaling. Nobody seems to know if it was intentional or a bug.

RipperDrone, 2019-12-23:
ok, I will send a request to FrSky and hope they will respond and clarify. 

Still curious why even on the 'buggy' R-XSR firmware, and with all the hardware and settings kept the same
- downgrading of OTX version on RC works for @klutvott123
- downgrading BF to 4.1.1 works for me

to fix it. 🤔

klutvott123, 2019-12-23:
@RipperDrone I downgraded the R-XSR firmware. Used opentx 2.3.3 for all tests. Please let us know if you get a response from them 👍 
For those using the FCC firmware I'm not sure if there's any benefit of upgrading to 191128. I suspect the fix for "abnormal high frame loss rate" mostly applies to LBT. I have tested both and the LBT firmware has had severe frame loss.

RipperDrone, 2019-12-23:
@klutvott123 Yep, let's see. Today, I tried to roll back OTX firmware to 2.3.2 (acc to your report this should be another route to fix the bug for BF4.1.2), but even Companion 2.3.2 suggests downloading 'latest' 2.3.3 firmware, and I haven'T been able to figure out a way to download a 2.3.2 firmware file. How did you do it, kept the old .bin? I didn't... :-(

klutvott123, 2019-12-23:
@RipperDrone I didn't. I used 2.3.3 the whole time and still am. I downgraded the R-XSR firmware. R-XSR firmware 190311 doesn't have this issue.

RipperDrone, 2019-12-23:
Ok, I got this hint from @Asizon - who had other issues with 2.3.3 and suggested it was due to OTX version. Re-reading his feedback it seems it is still uncertain to pertain to this issue here, so not confirmed yet.

So we actually have to conclude it's a re-scaled R-XSR frame rate firmware issue's interplay with BF versions 4.1.1 vs 4.1.2 somehow, even though nobody has an idea why 4.1.2 behavior should be different... 

Since I am using de-pinned R-XSR with univerted mod for F4 MCUs on FPORT for the last months, rolling RX firmware back on my recent, problematic is much more pain than staying on BF 4.1.1 for now, which runs fine with latest FrSky's FPORT FW. At some point, I will have to take a call, though :-)

I'll try fiddling with RSSI scaling next as a workaround :-)

PS: I found a thread where people where challenging the 300 max value of CRSF RX, and it seems it shows triple frame transmission rate, thus being rated as 300% RSSI 'fidelity'. If same rationale applies to FrSky's firmware update, maybe the recent update eliminates re-sending of non-changed frames, therefore reducing effective transmission frame rate? Just a shallow attempt to find an explanation... ;-)

RipperDrone, 2019-12-23:
For the record: rssi_scale = 200 is a valid workaround, telemetry RSSI max stays at around 100, OSD and BFC displayed RSSI max value increases from 50 to 99 :-)

etracer65, 2019-12-23:
Has nothing to do with OpenTx. It's completely related to the latest firmware for the R-XSR. Flash your receiver back to the previous firmware version and the problem will go away.

closed 2019-12-23 by RipperDrone

RipperDrone, 2019-12-24:
Replay from FrSky --> it seems we are back in this old 'blame the other side' kind of game... :-(

[image]

etracer65, 2019-12-24:
https://betaflightgroup.slack.com/archives/C221J1H54/p1574942378087200?thread_ts=1574942378087200&cid=C221J1H54

Read the thread.

RipperDrone, 2019-12-24:
Following this thread, it looks they might have tried to align aux channel based RSSI's scaling to FPORT integrated RSSI's, in case they have been scaled differently before... funny the FrSky reps don't know or want to comment...

McGiverGim, 2019-12-24:
It has been discussed here too: https://www.rcgroups.com/forums/showthread.php?3089797-Low-RSSI-signal-on-R-XSR/page7

In my case, the correct scaling now seems to be 175, not 200.

klutvott123, 2019-12-24:
Ray from frsky can say what he wants. I logged the raw rssi data from the fport control frame and it's half of what the telemetry RSSI is in opentx. Are we supposed to change/experiment with  the scale/offset/whatever whenever a new RX firmware is released? Why can't they send the same RSSI value both ways?

RipperDrone, 2019-12-24:
> It has been discussed here too: https://www.rcgroups.com/forums/showthread.php?3089797-Low-RSSI-signal-on-R-XSR/page7
> 
> In my case, the correct scaling now seems to be 175, not 200.

Thank you, I was already thinking that setting the scale factor to 200 seemed a little high... :-) Still difference btw OSD and shown-on-Taranis telemetry RSSI values :-)

RipperDrone, 2019-12-24:
> Ray from frsky can say what he wants. I logged the raw rssi data from the fport control frame and it's half of what the telemetry RSSI is in opentx. Are we supposed to change/experiment with the [...]

@klutvott123 Oh yeah, speaking more than true words - so confusing having to adjust bugs by scale factors differently with every version coming along... I'll stick to Ray, maybe one of the FrSky guys will reveal a little more...

RipperDrone, 2019-12-24:
Here's my conspiracy theory: FrSky is trying to squeeze us users out from ACCST by making us pizzed with changing scalings and semi-working RX firmware details again and again, just to make us finally shift over to the 'holy grail of protocols' ACCESS (which doesn't yet find a broad enough acceptance to make a real favorable-to-FrSky impact on the RC market...). Of course far-fetched somehow... ;-)

tlebrell, 2020-04-06:
I have to echo some of what RipperDrone has said. Similarly, I have a de-pinned R-XSR using uninverted FPORT pad, and I'm getting a max of 50% RSSI in BF OSD. Where we differ, is that I'm running BF 4.1.5 on a Speedybee F4 board.

I started out with a new RX with a fresh flash of ACCESS firmware. After being unable to find resolution of what appeared to be half RSSI value in BF OSD, I flashed it back to ACCST thinking that it would solve the problem, but the problem persists.

My TX is an X-Lite Pro, I'd be interested to know what kind of TX RipperDrone is using.

mikeller, 2020-04-06:
@tlebrell: See https://github.com/betaflight/betaflight/issues/9300#issuecomment-568777515.