betaflight/betaflight #9301: STM32F7X2 overflows with Debug symbols.

kmitchel, 2019-12-22:
When building STM32F7X2 target on current master with `DEBUG=GDB` `AXIM_FLASH1' overflows by 34184 bytes.

```make clean_all
git checkout master
git fetch --all
git reset --hard upstream/master
make TARGET=STM32F7X2 DEBUG=GDB``

By removing features from common_pre.h, I can link successfully.

etracer65, 2019-12-22:
Yep, 512K of flash.

mikeller, 2019-12-22:
STM32F405 is currently the preferred target for debug builds - the 1MB flash means that the target can be built with debug symbols / no optimisation and still fit into flash. For all other MCU types, irrelevant features have to be disabled for the debug build to fit into flash.

kmitchel, 2019-12-22:
Also a handful of `may be used uninitialized in this function` warnings.

Just consider it a known issue and close?

etracer65, 2019-12-22:
I fixed the build warnings in #9303

closed 2019-12-22 by kmitchel