betaflight/betaflight #9312: Bidirectional / Motor Tab Shows E: 100%

Nullstr1ng, 2019-12-25:
[image]

[image]

I have flashed JESC 2.2  firmware in my ESC, enabled Bidirectional in Configurations tab, plugged in the battery and setup any necessary setting (hopefuly shown in my CLI dump). But still shows E: 100% ? I was thinking should the ESC have a telemetry support before someone can use the JESC firmware? Or all BLHeli_S ESCs with x-H-## is supported with this feature? I can arm my quad (not flown) and spin the motor.

FAQ:
1. FC+ESC - Diatone Mamba F405 Mini Mk1 
https://www.rcgroups.com/forums/showthread.php?3126404-MAMBA-F405-stacks

2. Versions
Betaflight 4.1.1
JESC 2.2 24Khz

`# dump

# version
# Betaflight / STM32F405 (S405) 4.1.1 Nov 15 2019 / 12:54:53 (1e5e3d369) MSP API: 1.42
# manufacturer_id: DIAT   board_name: FURYF4OSD   custom defaults: YES

# start the command batch
batch start

board_name FURYF4OSD
manufacturer_id DIAT

# name: jar@

# resources
resource BEEPER 1 A08
resource MOTOR 1 A03
resource MOTOR 2 B00
resource MOTOR 3 B01
resource MOTOR 4 A02
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 C09
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 A00
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
[... 1002 lines cut ...]
# rateprofile 0
set rateprofile_name = -
set thr_mid = 50
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 60
set pitch_expo = 60
set yaw_expo = 60
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 65
set tpa_breakpoint = 1250
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end`

etracer65, 2019-12-25:
JESC is not a Betaflight project. Seek support from the vendor.

closed 2019-12-25 by etracer65

mikeller, 2019-12-25:
@Nullstr1ng: You might want to read the contents of screenshots before you post them - as it states, JESC requires you to buy licenses for your ESCs for it to support bidirectional Dshot which is needed for RPM filtering, which you apparently have not done:

[image]

So you either need to buy licenses for JESC, or you can use version 16.71 of BLHeli_S that has support for bidirectional Dshot:

[image]

Nullstr1ng, 2019-12-25:
> @Nullstr1ng: You might want to read the contents of screenshots before you post them - as it states, JESC requires you to buy licenses for your ESCs for it to support bidirectional Dshot which is [...]

I did read, but the not fully understood. 
It said I can flash the ESC with JESC firmware without using a license using the "Flash All" button. Although it said "consider purchasing" (optional I believe) purchased or not, I could still use the RPM Filtering feature. But anyway, if that's the case. Then I'll purchase it 8 today.