betaflight/betaflight #9313: telemetry problem

waynepatt, 2019-12-25:
cant get telemetry to work with R-XSR receiver and Succex E F405  flight controller. Have the latest firmware on R-XSR  and it is bound to transmitter Taranis X9D Plus SE 2019  Sticks work fine for throttle etc. so is communicating with the FC  The R-XSR is version 2 and firmware has been flashed for F. Port version. Signal must be OK coming through F. Port wire as I can move sticks.

Telemetry has been enabled in Beta flight. yet when I go on the taranis to discover sensors I only find the 2 located on the R-XSR receiver. So is this a configuration issue with Beta Flight?

mikeller, 2019-12-26:
@waynepatt:

> Signal must be OK coming through F. Port wire as I can move sticks.

Just guessing here since you did not supply the requested information, but I suspect you connected your receiver to the RX pin of the UART and not the TX pin - FPort will not work on the RX pin.

waynepatt, 2019-12-26:
thanks you were exactly correct, working OK now.

Yours Sincerely,
Wayne Patterson

Sent with [ProtonMail](https://protonmail.com) Secure Email.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

closed 2020-02-01 by stale[bot]