betaflight/betaflight #9314: Replace RC_SMOOTHING_RATE with proper debug streams

mikeller, 2019-12-26:
Currently, the `RC_SMOOTHING_RATE` debug stream is made up of a mishmash of measurements based on data input and measurements based on RX task timings.
This makes it impossible to use this stream for anything except for RC smoothing tuning.

This should be replaced with two debug streams, `RC_INPUT_RATE` and `RC_TASK_RATE` to track one distinct set of measurements each.

This will allow users to use these debug values without having to worry about what measurements they represent.

If needed, amalgamation of `RC_INPUT_RATE` and `RC_TASK_RATE` into a mixed data stream can be done on the client side.

etracer65, 2019-12-28:
`RC_SMOOTHING_RATE` should not be replaced. It contains more than the current inter-frame interval. It contains the details showing the behavior of the RC Smoothing rate training used to capture the frame rate and set the filter cutoffs.

Having people secondarily using it to display receiver frame timing doesn’t invalidate its original designed function.

mikeller, 2019-12-28:
If there are even more metrics mashed up into a single debug stream, can we please separate them into different stream, and restore the functionality of `debug` to provide value for users for debugging?

etracer65, 2019-12-29:
`RC_SMOOTHING_RATE` provides the metrics for debugging the RC Smoothing rate training as designed. It was added only for this reason. Subsequent uses of debug(0) developed later. If you want to provide some other metrics that’s fine, but `RC_SMOOTHING_RATE` is required as is.

mikeller, 2019-12-29:
I think we should either rearrange `RC_SMOOTHING` and `RC_SMOOTHING_RATE` to make space for the separate input based and task based rates in them, or else extract these streams into a new debug group, and drop or replace `debug[0]` in `RC_SMOOTHING_RATE`, and then direct users to use these for debugging, including debugging of RC smoothing.
Keeping code for the collection of data around after it has been replaced with the collection of more accurate and versatile data just adds to the effort required for maintenance.

etracer65, 2019-12-29:
I'm really at a loss to understand why this is apparently such a big problem. I don't see any problems with the data or its uses. `RC_SMOOTHING` and `RC_SMOOTHING_RATE` debug modes are different things and both appropriate to their purpose. `RC_SMOOTHING` is used to examine the performance of the smoothing filters on the input data. To see how well the cutoffs and filter types are performing for the input and derivative.

Debug 0: Raw input RC channel signal
Debug 1: Setpoint derivative after input smoothing but before derivative smoothing
Debug 2: Setpoint derivative after derivative smoothing
Debug 3: Current trained frame rate used to set the filters

`RC_SMOOTHING_RATE` is used to debug the performance of the RX frame rate training and filter auto-cutoff determination.

Debug 0: Last RX frame rate
Debug 1: Training step count
Debug 2: Current "locked" frame rate
Debug 3: Training state machine status

The one you seem so concerned about is debug(0) of `RC_SMOOTHING_RATE`. This contains the best information we have about the current RX frame rate. Previously it only contained the task-derived rate (as that's all we had), but now it can contain the protocol-derived (much more accurate) or the task-derived if protocol version is not available (not implemented or not possible). So the point is that it contains at worst the same information we had before, at best much more accurate information. Yes, this can fallback to the 30ms value if frames stop, but this is irrelevant because, **frames have stopped**. Receiver in failsafe not providing frames doesn't present any useful frame rate data and having it fall back to 30ms is just as useful an indicator of that as anything else. All the processing that relies on actual rx frame rates also take into account `isRXDataNew` which is not set in cases where we've fallen-back to the 30ms minimum processing.

Again, neither of these are meant to "debug" the receiver. Debug(0) of `RC_SMOOTHING_RATE` has been helpful to see the behavior of the various receiver protocols and helped us identify problems. But the point of identifying these problems has been in the context troubleshooting problems with RC Smoothing performance. By adding support for the more accurate protocol-level calculations we've been able to identify and fix problems that were previously unnoticed because they were lost in the "noise" of the previous task-based measurements. But at a high-level all we did is improve the accuracy of the debug(0) data. Nothing about it has changed in any other way.

mikeller, 2019-12-30:
It is not that I see this as 'such a big problem'. It is just that I think that single-use functionality should be replaced with generic functionality that provides more value to more users where this is possible. And collecting input based and task based RX timings individually is definitely more widely usable than combining them into a single data stream and discarding the information about what the values actually represent. Once this is done, users will get the same value out of using these data streams as they get out of using `debug[0]` in `RC_SMOOTHING_RATE`, and I do not think it is justified to keep and maintain the additional complexity and the hooks / statics in every protocol driver that is required to collect this information.

mikeller, 2020-03-15:
Implemented in #9511.

closed 2020-03-15 by mikeller