betaflight/betaflight #9318: FPV Camera Angle Compensation -Enhancement

KSCopter, 2019-12-27:
The current implementation of FPV Camera Angle Compensation does a pretty good job of correcting for the visual perceptual error created by tilting the camera.  It works exceptionally well when the quad is moving forward at EXACTLY the angle specified in the GUI.  

The problem, however,  is that the quad is almost never traveling at the exact specified angle set in the GUI.  Pitch is dynamic and therefore the effective compensatory angle is also dynamic.

A more accurate compensatory angle needs to take into account not only the fixed camera angle but also the pitch of the quad.  

For example, when flying a fast quad with very high camera tilt on the race course in a tight, slow section where the quad is moving slower and flying at low pitch (lower than the angle of camera tilt set in the GUI), the quad is essentially over-corrected.  This leads to increased visual perceptual error and makes slower or more technical sections of the race course more difficult to fly.  At higher speeds and higher tilt angles it is not generally a problem.

Personally, I have found setting the compensatory angle in the GUI at about half or a little less than half of the actual camera angle to give the best results on a race quad, as that seems to be a fairly good average for a flight that requires constantly changing pitch.  But, it could be so much better. 

THE PROPOSED ENHANCEMENT(S):
1.  Since the flight controller is already monitoring the pitch angle, make the FPV camera compensation angle a dynamic number that takes into account actual pitch.   This would provide the most accurate correction at any pitch angle.

(fixed camera angle) + or - (actual pitch angle of the quad) = Applied compensation angle

2.  A simpler but less perfect approach could be to change the compensation angle base on throttle, as throttle and pitch are generally proportional in a forward moving quad.  
Less throttle = less pitch = less compensation.  and vice versa.  I could see this working well within the first 60-70% of the throttle curve (perhaps that could be configurable, too.)

(FPV Camera Compensation angle) X (throttle %) = Applied compensation angle

Thank you developers for your consideration.  I truly appreciate everything you do.  I hope this will at least lead to a fruitful discussion.

simonbethke, 2019-12-28:
Why would an FPV Pilot want to use this compensation? In any case it would make the copter move unpredictable because there are physics...

KSCopter, 2019-12-30:
FPV Camera Angle Compensation is an existing feature within Betaflight.  Joshua Bardwell has a  good video explanation of it here:  https://www.youtube.com/watch?v=gEcWxHIH378.  

It is especially useful for CINEMATIC recording/flying.  It helps keep the horizon flat on yaw and the center of the screen centered on roll.  

Imagine driving or flying with your head tilted back.  Something has to compensate.  That is what this feature takes care of.  

Uptilt compensation (UC) is places the axis of rotation in the middle of the pilots point of focus in the goggles.   It is nearly perfect when the quad is moving forward at the same degree of tilt as the Compensation angle set in the Betaflight configuration tab .  In this scenario, the pilot point of focus is directly in the middle of the goggle screen, as is the axis of quad rotation.  

Think of the goggle screen as a window.  If the quad is tilted forward beyond the UC angle( and the fixed camera angle), pilot point of focus moves higher on the screen.  This requires more uptilt compensation to place the axis of rotation at the pilot point of focus.  If we were sitting in the quad, we would tilt our head up to make up for this (so we are not looking down at the ground. 
 
If the quad is travelling forward at a tilt angle below the user specified Compensation angle (and below the fixed camera angle), the pilot point of focus moves to the lower part of the screen. (so we don't look at the sky).  Less UC would move the axis of rotation down. 

The current implementation only works perfectly when the quad is moving at exactly the level of specified tilt in the configuration tab.  But flight is dynamic.

**Here is my proposed Enhancement:**  

For every degree of tilt the quad flies at above the specified Compensation angle set in the GUI -add an additional degree to the user specified Compensation and visa-versa.

**Example #1:**   A quad wtih 45 degrees of fixed camera uptilt traveling at a 50 degree angle of forward tilt:  Add 5 degrees to the user specified  uptilt compensation.

**Example #2**  A quad with 45 degrees of fixed camera uptilt, traveling forward at a tilt angle of 30 degrees:  Subtract 15 degrees from the user specified uptilt compensation. 
 
There is probably a useable range for this within 20-25  degrees on either side of the user specified angle in the Configuration tab.  The effects might be subtle but could turn out to be a big enhancement, especially for cinematic flyers and possibly for racers who choose to use it.

simonbethke, 2019-12-30:
How can you ever know the offset between the viewing vector and the motion vector? You never know the motion vector at all...

KSCopter, 2019-12-30:
The camera is set at a fixed angle (a known number).  The flight controler/accelerometer is monitoring the actual pitch of the quad.  You can even display it in the OSD, if you like.  The pitch data collected by the FC  is used for features like crash recovery, launch mode, air/horizon mode.

The current implementation of FPV angle compensation uses the fixed camera angle set in the configuration tab.  I am only proposing adding or subtracting the difference between the fixed camera angle set in the configuation tab and the actual/real-time pitch measured by the FC.  There is no motion vector.

simonbethke, 2019-12-30:
But you do know, that the accelleration sensor measures accelleration. It cannot tell if your copter is flying upside down or not.

KSCopter, 2019-12-30:
The flight controller certainly knows when the aircraft is upside down. That is why flight controllers use 3-axis accelerometers.  

Here is an article on how accelerometers are used for tilt sensing.
https://wiki.dfrobot.com/How_to_Use_a_Three-Axis_Accelerometer_for_Tilt_Sensing

Remember,  camera angle compensation is already an existing feature of Betaflight.  The math is quite brilliant.   I am merely proposing to enhance it with additional data already collected by the flight controller.  

Like I said, this would probably work in a specific range on either side of the user specified angle in the configuration tab.

simonbethke, 2019-12-31:
So if I put the copter upside down and accellerate with approx. 18.6m/s^2
towards the ground. The acc will tell that the copter is hovering.

KSCopter, 2019-12-31:
I think your questions/comments could be addressed more appropriately on the Betaflight Slack exchange.

senorblasto, 2020-02-20:
I'm not sure who you are trying to explain the features of betaflight to exactly, but most of your audience are the people who wrote the code themselves.  
The accelerometer is not used when you are flying in acro mode.  Only the gyro gets used.
But from what I can understand it sounds like you want your quad to behave the way it already does just in angle mode.
to quote the legendary @etracer65 the great one:

> @etracer65 
 The feature is designed to relative to the FPV camera's frame of reference - hence the name "FPV angle mix" and works the way it should. This is not a line-of-sight feature. If you want yaws to remain parallel to the ground then use angle or horizon mode with auto-leveling.

> @etracer65 
The feature is designed to compensate for the camera up-tilt to make it appear from the FPV pilot's perspective that the requested rotation around each axis is as if the camera up-tilt was zero degrees. The FPV angle mix performs the cross-coordination on the other axes so that the pilot doesn't have to do it. The frame of reference is the vector defined by the FPV camera angle. If you still don't understand, hold a quad with the up-tilted camera level to the ground. The frame will obviously be tilted forward with negative pitch. Now yaw around the center of the flight controller's Z axis and see what happens to roll and pitch. Normally the pilot compensates for this in flight. The FPV angle mix does it instead. All that is needed to calculate this is the angle of the camera relative to the flight controller.

AlessandroAU, 2020-07-20:
@KSCopter you have a misunderstanding about how an accelerometer measures tilt. Tilt can only be accurately measured when the acceleration vector is pointing directly down at the ground. Ie, level flight, it cannot possibly work during race or acro as the centripetal forces (which can hit 10g(!)) mean that the vector no longer points downwards. 

This can be observed directly if you look at 'craft orientation' indicator during a hard acro flight. For example: https://www.youtube.com/watch?v=GloaeXDtU6k

closed 2020-07-21 by KSCopter

a-marmot, 2021-09-04:
I don't think you want to change the angle compensation according to measured attitude.  This results in roll and yaw along axes not aligned with the camera, which defeats the purpose of the correction.