betaflight/betaflight #9322: Throttle limit on a switch / knob - Feature Request

riveraj082, 2019-12-27:
Is your feature request related to a problem? No

I would like to control the throttle limit with a knob switch. Currently they are tied to a rate profile. I would like each to be independent from each other. I would like to control the throttle limit separate from rates.

I have not found an alternate way of doing this in betaflight. There is a similar way of doing this manually on the remote itself.

mikeller, 2019-12-27:
@riveraj082: I think you are overthinking this - if you want to use a knob or a switch on your TX to control the throttle, it will be much easier if you set this up on your TX, before the throttle stick value is sent to the craft.

riveraj082, 2019-12-27:
You might be correct. I like controlling everything via betaflight, it makes it easier for me when i clone multiple racers. It get to use one model on the remote. I have also been told you loose some throttle resolution when doing it on the remote. Either way, i would like this feature.

mikeller, 2019-12-27:
@riveraj082: 

> I like controlling everything via betaflight, it makes it easier for me when i clone multiple racers.

While I agree with this in most cases, this seems to be a case where you want to use Betaflight as a straightforward mixer of two values on the TX, at the waste of an entire RC channel, which does not make much sense. Just set up a profile for this on your TX?

> I have also been told you loose some throttle resolution when doing it on the remote.

Can you specify how much throttle resolution you require? We can then check if configuring this on the TX side will still leave you with sufficient resolution.

riveraj082, 2019-12-28:
Since you mentioned controlling throttle cut on the remote i have been trying to figure out how to perform this action. On OpenTX 2.3.X the knob is not a select-able switch for most functions like, logical switch, special functions, etc. I do see the capability to use a regular switch for most functions. Do you happen to know how to perform a throttle cut using a knob on the remote? Currently i perform this function by manually reducing or increasing the max value on the throttle output.

mikeller, 2019-12-28:
@riveraj082: That's simple - in the 'Mixes' tab configure a channel that is a Multiplex ('multiply' type) of the throttle channel (scaled to be 0 ... 100 with Weight 50, Offset 50 for simplicity) and an analogue input scaled to output 80 ... 100 (Weight 10, Offset 90):

[image]

Then on the 'Outputs' tab you will have to set the Min to 0 to match the mix for the throttle channel.
This will allow you to seamlessly scale your throttle from 80% to 100% with the analogue input.

d3c0deFPV, 2019-12-28:
>Can you specify how much throttle resolution you require? 

@mikeller At which point would the reduction in throttle resolution via OpenTX start to become a perceptible change on your average 5" quad and a Taranis? I'm curious where the bounds would be of what we can perceive or would affect performance in any noticeable way. If you don't already know, I may explore this.

Also, I realize that this is probably a good option for OpenTX users, but bringing it to Betaflight would open this option up to anyone, As far as wasting an RC channel, channel packing gets us around that as an issue.

WrathPak, 2019-12-28:
@mikeller not everyone has a radio that can do this. For example, I use the DJI Digital FPV Radio. I love it. But it has no ability in the radio to change throttle limit. It does however have a horribly slow menu to attempt to do a few things in limited fashion similar to this from the goggles. I would love to have the option for this on a switch or dial on my radio given the opportunity. Spending 2+ min each time you want to switch it with alternative measures instead of what a switch or knob can do in less than a second is a bit annoying.

mikeller, 2019-12-28:
@d3c0deFPV:

> At which point would the reduction in throttle resolution via OpenTX start to become a perceptible change on your average 5" quad and a Taranis? I'm curious where the bounds would be of what we can [...]

I don't know, and I guess it depends on the user. But I suspect that a reduction from 1000 steps to 800 steps as I proposed above is not enough to make a noticeable difference if tested blindly.

> Also, I realize that this is probably a good option for OpenTX users, but bringing it to Betaflight would open this option up to anyone, As far as wasting an RC channel, channel packing gets us around that as an issue.

Channel packing will not work if you want to be able to seamlessly regulate limiting as in the example above. Also, as stated right at the start, for users not wanting to / able to do this on the TX, there is always the possibility to do this through rate profiles.

You have to realise that, when asked, every user will probably come up with a different setting that they would like to have inside / outside of rate profiles or profiles, and there is no reason whatsoever that your preference should be chosen over everybody else's - especially as this is something that can be easily worked around on the TX for the majority of the hardware that is in use.

@WrathPak: Set up different rate profiles with different throttle limits, and use a switch on the TX to switch between the rate profiles.

closed 2020-02-04 by stale[bot]

gavinwilliams, 2021-03-01:
I'm willing to pick this up, it would be nice to do this through BetaFlight as there is already functionality to set the throttle limit.

Being able to set a 100%, 50% and 0% throttle limit and map it to a three or two position switch would be useful for anyone who doesn't have a teranis. I find this particularly useful if I'm teaching someone how to fly personally and don't want to give them access to full throttle yet.