betaflight/betaflight #9324: Flyaway pid auto disarm

frizzle101101, 2019-12-30:
**Is your feature request related to a problem? Please describe.**

So i just armed my 3 inch for the first time and it sits on the ground nicely and not to noisy seeming but as soon as the weight goes into hover the pid takes over and my quad sails to the moon. Unfortunately i did this in my house and theres a good impression in the ceiling of my diatone frame. Other quads have done this in the past because of the classic d noise. All it took was the pid gain of 0.8 for the 3inch and pd gain of 0.7 cuz its a 6s...but dont be stupid like me and try to hover test a new quad in the house

**Describe the solution you'd like**
Somthing like a disarm a soon as the d or pid wants to use full osilitaions on the axis for a cupple cycles soon after arm then display a "flyaway pid may not be tuned" message in the osd

**Describe alternatives you've considered**
This may be implemented already with the crash prevention or some or safety feature... luckily after the quad hit the roof it disarmed so that worked but can we stop it earlier?

frizzle101101, 2020-01-01:
Did no one reply cuz its the Holidays...idk i thought it may be a good idea.

bizmar, 2020-01-03:
@joelucid had a prototype of something with the same goal, but it was never finished. It seemed promising in the few preliminarily flights I tried it (quad with NO filters didn't fly away)

Here's the commit, if anyone wants to run with it. :) https://github.com/joelucid/betaflight/commit/fce04052f37ff1f729d86578aa6f28bd36e850e4

closed 2020-01-17 by no-response[bot]