betaflight/betaflight #9325: G-force throttle

TiagoTiago, 2019-12-31:
**Is your feature request related to a problem? Please describe.**
When changing props, motors, and of course, quads, any given throttle stick position may mean different things even if all the numbers set on Betaflight are the same; not to mention battery sag.

**Describe the solution you'd like**
Instead of mapping the throttle stick directly to motor RPM (or whatever is done now), make use of the accelerometer, and perhaps a 4th PID loop or some other more appropriate algorithm, to make it so a given stick position makes the quad attempt to pull a certain amount of g's along it's own vertical axis (hoverpoint, while the quad is perfectly flat relative to gravity, would pull 1g; min throttle would aim for zero-g, and max throttle would aim for a user selected maximum g-force). Understandably, if the quad is not perfectly flat, then the hover-point would not be enough to keep the quad from accelerating downwards, a separate mode that takes the angle relative to gravity in consideration would be required to address this, such additional mode is not the focus of this feature request.

The extrapolation to 3D mode is obvious, simply allow for negative g's to be targeted as well, automatically reversing motors as needed.

I'm not proposing for the conventional throttle method to be fully deprecated; in some cases it makes sense to be able to go to exactly the maximum RPM at full stick, whatever it  that RPM may be. I'm just proposing an extra toggle, and accompanying settings, the pilot may elect to use to normalize behaviour between quads, and different sets of props, motors, and battery states.

It might be useful to add a new warning to be displayed on the OSD and/or trigger beeps, for when the motors are maxed out while the quad is still not reaching the targeted g-force; to help pilots tune the maximum g-force target to be more suited to the quad at hand if desired (though, that will only be useful for blips; since attempts of continuous acceleration will eventually start suffering from aerodynamic challenges, allowing for no further, or at least only reduced, acceleration).

**Additional context**
I am aware that some boards allow for Altitude-Hold functionality; but that's depends on the presence of a barometer, which not all boards have; and makes things too stiff and slow for freestyling. The idea is to automate Throttle the way Pitch, Roll, and Yaw are automated in Acro mode, where stick deflection is mapped to an specific physical movement rate regardless of the details of the hardware being flown.

etracer65, 2020-01-01:
This is not possible. “G-forces” are acceleration, not a measure of speed. To maintain a constant “g-force” it would require constantly increasing the throttle to accelerate faster and faster upward. Also the accelerometer is measuring **ALL** acceleration. So in flight we can’t differentiate between acceleration caused by a turn for example and gravity.

TiagoTiago, 2020-01-01:
> This is not possible. “G-forces” are acceleration, not a measure of speed. To maintain a constant “g-force” it would require constantly increasing the throttle to accelerate faster and faster upward.

I don't see what's the issue. Isn't the quad already experiencing acceleration whenever the throttle is not at zero anyway? I understand there is a limit, I mentioned in the OP that there would be a point where the throttle is maxed out but the airdrag won't allow further acceleration; but quads aren't always constantly flying at that limit, are they?

>Also the accelerometer is measuring ALL acceleration. So in flight we can’t differentiate between acceleration caused by a turn for example and gravity.

This proposal doesn't require gravity to be differentiated from acceleration; I understand gravity is acceleration; I'm just proposing for a more explicit way to control the acceleration of the quad along the thrust axis.

etracer65, 2020-01-01:
Again, velocity is not acceleration. If you increase the throttle there will be a momentary spike in positive acceleration in the z-axis but then it will immediately return to 1.0g. The only way we can measure vertical velocity is with a barometer (as a vario) and that low resolution.

TiagoTiago, 2020-01-01:
Do quads instantly achieve a constant velocity once the throttle stick is moved? If not, then there you are, acceleration. I'm well aware acceleration is change of speed.

And regarding "vertical", I mean relative to the quad, not to the gravitational field. And again, not speed, acceleration.

TiagoTiago, 2020-01-02:
Btw, I've been reviewing some DVR footage recently, and noticed that indeed the quad is staying above 1g for several continuous seconds during flight.

closed 2020-02-08 by stale[bot]