betaflight/betaflight #9328: Make OSD menu readable under every condition

lbedeschi, 2020-01-01:
**Problem**
Depending on what the FPV cam is pointing at, it might be hard or impossible to read the OSD menu text.

**Solution (feature request)**
Dim the FPV feed with a dark overlay, or at least add a black background to every text item.

etracer65, 2020-01-02:
Sorry, not possible with the hardware on the flight controller. The OSD chip can only overlay text, not affect the base video signal. There are a variety of fonts available in the Configurator. Choose one that works better for you.

etracer65, 2020-01-02:
Potentially you could design a font that has a black background, but this would affect the characters during normal OSD when you're flying as well (only one font can be active).

Jetrell, 2020-01-04:
@lbedeschi  Here's a reverse colored font like @etracer65 suggested. I haven't tried it. So let us know if its the sort of thing your after?
[impact font reverse color.zip](https://github.com/betaflight/betaflight/files/4021644/impact.font.reverse.color.zip)

Unpack the zip to get .mcm. Then load it through - `OSD` > `font manager` > `open font file`

ModdQuad, 2020-01-11:
I can fix the menu by adding grey under the text.  (software osd, lots of fonts)  I made a demo drop-down menu demo a while back,  but the fc will not be available for another 3 months.

closed 2020-02-17 by stale[bot]