betaflight/betaflight #9330: Kakute F7 HDV, Tekko32 F3 Combo current sensor reading very inaccurate

Xedos9er, 2020-01-02:
I found some topics about this problem on other forums, where nobody was able to fix this issue.
The current sensor reading is very inaccurate on my Kakute F7 HDV / Tekko32 F3 Stack. It jumps around from 0.1-0.8A on table. While cruise on constant throttle it´s reading 7-10A. 

Is there any way to smooth out the current reading or is it a hardware flaw? I noticed that the refresh rate is much higher compared to my other FC´s. Mabe there´s some way to implement some kind of smoothing algorithm??

klutvott123, 2020-01-02:
``ibat_lpf_period`` lets you adjust the current filtering. Would help if you gave us the requested info.
Anyway this sounds very similar to #8233. You could try the solution posted there and see if it helps.

Xedos9er, 2020-01-03:
Thank you very much. The capacitor has done the trick! I soldered a and capacitor between GND and Current Pin directly on the connector and it fixed it. So it's a hardware flaw.

dzenbot, 2020-01-04:
I'm glad you fixed it. @Xedos9er you should then close this issue.

closed 2020-01-04 by Xedos9er