betaflight/betaflight #9331: Barometer shows always 0 in BF 4.1.2 OMNIBUSF4SD Airbot

3DRacer, 2020-01-03:
At BF 4.1.2 Barometer shows always 0 at my Airbot F4SD clone (OMNIBUSF4SD (AIRB)).
At BF 4.1.1 OMNIBUSF4SD (Legacy) it works flawless.

3DRacer, 2020-01-03:
Board is known (OMNIBUSF4SD), Version is known (Betaflight 4.1.2) and Issue is described (Fix Value "zero" altitude shown in OSD).
what information should i submit?

McGiverGim, 2020-01-03:
When you open an issue there is a text explaining the info needed to help the developers to find the problem. For example a diff. Push the open issue and read the text.

3DRacer, 2020-01-04:
Since BF 4.1.2 OMNIBUSF4SD (Legacy) is not available anymore. Therefore I installed OMNIBUSF4SD (AIRB). In that configuration Barometer is shown as available, but ahown value is always zero. Tested in osd settings altitude. No problems in latest available BF 4.1.1 OMNIBUSF4SD (Legacy).

see diff all below.

# 

# diff all

# version
# Betaflight / STM32F405 (S405) 4.1.2 Dec 22 2019 / 02:19:35 (89051e256) MSP API: 1.42
# manufacturer_id: AIRB   board_name: OMNIBUSF4SD   custom defaults: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name OMNIBUSF4SD
manufacturer_id AIRB
mcu_id 002100243037471639333631
signature 

# master
set osd_altitude_pos = 14570
set gyro_1_align_yaw = 2700

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#

closed 2020-01-04 by 3DRacer