betaflight/betaflight #9335: Barometer shows always 0 in BF 4.1.2 OMNIBUSF4SD Airbot

3DRacer, 2020-01-04:
Since BF 4.1.2 OMNIBUSF4SD (Legacy) is not available anymore. Therefore I installed OMNIBUSF4SD (AIRB). In that configuration Barometer is shown as available, but shown value is always zero. Tested in osd settings altitude. No problems in latest available BF 4.1.1 OMNIBUSF4SD (Legacy).

see diff all below.
diff all
version
Betaflight / STM32F405 (S405) 4.1.2 Dec 22 2019 / 02:19:35 (89051e2) MSP API: 1.42
manufacturer_id: AIRB board_name: OMNIBUSF4SD custom defaults: YES
start the command batch

batch start
reset configuration to default settings

defaults nosave

board_name OMNIBUSF4SD
manufacturer_id AIRB
mcu_id 002100243037471639333631
signature
master

set osd_altitude_pos = 14570
set gyro_1_align_yaw = 2700

profile 0

profile 1

profile 2
restore original profile selection

profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5
restore original rateprofile selection

rateprofile 0
save configuration

save

Asizon, 2020-01-04:
You need to try same version for unified target, 4.1.1(Legacy)vs 4.1.1(AIRB) to make sure that it is Unified Target system problem, please try both several times. OMNBUSF4SD unified config seems correct.

3DRacer, 2020-01-04:
I can confirm. the barometer does not work with 4.11 airb either. the error only became apparent to me after the compulsory change from legacy to airb.

Asizon, 2020-01-04:
My Airbot OmnibusF4SD board seems Ok.
[image]

Asizon, 2020-01-04:
4.1.2 this
[image]

3DRacer, 2020-01-04:
what should these pictures show? as i wrote at the beginning, the barometer is recognized. put altitude in osd and look at it via vtx. no change in value, always zero

3DRacer, 2020-01-04:
[image]

3DRacer, 2020-01-04:
[image]

and here same copter with bf 4.1.1 legacy

Asizon, 2020-01-04:
I don't know what can happens, maybe this clone board is not able to correctly initialize the baro spi  device (BMP280) with the unified target system

3DRacer, 2020-01-04:
I agree. in any case, the effect can be found in the difference between the two BF versions mentioned

Asizon, 2020-01-04:
The problem you can be seen in all versions under the unified target system. I don't know if baro detection can be modified in unified target code to work on clone boards, maybe @etracer65  can offer an answer to this.

3DRacer, 2020-01-04:
We will see. otherwise, many users will have to stay on legacy 4.1.1, when baro is used.

jflyper, 2020-01-05:
Try pulling up PB3.

3DRacer, 2020-01-05:
resource is already B3. With B3 is in BF4.1.1 Legacy no problem with barometer.

Asizon, 2020-01-05:
@3DRacer I think he said try to pull up baro CS pin with a resistor

3DRacer, 2020-01-05:
ok. pullup resistor. basically I am capable of this. But to fix a malfunction caused by a software change by a hardware change is not useful

Asizon, 2020-01-05:
I think that the malfuction is in clone boards manufacturers making bad hardware, you have alot of recently examples like Eachine Vtx.

etracer65, 2020-01-05:
@3DRacer Is your board a genuine Airbot OmnibusF4SD flight controller? If not it's a clone that has likely implemented the hardware incompletely. Targets are only designed to work with genuine boards and if at some point a poorly implemented clone starts failing through software changes (that still work on the genuine board) then it's not something that we'll fix.

But even so, the idea to add a pull-up is just to test a possible cause (and common problem with clones). It still may not fix your problem - but is a valid test.

3DRacer, 2020-01-05:
As test a 1K resistor from 5V to PB3? But that still has to wait. At the moment this board is in a super flying copter with rpm filtering and a properly working 3D mode. That is why I currently have little motivation to test it. I was hoping for an easier solution

etracer65, 2020-01-05:
`resource PULLUP 1 B03`

3DRacer, 2020-01-06:
I set resource PULLUP 1 B03 in CLI, saved and restarted FC. But Barometer still in zero. So that test was not success.

diff

# version
# Betaflight / STM32F405 (S405) 4.1.2 Dec 22 2019 / 02:19:35 (89051e256) MSP API: 1.42
# manufacturer_id: AIRB   board_name: OMNIBUSF4SD   custom defaults: YES

# start the command batch
batch start

board_name OMNIBUSF4SD
manufacturer_id AIRB

# resources
resource PULLUP 1 B03

# master
set gyro_1_align_yaw = 2700

profile 0

rateprofile 0

# end the command batch
batch end

#

mikeller, 2020-01-13:
@3DRacer: When running the 4.1.1 legacy target (with baro working), what do `status` and `resource show all` return?

3DRacer, 2020-01-13:
Here it is:

# 

# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.28V, Core temp=31degC
Stack size: 2048, Stack address: 0x1000fff0
Config size: 3632, Max available config: 16384
GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
System Uptime: 46 seconds, Current Time: 2020-01-13T16:19:13.504+00:00
CPU:10%, cycle time: 125, GYRO rate: 8000, RX rate: 33, System rate: 9
Voltage: 2 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: None inserted
Arming disable flags: RXLOSS CLI MSP

# 

resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 3
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A15: OSD_CS
B00: MOTOR 1
B01: MOTOR 2
B03: BARO_CS
B04: BEEPER
B05: LED 1
B06: FREE
B07: SDCARD_DETECT
B08: PWM 1
B09: PWM 2
B10: FREE
B11: FREE
B12: SDCARD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: USB_DETECT
C06: PWM 3
C07: PWM 4
C08: INVERTER 6
C09: INVERTER 3
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
D02: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3: MOTOR 4
    CH4: MOTOR 3
TIM3:
    CH3: MOTOR 1
    CH4: MOTOR 2
TIM4:
    CH4: PWM 2
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: PWM 3
    CH2: PWM 4
    CH3: PWM 5
    CH4: PWM 6
TIM9: FREE
TIM10:
    CH1: PWM 1
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: ADC 2
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

#

mikeller, 2020-01-15:
@3DRacer: That all looks ok.
The problem is that a lot of the cloned boards out there have subtle differences in the hardware, like missing pull ups and stuff like that, and no clone manufacturer ever bothers to support us or their users by supplying us with engineering samples. This makes it very hard to effectively debug problems with these boards.

3DRacer, 2020-01-15:
The mistake with the Baro is not tragic. (I) can easily do without it. But I am afraid that in the future the unified target system will also lose other functions that are possible with the legacy devices.

if you think there will be no solution, this can be closed here. should I?

mikeller, 2020-01-15:
Yes, functionality from boards sold by entities not working with Betaflight might be lost - there is not much that we can do about this. And yes, this can probably be closed.

closed 2020-02-22 by stale[bot]