betaflight/betaflight #9336: Option to inject accel/gyro data as audio in the FPV feed for mic-less quads

TiagoTiago, 2020-01-04:
**Is your feature request related to a problem? Please describe.**
Some quads lack microphones, but some pilots still prefer to have some audio feedback of how the quad is performing.

**Describe the solution you'd like**
An option to inject the accelerometer and/or gyroscope data as audio in the FPV feed; a highpass filter would likely be required to remove the DC offset of gravity and of some maneuvers (though maybe that could be optional, perhaps some pilots wouldn't mind have the popping as an indication or flipping, or hearing the shaking of sticks or whatever). I image obtaining the audio waveform would be just a matter of getting the magnitude (sqrt(x²+y²+z²)) and multiplying it by the sign of any one of the axes (that last part, to ensure the waveform goes both positive and negative instead of bouncing at zero; though perhaps to avoid noise from one axis having very low magnitude and jittering around zero, perhaps it could pick whichever axis had the largest absolute value in the last N seconds or something of the sort). 

A setting for sensitivity/volume would likely be welcome; and the option of selecting accel, gyro, or both, would probably be a good idea too (at least until enough people use it for a while to figure out which would work better). I understand the sampling rate would be less than of conventional microphones, but for some pilots any audio could be better than nothing.

duezero, 2020-01-04:
What do you mean by 

> how the quad is performing

 Audio based on RPM data would make more sense IMO

etracer65, 2020-01-04:
The audio signal line from the vtx is not connected to the flight controller on any existing board (except maybe a couple that have on-board microphones).

TiagoTiago, 2020-01-04:
> What do you mean by
> 
> > how the quad is performing
> 
> Audio based on RPM data would make more sense IMO

This would provide more than just the RPM though, would help with getting a feel for prop/motor health, and maybe some extra insight on what's going on during tricky "crash flip" situations where the props may be hitting something.

TiagoTiago, 2020-01-04:
> The audio signal line from the vtx is not connected to the flight controller on any existing board (except maybe a couple that have on-board microphones).

How is the OSD injected into the FPV signal?  I had assumed it was after it had already been made into a "TV" signal, but before it was modulated into the selected band. But to be honest, I haven't looked up how it's done. Is it before there is any audio channel available to be manipulated? :(

Would it be possible to hook up an output pin to the audio input of the VTX and output audio that way? (of course the signal wouldn't be clean analog, but it's possible to make sound with digital outputs; pulse-density-modulation IIRC)

Or are the characteristics of the circuits inappropriate for that (impedance, voltage etc)?

etracer65, 2020-01-05:
> How is the OSD injected into the FPV signal? I had assumed it was after it had already been made into a "TV" signal, but before it was modulated into the selected band. But to be honest, I haven't looked up how it's done. Is it before there is any audio channel available to be manipulated? :(

Using a MAX7456 which only uses the video component. There is no connection for an inbound audio signal or opportunity to manipulate it.

The flight controller technically *could* generate an audio output signal that could be fed to the VTX. The problem is that it would require a DAC pin which is not accessible on nearly every flight controller. Also needs support circuitry to convert the 0-3.3v DC output to an appropriate range AC waveform (op-amp, filters, etc.) - again not available. The exception would be to look at the recent SPRACING flight controllers as they have a "PID Audio" function that's similar to what you're describing. But this is not possible with general flight controller hardware.

TiagoTiago, 2020-01-05:
That's disappointing :(

richardbulger, 2020-01-10:
@TiagoTiago This seems a feature you may be interested in: https://youtu.be/mjRWuA7ICZ8?t=127

closed 2020-02-16 by stale[bot]