betaflight/betaflight #9337: SPRACINGH7EXTREME external current measurement via ADC  pad

Xplorer88, 2020-01-04:
Hi,
first of all, thanks for this great software!
I’m having the following problem, I’m not able to make the current sensing work on my SPRACINGH7EXTREME FC. I'm using the following hardware.
FC: SPRACINGH7EXTREME 
ESC: T-Motor F55A PROⅡ 6S 4IN1 32bit

The main battery plug is connected to the ESC and the current sensing output of the ESC is connected to the ADC pad of the flight controller. The resources are set to default (resource ADC_CURR 1 C00)
The current meter source is set to "Onboard ADC". If i plug in the power supply the measured value is 1.18A (power supply shows 0.37A) but if i just keep the USB cable plugged in, the value shows 15.86A.
The 1.18A value does not change even if the motors are spinning.

I did the following tests (ESC current sensor unplugged from ADC pad).
1. Plugged in the power supply (15V) and increased the motor speed until the PS showed 1A
-- the voltage of the ESC current sensor output was at 33,4mV
2. increased the motor speed even more until the PS showed 2A
-- the voltage of the ESC current sensor increased to 84,2mV

I have found the following information on t-motors product page "Amp monitoring port with the amperage of 15.2mV/A...", that would not make sense compared to my measurements.

Did i understand something wrong? Maybe I have to change the resource?

One additional thing that i have noticed is that if the PS is plugged in there is a voltage of about 64mV on the ADC pad (current sensor not connected).

Thanks for Your help

Xplorer88, 2020-01-05:
@etracer65 I have added the full configuration dump to my first post. Do you need some additional information?
Thank You

etracer65, 2020-01-05:
Provide the information requested when you opened the issue.

> **Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)
>
> Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

Xplorer88, 2020-01-05:
Sorry i just selected the configuration issue type so no hint about what information is need was provided.

diff:
```# version
# Betaflight / SPRACINGH7EXTREME (SP7E) 4.1.2 Dec 22 2019 / 02:18:18 (89051e256) MSP API: 1.42

# start the command batch
batch start

board_name SPRACINGH7EXTREME

# name: JS-ApeX

# feature
feature -RSSI_ADC
feature -LED_STRIP
feature -TRANSPONDER
feature GPS
feature ESC_SENSOR

# beeper
beeper -ARMED
beeper -ON_USB

# serial
serial 0 64 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 2 2 115200 57600 0 115200
serial 3 2048 115200 57600 0 115200
serial 4 1024 115200 57600 0 115200

# aux
aux 0 0 0 1900 2100 0 0
aux 1 1 1 900 1100 0 0
aux 2 2 1 1400 1600 0 0
aux 3 46 3 1900 2100 0 0
aux 4 13 2 1275 1475 0 0
aux 5 35 2 1900 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745    0
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860    0
vtxtable band 4 RACEBAND R FACTORY    0    0 5732 5769 5806 5843    0    0
vtxtable band 5 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 3
vtxtable powervalues 14 20 26
vtxtable powerlabels 25 100 400

# master
set acc_calibration = 27,-8,-24
set mag_hardware = NONE
set max_check = 2000
set rssi_channel = 16
set serialrx_provider = FPORT
set serialrx_inverted = ON
set rx_min_usec = 970
set rx_max_usec = 2030
set serialrx_halfduplex = ON
set blackbox_p_ratio = 64
set blackbox_device = SPIFLASH
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set gps_provider = UBLOX
set gps_rescue_sanity_checks = RESCUE_SANITY_OFF
set gps_rescue_min_sats = 6
set pid_process_denom = 1
set osd_cap_alarm = 1300
set osd_alt_alarm = 150
set osd_vbat_pos = 6604
set osd_rssi_pos = 6617
set osd_link_quality_pos = 473
set osd_tim_2_pos = 2081
set osd_throttle_pos = 6209
set osd_current_pos = 6561
set osd_mah_drawn_pos = 6593
set osd_gps_speed_pos = 4322
set osd_gps_lon_pos = 4097
set osd_gps_lat_pos = 4112
set osd_gps_sats_pos = 4129
set osd_home_dir_pos = 4186
set osd_home_dist_pos = 4152
set osd_altitude_pos = 4342
set osd_stat_max_alt = ON
set osd_stat_max_esc_rpm = ON
set osd_stat_total_time = ON
set osd_profile = 2
set debug_mode = DUAL_GYRO_SCALED
set vtx_band = 4
set vtx_channel = 6
set vtx_power = 1
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5843
set camera_control_ref_voltage = 325
set name = JS-ApeX

profile 0

rateprofile 0

# end the command batch
batch end
```

resource:
```
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 1
A01: MOTOR 2
A02: MOTOR 3
A03: MOTOR 4
A04: FREE
A05: FREE
A06: VTX_DATA 10
A07: VTX_CLK 10
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: GYRO_CS 1
B00: VTX_CS 10
B01: VTX_POWER
B02: QSPI_CLK 1
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SERIAL_RX 5
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: QSPI_BK1CS 1
B11: FREE
B12: GYRO_CS 2
B13: FREE
B14: SERIAL_TX 1
B15: FREE
C00: ADC_CURR
C01: ADC_BATT
C02: SPI_MISO 2
C03: SPI_MOSI 2
C04: FREE
C05: ADC_EXT
C06: FREE
C07: FREE
C08: SDIO_D0
C09: SDIO_D1
C10: SDIO_D2
C11: SDIO_D3
C12: SDIO_CK
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: SERIAL_TX 4
D02: SDIO_CMD
D03: SPI_SCK 2
D04: GYRO_EXTI
D05: FREE
D06: SPI_MOSI 3
D07: BEEPER
D08: SERIAL_TX 3
D09: SERIAL_RX 3
D10: FREE
D11: QSPI_BK1IO0 1
D12: QSPI_BK1IO1 1
D13: QSPI_BK1IO3 1
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: QSPI_BK1IO2 1
E03: LED 1
E04: SYSTEM
E05: CAMERA_CONTROL
E06: FREE
E07: QSPI_BK2IO0 1
E08: QSPI_BK2IO1 1
E09: QSPI_BK2IO2 1
E10: QSPI_BK2IO3 1
E11: OSD_CS
E12: SPI_SCK 4
E13: SPI_MISO 4
E14: SPI_MOSI 4
E15: GYRO_EXTI
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE
G00: FREE
G01: FREE
G02: FREE
G03: FREE
G04: FREE
G05: FREE
G06: FREE
G07: FREE
G08: FREE
G09: FREE
G10: FREE
G11: FREE
G12: FREE
G13: FREE
G14: FREE
G15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5:
    CH1: MOTOR 1
    CH2: MOTOR 2
    CH3: MOTOR 3
    CH4: MOTOR 4
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE
TIM15:
    CH1: CAMERA_CONTROL
TIM16: FREE
TIM17: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 1
DMA1 Stream 1: MOTOR 2
DMA1 Stream 2: MOTOR 3
DMA1 Stream 3: MOTOR 4
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: ADC 3
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

etracer65, 2020-01-05:
That flight controller has an onboard current sensor so I would presume that it would be connected to the `ADC_CURR` pin. So not sure what the "ADC" pad is connected to and how the user would change between the onboard current sensor and the external sensor connected to the ADC pin. @hydra ?

Maybe you have the board wired incorrectly and the onboard current sensor is active and conflicting with the external connection.

hydra, 2020-01-07:
there's an external ADC pad which can be used instead of the onboard current sensor, to make it work swap the resources for the ADC_CURR and ADC_EXTERNAL over using the cli `resource` command.

The SPRacingH7EXTREME manual has the instructions for wiring up the FC to 4in1 or separate ESCs.  In short, for a 4in1 it goes:

Battery->XT60->FC Battery IN pads->FC 4in1 OUT pads->4in1 ESC Battery IN pads.

Let us know how you get on.

Xplorer88, 2020-01-07:
Today i swapped the resources of ADC_CURR and ADC_EXT. Now if just the USB cable is plugged in the amps are 0, if i connect the power supply the value jumps to about 7A. 
The value is not very stable and jumps from 6.2 to 7.8A up and down (power supply: 0.38A). If i switch on the motors the value is not increasing much, about 0.2A (power supply: 1A ). The upcoming weekend i will try see how the value changes during flight.

Btw i don't know which scale i have to set, unfortunately it is not mentioned in the ESC manual. They just wrote "Amp monitoring port with the amperage of 15.2mV/A...". The Calibration tool in the configurator suggests a scale of about 5000. At the moment the value is set to 220.

I would prefer to not connect the battery directly to the FC and from FC 4in1 to the ESC because there is not much space in the frame.

A picture how i wired up the FC with the ESC, i hope this works as well
[image]

hydra, 2020-01-09:
@Xplorer88 yes that connection diagram looks fine if you don't want to use the on-board current sensor.  Let us know how you get on with the calibration, remember there is an offset *and* a scale, you generally need to configure both.

Xplorer88, 2020-02-09:
@hydra finally i had some time to test if the current sensing is working as expected. But unfortunately, it's not the case. I flew some freestyle with several punches but the max current at the end was on 14A. Do you have another idea what could be wrong? I'll ask T-Motor if they have an idea what could be wrong and how to figure out if the current sensor on the ESC is working properly.

hydra, 2020-02-17:
suggest logging the flights using blackbox logging, there might be a debug mode for the current sensor too.  once you've done that post photos of your wiring, cli `dump` command output and a flight log.

Xplorer88, 2020-03-08:
Thank you hydra, i was a bit busy and hadn't time the fly the past weeks. I'll try to post all the logs and configs asap.

Xplorer88, 2020-03-08:
@hydra on the following link you can find the Blackbox file with DEBUG on CURRENT_SENSOR enabled. There are different logs in the file, the second is the best one imo. I just flew LOS without goggles.
https://1drv.ms/u/s!AlMvqDMe29VagdwTAf8KwngkQ4H6aw?e=ZwPA8k

It looks like the current sensor is kinda working, but the amperage value is always very low. Even on punches with  >80% throttle. It is a 6S quad with 2000kV motors with a throttle limit of 90%. I'll post some photos next week if still needed.

**dump/resource**

```
# dump

# version
# Betaflight / SPRACINGH7EXTREME (SP7E) 4.1.3 Jan 16 2020 / 11:32:30 (543a5e1ba) MSP API: 1.42

# start the command batch
batch start

board_name SPRACINGH7EXTREME
manufacturer_id 

# name: JS-ApeX

# resources
resource BEEPER 1 D07
resource MOTOR 1 A00
resource MOTOR 2 A01
resource MOTOR 3 A02
resource MOTOR 4 A03
resource MOTOR 5 B06
resource MOTOR 6 B07
resource MOTOR 7 C06
resource MOTOR 8 C07
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B15
resource PWM 1 B15
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 B14
resource SERIAL_TX 2 D05
resource SERIAL_TX 3 D08
resource SERIAL_TX 4 D01
resource SERIAL_TX 5 B13
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 E01
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_RX 1 B15
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 D09
resource SERIAL_RX 4 D00
resource SERIAL_RX 5 B05
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 E00
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource I2C_SCL 1 B08
[... 1109 lines cut ...]
resource SPI_MOSI 4 E14
resource CAMERA_CONTROL 1 E05
resource ADC_BATT 1 C01
resource ADC_RSSI 1 C04
resource ADC_CURR 1 C05
resource ADC_EXT 1 C00
resource SDCARD_DETECT 1 D10
resource SDIO_CK 1 C12
resource SDIO_CMD 1 D02
resource SDIO_D0 1 C08
resource SDIO_D1 1 C09
resource SDIO_D2 1 C10
resource SDIO_D3 1 C11
resource OSD_CS 1 E11
resource GYRO_EXTI 1 D04
resource GYRO_EXTI 2 E15
resource GYRO_CS 1 A15
resource GYRO_CS 2 B12
resource VTX_POWER 1 B01
resource VTX_CS 1 B00
resource VTX_DATA 1 A06
resource VTX_CLK 1 A07

# 
```

closed 2020-04-14 by stale[bot]