betaflight/betaflight #9339: Crossfire frequency switch back to previous selectio

oranmike28, 2020-01-06:
**Describe the bug**
IDK of it's because BF 4.1.1 or not but it happens after switching to BF 4.1.1 and
 my CRSF Tx and Rx was updated to latest version by TBS AgentX (as of 3rd JAN 2020) and OpenTX to version 2.3.4

My Crossfire Micro Tx and CRSF nano Rx were set to 915mhz then I changed it to 868 mhz by
1.Turn on Taranis with CRSF micro Tx on its back
2. Plug in the quad's battery
3. Execute [CROSSFIRE. LUA] in my taranis
4. Go into Tx option and change from 915 to 868
5. Exit the page
 Then it works normally but when I finished a flight, unplug quad's battery and turn off the Taranis the next time I turn on Taranis and plug in the quad they are in differ frequency again( It looks like CRSF Rx switch back to 915 Mhz) . I have to redo it again; change Tx frequency to 915Mhz then plug in quad battery, then change to 868Mhz then fly. 

But if I leave the Taranis on all the time then the CRSF Rx stay on 868Mhz.

**To Reproduce**
Just simply changing CRSF frequency. 

**Expected behavior**
Nothing serious just the CRSF Rx go back to the previous frequency everytime transmitter is turned off. 

**Flight controller configuration**
I don't think it's related. Just normal CRSF configuration. 

**Setup / Versions**
 - BrainFPV RadixLi from the BrainFPV website with BF4.1.1
 - TBS CRSF micro Tx and Nano Rx with latest FW as of 3rd JAN 2020. Taranis X9D plus with OpenTX 2.3.4
 - CRSF Nano Rx to FC like what BrainFPV instruction said.

etracer65, 2020-01-06:
First of all, this is not a Betaflight issue. Betaflight has no control over your crossfire settings. Recommend you contact TBS for support.

Second, we don't support the BrainFPV boards. They have their own fork and develop a custom version of Betaflight for them.

closed 2020-01-06 by etracer65

oranmike28, 2020-01-07:
Totally understand that it is not BF issue as it happens with another quad
with BF4.0.5 too. But Woww! just know that BrainFPV FW is out of BF dev
hands. Things get more complicated!

Thanks for your works!
Oran