betaflight/betaflight #9340: SPRACINGH7EXTREME UART6 not working

danielolsman, 2020-01-06:
1. SPRACINGH7EXTREME UART6 not working
2. GPS is wired to UART6 and UART3, when using UART6 (and configuring it) it's not detecting the gps (BN-180) module, when using UART3 everything is working fine
3. SPRACINGH7EXTREME with a Beitian BN-180 GPS module wired to UART6
4. # Betaflight / SPRACINGH7EXTREME (SP7E) 4.1.0 Oct  6 2019 / 06:46:02 (c6452a55c) MSP API: 1.42
5. 
```# version
# Betaflight / SPRACINGH7EXTREME (SP7E) 4.1.0 Oct  6 2019 / 06:46:02 (c6452a55c) MSP API: 1.42

# start the command batch
batch start

board_name SPRACINGH7EXTREME
manufacturer_id 
mcu_id 0041002f3437511931353431
signature 

# name: DanO

# resources
resource BEEPER 1 D07
resource MOTOR 1 A00
resource MOTOR 2 A01
resource MOTOR 3 A02
resource MOTOR 4 A03
resource MOTOR 5 B06
resource MOTOR 6 B07
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B15
resource PWM 1 B15
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 B14
resource SERIAL_TX 2 D05
resource SERIAL_TX 3 D08
resource SERIAL_TX 4 D01
resource SERIAL_TX 5 B13
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 E01
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_RX 1 B15
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 D09
resource SERIAL_RX 4 D00
resource SERIAL_RX 5 B05
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 E00
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource I2C_SCL 1 B08
resource I2C_SCL 2 NONE
resource I2C_SCL 3 NONE
resource I2C_SCL 4 NONE
[... 1349 lines cut ...]
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 65
set tpa_breakpoint = 1250
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# restore original rateprofile selection
rateprofile 0

# save configuration
save```

etracer65, 2020-01-06:
Can you please provide the `diff` as requested? The `dump` shows every setting and it's very difficult to review.

danielolsman, 2020-01-06:
> Can you please provide the `diff` as requested? The `dump` shows every setting and it's very difficult to review.

sorry, my bad. changed it

etracer65, 2020-01-06:
@hydra Any ideas?

hydra, 2020-01-07:
@etracer65 Not currently, needs some investigation.  Seems a bit odd for sure.

@danielolsman Can you give the output of the CLI `dump hardware` please.

danielolsman, 2020-01-08:
```
# dump hardware

# version
# Betaflight / SPRACINGH7EXTREME (SP7E) 4.1.0 Oct  6 2019 / 06:46:02 (c6452a55c) MSP API: 1.42

# start the command batch
batch start

board_name SPRACINGH7EXTREME
manufacturer_id 

# resources
resource BEEPER 1 D07
resource MOTOR 1 A00
resource MOTOR 2 A01
resource MOTOR 3 A02
resource MOTOR 4 A03
resource MOTOR 5 B06
resource MOTOR 6 B07
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B15
resource PWM 1 B15
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 B14
resource SERIAL_TX 2 D05
resource SERIAL_TX 3 D08
resource SERIAL_TX 4 D01
resource SERIAL_TX 5 B13
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 E01
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_RX 1 B15
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 D09
resource SERIAL_RX 4 D00
resource SERIAL_RX 5 B05
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 E00
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource I2C_SCL 1 B08
resource I2C_SCL 2 NONE
resource I2C_SCL 3 NONE
resource I2C_SCL 4 NONE
resource I2C_SDA 1 B09
resource I2C_SDA 2 NONE
resource I2C_SDA 3 NONE
resource I2C_SDA 4 NONE
resource LED 1 E03
resource LED 2 NONE
[... 226 lines cut ...]
set max7456_spi_bus = 4
set max7456_preinit_opu = OFF
set led_inversion = 0
set dashboard_i2c_bus = 1
set dashboard_i2c_addr = 60
set flash_spi_bus = 0
set gyro_1_bustype = SPI
set gyro_1_spibus = 3
set gyro_1_i2cBus = 0
set gyro_1_i2c_address = 0
set gyro_1_sensor_align = CW180
set gyro_1_align_roll = 0
set gyro_1_align_pitch = 0
set gyro_1_align_yaw = 1800
set gyro_2_bustype = SPI
set gyro_2_spibus = 2
set gyro_2_i2cBus = 0
set gyro_2_i2c_address = 0
set gyro_2_sensor_align = CUSTOM
set gyro_2_align_roll = 0
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 2250

# end the command batch
batch end

jflyper, 2020-01-13:
Simple GPS config detect the GPS on UART6 with NUCLEOH743 target.

It must be one of
1. H7EXTREME specific hardware issue
2. An issue related with other config items
3. Config problem when moving the GPS from UART3 to UART6 (most unlikely).

[image]

hydra, 2020-01-17:
@jflyper what CPU pins did you use for UART6?  You don't have a dump do you by any chance?

jflyper, 2020-01-21:
They are NUCLEOH743 target default... PC6 and PC7 for TX and RX, respectively.

hydra, 2020-01-21:
Should be fine then, I'm wondering if the specific board that @danielolsman has perhaps has a soldering issue with the two pins on the CPU.

@danielolsman I'll attach a picture of the CPU pins to check, you can check them for continuity to adjacent CPU pins and to each other and check for continuity from pads to CPU pins.

I'm at the post office now but will try and update this later today with the pin numbers or you can look in the H750 CPU datasheet.

hydra, 2020-01-21:
[image]
[image]

@danielolsman  There you are.  Those are the pin names, IO assignments and physical location.

Please take a couple of high-res, in-focus photo of the CPU legs *before* probing and attach/upload them here.

Please do as above and check continuitity, measure from the top of the CPU pins to.  Don't exert too much pressure when pressing your probe tips to the legs of the CPU, to avoid damaging them.

The PCB's themselves should be fine, as each PCB is tested in the factory for continuity on all 'nets'.

hydra, 2020-01-21:
@danielolsman @jflyper I just tested enabling MSP on UART6 and hooked up a 3v USB-UART to M7/M8 pads and all is working as expected.

```
# 
# Building AutoComplete Cache ... Done!
# 
# diff

# version
# Betaflight / SPRACINGH7EXTREME (SP7E) 4.1.3 Jan 16 2020 / 11:32:30 (543a5e1ba) MSP API: 1.42

# start the command batch
batch start

board_name SPRACINGH7EXTREME

# serial
serial 0 0 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 5 1 115200 57600 0 115200

profile 0

rateprofile 0

# end the command batch
batch end
```

[image]

[image]

Using 4.1.3 firmware, as per dump.

I suspect a wiring or other hardware issue.  Note that on the CP2102 the TXD/RXD labels on the PCB silk screen are transposed.  i.e.  where it says 'TXD', that's actually RXD.  Note also that on my photo the white wire is joined to a green wire.

DeepnineRC, 2020-01-25:
Same problem here.

Uart 6 MSP with FTDI is working
Uart 6 MSP with DJI air unit not working
Uart 6 GPS BN220 not working

danielolsman, 2020-01-25:
If I have to take a picture of the H7 chip I need to dissemble almost the entire quadcopter, so I tried something different. @DeepnineRC reported that MSP with a FTDI is working so I wired up an FTDI and it is working! So there should not be a problem with the soldering of the H7 chip. Then I tried connecting a bn-180 gps module and strangely enough it's working now. I'm sure I triple checked the soldering before and also tried swapping tx/rx. I'm going to do some more testing (probaly flying) to see if it keeps working. @hydra thnx for the effort of testing, I will report back

DeepnineRC, 2020-01-25:
I swapped esc telemetry (uart4) and msp4dji (uart6). Now esc telemetry on uart6 is not working, but msp4dji is working on uart4.

hydra, 2020-01-31:
@danielolsman and @DeepnineRC ok. thanks for the update.   I'll close this issue.

@danielolsman perhaps in your case the GPS device and/or BF GPS related setting were not initially configured correctly.  You can also use serial passthrough mode and the UBlox U-Center app to communicate/ with and configure the GPS from your PC.

If anyone else reading this has issues specifically with the H750 and UART6 on any FC please comment on this issue.

closed 2020-01-31 by hydra