betaflight/betaflight #9344: ADC transposed values bug.

hydra, 2020-01-07:
**Describe the bug**
Transposed ADC readings.

**To Reproduce**
define an H7 target (e.g. NucleoH743) with the following ADC configuration

```
#define USE_ADC
#define USE_ADC_INTERNAL        // ADC3
#define ADC2_INSTANCE ADC2      // ADC2 for RSSI
#define ADC3_INSTANCE ADC3      // ADC3 for TEMP/REF/BATTERY/CURRENT
#define CURRENT_METER_ADC_PIN   PC0
#define VBAT_ADC_PIN            PC3
#define RSSI_ADC_PIN            PC4
```

Note ADC1 is SPECIFICALLY NOT USED.

Use a 10:1 voltage divider, apply 12V to input of divider.  let other ADC pins float.

Observe that battery reading is ~1v, not 12V.

**Expected behavior**
12V reading.

**Setup / Versions**
 - Any debug H7 target with above ADC config.

**Additional context**
From slack:

I'm getting incorrect readings for ADC values for H7, and looking at the code something seems a little off to me.

Code walkthrough:
so, `adcGetChannel`, is called, with a `channel` value of `0` (edited)
https://github.com/betaflight/betaflight/blob/master/src/main/drivers/adc.c#L85

that then calls the H7 driver's `adcGetChannelValues` method.
https://github.com/betaflight/betaflight/blob/master/src/main/drivers/adc_stm32h7xx.c#L450-L458

that code proceeds to copy the values from the internal `adcConversionBuffer` to the `adcValues` global variable, channel by channel, in the order defined by the `AdcChannel` enum in `adc.h`
`ADC_BATTERY` is 0.

after the code in the driver is called, flow returns back to `adcGetChannel` when this does this:
`return adcValues[adcOperatingConfig[channel].dmaIndex];`
https://github.com/betaflight/betaflight/blob/master/src/main/drivers/adc.c#L101

and that doesn't look right to me at all since the driver code already used the dmaIndex to put the right values in `adcValues`. it feels like it should just be return `adcValues[channel];`

with 12V supply connected, and a voltage divider on PC3 giving about 1.08v (10:1 divider)
the values I get are as follows

`adcValues` dump from eclipse/gdb
```
{1355, 122, 122, 104, 0, 0}
```

the values in adcOperatingConfig look correct.  I've NOT enabled ADC1, only ADC2 and 3.

`adcOperatingConfig` dump from eclipse/gdb:
```
{{\n    tag = 51 '3', \n    adcDevice = ADCDEV_3, \n    adcChannel = 70254594, \n    dmaIndex = 2 '\002', \n    enabled = true, \n    sampleTime = 7 '\a'\n  }, {\n    tag = 48 '0', \n    adcDevice = ADCDEV_2, \n    adcChannel = 704644096, \n    dmaIndex = 0 '\0', \n    enabled = true, \n    sampleTime = 7 '\a'\n  }, {\n    tag = 0 '\0', \n    adcDevice = ADCDEV_1, \n    adcChannel = 0, \n    dmaIndex = 0 '\0', \n    enabled = false, \n    sampleTime = 0 '\0'\n  }, {\n    tag = 52 '4', \n    adcDevice = ADCDEV_2, \n    adcChannel = 281018384, \n    dmaIndex = 1 '\001', \n    enabled = true, \n    sampleTime = 7 '\a'\n  }, {\n    tag = 0 '\0', \n    adcDevice = ADCDEV_3, \n    adcChannel = 3414425600, \n    dmaIndex = 3 '\003', \n    enabled = true, \n    sampleTime = 7 '\a'\n  }, {\n    tag = 0 '\0', \n    adcDevice = ADCDEV_3, \n    adcChannel = 3484942336, \n    dmaIndex = 4 '\004', \n    enabled = true, \n    sampleTime = 7 '\a'\n  }}
```

certainly, when I change the code to use `return adcValues[channel];` the values  in the OSD and configurator UI are correct.

I've not looked into the other MCU adc drivers, but this requires some investigation.

hydra, 2020-01-07:
After some investigation, other MCU's have a no-op `adcGetChannelValues` so the problem is H7 specific.

The current H7 code also doesn't handle channels that are not enabled, in the example above `ADC_EXTERNAL1` is NOT enabled.

I think the fix is changing this:

```
void adcGetChannelValues(void)
{
    for (int i = 0; i < ADC_CHANNEL_INTERNAL; i++) {
        adcValues[i] = adcConversionBuffer[adcOperatingConfig[i].dmaIndex];
    }
}
```

to this:

```
void adcGetChannelValues(void)
{
    for (int i = 0; i < ADC_CHANNEL_INTERNAL; i++) {
        if (adcOperatingConfig[i].enabled) {
            adcValues[adcOperatingConfig[i].dmaIndex] = adcConversionBuffer[adcOperatingConfig[i].dmaIndex];
        }
    }
}
```

at least it works on my test target, needs testing on other targets.

hydra, 2020-01-07:
with the other ADC pins floating, here's what the configurator looks like with the above change in place.

[image]

jflyper, 2020-01-15:
@hydra Your fix is correct. I'll do a PR.

PR #9367 (2020-01-16): [H7] Fix ADC DMA buffer transposing bug
Fixes #9344

EDIT
This bug probably needs some explanation.
As H7's DMA buffer coherency management is done with MPU regions, it is necessary to attribute DMA buffer as appropriate. When I've done this, I somehow came to believe that `drivers/adc.c` should be free from DMA details and decided NOT to attribute `adcValues` as DMA buffers, and transposing be done in H7 specific code (hence the current implementation of `adcGetChannelValues()` that transpose elements). The actual transposing is done in `adcGetChannel()` in `drivers/adc.c`, which resulted in  [... cut]
closed 2020-01-16 by mikeller