betaflight/betaflight #9346: MCO to output either I2S or HSE

ModdQuad, 2020-01-07:
I am designing a new F722 FC that needs MCO to output an 8MHz clock to an osd chip.
There is already the ability to output the PLL_I2S_CLK clock on an MCO2 pin in /drivers/mco.c
What is needed is to add a flag to select either PLL_I2S_CLK or HSE_CLK output to MCO2.  (or HSE_CLK to MCO1)

It works on my local build, this change seems easy enough to do myself.  Do I need any special permissions to contribute code?

jflyper, 2020-01-07:
Feel free to contribute. We can discuss on an actual PR.

(In fact, `mcoConfig()` was written for somewhat flex source selection; only CLI interface was missing.)

ModdQuad, 2020-01-07:
Thanks,  I will look into adding a CLI interface before I submit.   My alpha board (flywoo f7 clone) does not require the clock.  So this would be for the beta board.  I hate to add yet another target, but with the osd changes it seems to make sense.

closed 2020-01-07 by ModdQuad